openwrt/openwrt #538: dnsmasq CVEs with RCE and DoS

spinkham, 2017-10-02:
https://security.googleblog.com/2017/10/behind-masq-yet-more-dns-and-dhcp.html
dnsmasq 2.78 fixes these issues.

wigyori, 2017-10-02:
https://github.com/openwrt/openwrt/pull/536

Although we're waiting for the merge, I've already started porting these into trunk this week.

l0ddej0kk0, 2017-10-03:
Am positively delighted to hear that!
Thx a lot.

scoresofunderscores, 2017-10-03:
Awesome.

scoresofunderscores, 2017-10-03:
Oh right will the fix be in a new chaos calmer binary or will we need to build openwrt from source after the merge?

l0ddej0kk0, 2017-10-03:
Here's hoping it will take the form of an OPK package dnsmasq with the new version 2.78 ...

the-real-tokai, 2017-10-16:
Since "upgrading" to LEDE bricked my Linksys WRT200AC (luckily I was able to revert to the previous CC installation) and I simply don't have time to do further experiments with my production router I'm also hoping for a simple package update for dnsmasq.

wigyori, 2017-10-16:
The fix was merged with #543 . At this time I cannot promise a simple update to CC.1, but I've received several requests for this (understandibly) - we need to coordinate that, along with the KRACK fix as well. (Low on manpower, yadayada, please bear with us.)

closed 2022-02-09 by aparcar