openwrt/openwrt #558: WPA2 KRACK

bogdik, 2017-10-16:
Please check  this:

CVE-2017-13077: Reinstallation of the pairwise encryption key (PTK-TK) in the 4-way handshake.
    CVE-2017-13078: Reinstallation of the group key (GTK) in the 4-way handshake.
    CVE-2017-13079: Reinstallation of the integrity group key (IGTK) in the 4-way handshake.
    CVE-2017-13080: Reinstallation of the group key (GTK) in the group key handshake.
    CVE-2017-13081: Reinstallation of the integrity group key (IGTK) in the group key handshake.
    CVE-2017-13082: Accepting a retransmitted Fast BSS Transition (FT) Reassociation Request and reinstalling the pairwise encryption key (PTK-TK) while processing it.
    CVE-2017-13084: Reinstallation of the STK key in the PeerKey handshake.
    CVE-2017-13086: reinstallation of the Tunneled Direct-Link Setup (TDLS) PeerKey (TPK) key in the TDLS handshake.
    CVE-2017-13087: reinstallation of the group key (GTK) when processing a Wireless Network Management (WNM) Sleep Mode Response frame.
    CVE-2017-13088: reinstallation of the integrity group key (IGTK) when processing a Wireless Network Management (WNM) Sleep Mode Response frame.
On DD-WRT patch svn.dd-wrt.com/changeset/33525

crosser, 2017-10-16:
I did not look into the issue very closely, but my initial understanding is that hostapd and wpa_supplicant may need updating.

thodt-md, 2017-10-16:
They are already updated, if I read the log correctly: https://w1.fi/cgit/hostap/log/

makedir, 2017-10-16:
So doing this:

opkg update
opkg upgrade hostapd-common
opkg upgrade wpad-mini

should fix it? Why are those builds though called 2016-06-15? Is it save to do these updates or would it break something or need newer kernel modules too or any other dependencies?

darkprof83, 2017-10-17:
[patches](https://w1.fi/security/2017-1/) for WPA2 KRACK

crosser, 2017-10-18:
As far as I understand (I may be wrong), if you flash `trunk` firmware, you will get fixed wpad (hostapd and wpa_supplicant). But I do not see updated packages for the stable release.

Attempt to install hostapd/wpad packages from trunk onto the system running 15.05.1 does not work (breaks the system).

It would be nice if patched packages for stable system where made available in the package repos. Then upgrading would indeed be as easy as `opkg update; opkg upgrade ...`

wigyori, 2017-10-18:
Although initially we only wanted to provide sources for upgrading hostapd in CC, we're currently doing a build for all targets supported by CC.1. Will be ready in the next few days. 

Trunk users should be able to do an opkg update/upgrade by now (with the buildbots munching through the daily builds).

hzjnet, 2017-10-20:
如何下载补丁文件啊？

r57zone, 2017-10-28:
Maybe also make stable release 15.05.2 with these fixes.

DalSoft, 2017-11-25:
Sorry can I now do:
```bash 
opkg` update
opkg upgrade hostapd-common wpad-mini
```
On 15.05.2 stable?

closed 2022-02-09 by aparcar