openwrt/openwrt #566: CC: hostapd-supplicant-full unknown type name error

peterwillcn, 2017-10-23:
LIBS="-L/home/openwrt/15.05.1/openwrt/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/lib -L/home/openwrt/15.05.1/openwrt/staging_dir/target-x86_64_uClibc-0.9.33.2/lib -L/home/openwrt/15.05.1/openwrt/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/openwrt/15.05.1/openwrt/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/lib -Wl,--gc-sections -lm -lnl-tiny" LIBS_c="" BCHECK=  wpa_cli wpa_supplicant
make[4]: Entering directory `/home/openwrt/15.05.1/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/hostapd-supplicant-full/hostapd-2016-06-15/wpa_supplicant'
make[4]: `wpa_cli' is up to date.
In file included from ../src/drivers/driver.h:22:0,
                 from ../src/drivers/driver_wext.c:32:
/home/openwrt/15.05.1/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/hostapd-supplicant-full/hostapd-2016-06-15/src/drivers/nl80211_copy.h:4746:2: error: unknown type name '__s8'
  __s8 delta;
  ^
make[4]: *** [../src/drivers/driver_wext.o] Error 1
make[4]: Leaving directory `/home/openwrt/15.05.1/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/hostapd-supplicant-full/hostapd-2016-06-15/wpa_supplicant'
make[3]: *** [/home/openwrt/15.05.1/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/hostapd-supplicant-full/hostapd-2016-06-15/.built] Error 2
make[3]: Leaving directory `/home/openwrt/15.05.1/openwrt/package/network/services/hostapd'
make[2]: *** [package/network/services/hostapd/compile] Error 2
make[2]: Leaving directory `/home/openwrt/15.05.1/openwrt'
make[1]: *** [/home/openwrt/15.05.1/openwrt/staging_dir/target-x86_64_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/openwrt/15.05.1/openwrt'
make: *** [world] Error 2

dztdeng, 2017-11-01:
I do the fellowing change , then compile procedures can continue
`--- a/src/drivers/linux_wext.h.org      2017-11-01 23:10:41.571306502 +0800
+++ b/src/drivers/linux_wext.h  2017-11-01 23:10:59.523179251 +0800
@@ -26,6 +26,7 @@
 typedef uint16_t __u16;
 typedef int16_t __s16;
 typedef uint8_t __u8;
+typedef int8_t __s8;
 #ifndef __user`

dztdeng, 2017-11-01:
I do the fellows change , then compile procedures can continue
`--- a/src/drivers/linux_wext.h.org      2017-11-01 23:10:41.571306502 +0800
+++ b/src/drivers/linux_wext.h  2017-11-01 23:10:59.523179251 +0800
@@ -26,6 +26,7 @@
 typedef uint16_t __u16;
 typedef int16_t __s16;
 typedef uint8_t __u8;
+typedef int8_t __s8;
 #ifndef __user`

wigyori, 2017-11-01:
Are you using any of the drivers that'd require WEXT support (i.e. airo, hermes, ipw*) ?

closed 2022-02-09 by aparcar