openwrt/openwrt #567: OpenWrt hangsup/restarts when connecting the USB device DVBSKY S960

groesser, 2017-10-23:
Hi,
Target-HW: TP-Link TL-WR1043N/ND v1
Firmware Version: OpenWrt Chaos Calmer 15.05.1 r49389 / LuCI for-15.05 branch (git-17.136.58961-13aa5ff)

I built an image based on "git clone -b chaos_calmer git://github.com/openwrt/openwrt.git".
(Current trunk has the same fault behaviour)

I updated the "packages" and "luci" feeds.

In "make kernel_menuconfig" i added the following options in addition:
----------------------------------------------------------------------
Device Drivers  --->
	Input device support  --->
		<*> Generic input layer (needed for keyboard, mouse, ...)
	[*] USB support  --->
		<*>   Support for Host-side USB 
	<*> Multimedia support  --->
		[*]   Digital TV support
		[*]   Remote Controller support
		[*]   Media USB Adapters  --->
			<*>   Support for various USB DVB devices v2
				<M>     DVBSky USB support
		[*]   Autoselect ancillary drivers (tuners, sensors, i2c, frontends)     

After upgrading the device to the new generated image (sysupgrade) i copied the driver (dvb-usb-dvbsky.ko) from:
build_dir/target-mips_34kc_uClibc-0.9.33.2/Linux-ar71xx_generic/Linux-3.18.75/drivers/media/usb/dvb-usb-v2
to:
/lib/modules/3.18.75

With "modprobe dvb-usb-dvbsky" i loaded the driver:
[   89.450000] usbcore: registered new interface driver dvb_usb_dvbsky

Plugging in the usb device (1st Try):
[  104.040000] usb 1-1: new high-speed USB device number 3 using ehci-platform
[  104.390000] usb 1-1: dvb_usb_v2: found a 'DVBSky S960/S860' in warm state
[  104.390000] usb 1-1: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[  104.400000] DVB: registering new adapter (DVBSky S960/S860)
[  104.410000] usb 1-1: dvbsky_usb MAC address=00:17:42:54:96:0c
[  104.410000] usb 1-1: dvb_usb_v2: MAC address: 00:17:42:54:96:0c
[  104.420000] DVB: Unable to find symbol m88ds3103_attach()
[  104.430000] usb 1-1: dvbsky_s960_attach fail.

After that i copied the file "m88ds3103.ko" from:
build_dir/target-mips_34kc_uClibc-0.9.33.2/Linux-ar71xx_generic/Linux-3.18.75/drivers/media/dvb-frontends
to:
/lib/modules/3.18.75

With "modprobe m88ds3103" i loaded it. -> No output in kernel log, but listed by "lsmod".

Now, plugging in the usb device causes a reboot of OpenWrt. I am not able to connect the device correctly.

=> Any ideas what's going wrong here? It seems that the problem is located in "dvb-usb-dvbsky.ko" or "m88ds3103.ko".
May be i am doing something wrong here. I loaded "m88ds3103" because this symbol couldn't be found in the first try. Is this correct? modprobe shows no dependency of "dvb-usb-dvbsky.ko" to "m88ds3103.ko".

Regards,
Marcus Grösser

wigyori, 2017-10-23:
Might be a dumb question, but is the PSU powerful enough? If it's a v1, then the PSU might be quite old, and/or not performing to its full power, causing a reset when the device is properly initialized.

Do you have an output of the serial console? That might tell if there was simply a reset, or a kernel panic.

Thanks,
-w-

groesser, 2017-10-24:
Hi,Thank you for your fast response :).I don't think that the PSU has too less power, because the connected usb tuner has it's own power supply.I try to get a system log. I am not so familiar with that but I am looking for it. Or do you have a short description how to do this?Thanks,Marcus-------- Originalnachricht --------Betreff: Re: [openwrt/openwrt] OpenWrt hangsup/restarts when connecting the USB device DVBSKY S960 (#567)Von: Zoltan Herpai An: openwrt/openwrt Cc: groesser ,Author Might be a dumb question, but is the PSU powerful enough? If it's a v1, then the PSU might be quite old, and/or not performing to its full power, causing a reset when the device is properly initialized.
Do you have an output of the serial console? That might tell if there was simply a reset, or a kernel panic.
Thanks,
-w-

wigyori, 2017-10-24:
https://wiki.openwrt.org/toh/tp-link/tl-wr1043nd#serial

Use this to get the serial console. :)  It will require some soldering - but I'd say this is the easiest to see what's going on with the router when you enable the DVBSKY device.

-w-

groesser, 2017-10-24:
Perfekt,When the connection is established, does openwrt directly write to putty or needs it any debugging command?-------- Originalnachricht --------Betreff: Re: [openwrt/openwrt] OpenWrt hangsup/restarts when connecting the USB device DVBSKY S960 (#567)Von: Zoltan Herpai An: openwrt/openwrt Cc: groesser ,Author https://wiki.openwrt.org/toh/tp-link/tl-wr1043nd#serial
Use this to get the serial console. :)  It will require some soldering - but I'd say this is the easiest to see what's going on with the router when you enable the DVBSKY device.
-w-

groesser, 2017-10-24:
You are right:

I logged the serial console to putty.log (see attached file):

Line 187-189: I loaded the driver module (dvb-sky-dvbsky) and m88ds3103.

Line 196: I plugged in the usb tuner.

Line 598: A Kernel panic is detected

 

Marcus 

 

https://wiki.openwrt.org/toh/tp-link/tl-wr1043nd#serial

groesser, 2017-10-30:
Issue resolved...
For proper work, there are some more kernel modules need:
Here is the list of all the needed additional kernel modules:
dvb-usb-dvbsky.ko
m88ds3103.ko
m88ts2022.ko ( -> needs: regmap-core.ko, regmap-i2c.ko, lzo_compress.ko, lzo_decompress.ko)

Successful connection:
[  193.430000] usb 1-1: new high-speed USB device number 3 using ehci-platform
[  193.780000] usb 1-1: dvb_usb_v2: found a 'DVBSky S960/S860' in warm state
[  193.780000] usb 1-1: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[  193.790000] DVB: registering new adapter (DVBSky S960/S860)
[  193.800000] usb 1-1: dvbsky_usb MAC address=00:17:42:54:96:0c
[  193.810000] usb 1-1: dvb_usb_v2: MAC address: 00:17:42:54:96:0c
[  193.820000] i2c i2c-0: Added multiplexed i2c bus 1
[  193.930000] m88ts2022 1-0060: Montage M88TS2022 successfully identified
[  193.940000] usb 1-1: DVB: registering adapter 0 frontend 0 (Montage M88DS3103)...
[  193.940000] IR keymap rc-dvbsky not found
[  193.950000] Registered IR keymap rc-empty
[  193.950000] input: DVBSky S960/S860 as /devices/platform/ehci-platform/usb1/1-1/rc/rc0/input0
[  193.960000] rc0: DVBSky S960/S860 as /devices/platform/ehci-platform/usb1/1-1/rc/rc0
[  193.970000] usb 1-1: dvb_usb_v2: schedule remote query interval to 300 msecs
[  193.980000] usb 1-1: dvb_usb_v2: 'DVBSky S960/S860' successfully initialized and connected

Thanks to wigyori !!! ;-)
Regards,
groesser

closed 2017-10-30 by groesser

wigyori, 2017-11-01:
No problem. :)   Would you want to send a patch to support this kmod as a package? ;)

-w-