openwrt/openwrt #5048: Kernel 5.10 progress

aparcar, 2022-02-09:
Targets using 5.10

ToDo:
- [ ] ath25 https://github.com/openwrt/openwrt/pull/4890
- [ ] bcm63xx https://github.com/openwrt/openwrt/pull/4616
- [ ] octeon https://github.com/openwrt/openwrt/pull/4610

Done:
- [x] apm821xx (testing https://github.com/openwrt/openwrt/pull/4617)
- [x] arc770 (dropped https://github.com/openwrt/openwrt/commit/fcb29171bc9a3a80c2685ef23ba6552abef04bbb)
- [x] archs38 via da9a8fc
- [x] armvirt (via: https://github.com/openwrt/openwrt/pull/4618)
- [x] at91 (test [patchwork](http://patchwork.ozlabs.org/project/openwrt/patch/mailman.22615.1632130083.1072.openwrt-devel@lists.openwrt.org/))
- [x] ath79 (testing https://github.com/openwrt/openwrt/pull/4619)
- [x] bcm27xx (via: https://github.com/openwrt/openwrt/commit/e1142041b0e65d9b20b65653fdb55d15abd37a42)
- [x] bcm47xx via https://github.com/openwrt/openwrt/commit/29f12fefc01039e9a14fade54b24e5ba8be35f9e
- [x] bcm4908 via f1a074d0ce
- [x] bcm53xx (via: https://github.com/openwrt/openwrt/commit/f55f1dbaad33a259e854bdad93d8adcef00a5629)
- [x] bmips (via: https://github.com/openwrt/openwrt/commit/029093a302c9a66b74bec46285a179abd122a40a)
- [x] gemini (https://github.com/openwrt/openwrt/commit/f7e00d81bca720a5a5b073d614982b2f57f7da71)
- [x] imx6 via adc37b8
- [x] ipq40xx (testing https://github.com/openwrt/openwrt/pull/4620)
- [x] ipq806x (via: https://github.com/openwrt/openwrt/commit/0470159552641c2b11ccc1b0fcfcb4ea08f2c6ab)
- [x] kirkwood (testing https://github.com/openwrt/openwrt/pull/4614)
- [x] lantiq (testing: https://github.com/openwrt/openwrt/pull/4542)
- [x] layerscape https://github.com/openwrt/openwrt/pull/4894
- [x] malta (via: https://github.com/openwrt/openwrt/pull/4613)
- [x] mediatek (via: https://github.com/openwrt/openwrt/commit/8f2cace0b7f2f9221ce3dd5a01772c17e5244c3e https://github.com/openwrt/openwrt/commit/c46ccb69d17e584479df849a107423175a143c83)
- [x] mpc85xx (testing https://github.com/openwrt/openwrt/pull/4612)
- [x] mvebu (via: https://github.com/openwrt/openwrt/commit/c9042202ad39e447a21afd91ec777bbc6cc8b72c)
- [x] mxs (testing https://github.com/openwrt/openwrt/pull/4611)
- [x] octeontx https://github.com/openwrt/openwrt/pull/4609
- [x] omap (https://github.com/openwrt/openwrt/pull/4539)
- [x] oxnas (testing https://github.com/openwrt/openwrt/pull/4608)
- [x] pistachio via 646c011
- [x] ramips (testing [patchwork](https://patchwork.ozlabs.org/project/openwrt/patch/20210909084441.1756-1-rsalvaterra@gmail.com/)) assigned to @rsalvaterra
- [x] realtek (testing https://github.com/openwrt/openwrt/pull/4607)
- [x] rockchip (testing https://github.com/openwrt/openwrt/pull/4604)
- [x] sunxi (testing https://github.com/openwrt/openwrt/pull/4606)
- [x] tegra (testing https://github.com/openwrt/openwrt/pull/4605)
- [x] uml (testing https://github.com/openwrt/openwrt/pull/4704) @aparcar
- [x] x86 (via: https://github.com/openwrt/openwrt/commit/64be0eadc17988f29d0a4b89569840d917746498)
- [x] zynq via 88537e7

rsalvaterra, 2022-02-09:
@aparcar, not only ramips is done, 5.4 support has been dropped already. 😉

aparcar, 2022-02-27:
All updated to 5.10.

closed 2022-02-27 by aparcar