openwrt/openwrt #5053: Archer c6 v3 snapshot: unstable connection, some websites work, others don't

o-alquimista, 2022-02-09:
After the upgrade to `r18776-09f383465e` the connection has been extremely unstable. Sometimes a certain website loads, others don't; sometimes nothing loads.

TP-Link Archer c6 v3.

jow-, 2022-02-09:
Try disabling flow offloading, see if the situation improves. Also wired or wireless? IPv4 or IPv6? If disabling offloading does not help, please try disabling MSS clamping on the WAN zone.

o-alquimista, 2022-02-09:
Wired and wireless, same situation. ipv4 and ipv6 both affected (tested with ipv6.google.com). Will try your suggested workaround.

I regret not having made a backup of an older working snapshot :/ I'll have to build it myself.

o-alquimista, 2022-02-09:
Remember that bug report I opened about not being able to reach a local web server? I had no problems with the connection back then. After you fixed that other problem I downloaded a new snapshot and that's when the instability began.

o-alquimista, 2022-02-09:
Flow offloading must be disabled (its default state), because I didn't explicitly set it.

My defaults in `/etc/config/firewall`:
```
config defaults
	option syn_flood '1'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'
```

o-alquimista, 2022-02-10:
I've built an image based on this revision: https://github.com/openwrt/openwrt/commit/f9782f5bcd13e419abf9c84017ada27fa9764011

I picked it because it's from around December 2021, back when it was working well. This is my workaround for now.

I'll still be able to try newer snapshots. Now I'm comfortable with the build system, should you need me to try a different revision.

svlobanov, 2022-02-10:
> I've built an image based on this revision: [f9782f5](https://github.com/openwrt/openwrt/commit/f9782f5bcd13e419abf9c84017ada27fa9764011)
> 
> I picked it because it's from around December 2021, back when it was working well. This is my workaround for now.
> 
> I'll still be able to try newer snapshots. Now I'm comfortable with the build system, should you need me to try a different revision.

Please try to bisect (between working and problematic commits). When you find the commit that causes the issue, please report it here

o-alquimista, 2022-02-10:
After several tests using different commits of OpenWrt and routers with their original firmware, I've come to the conclusion this is not a problem with the router, but something on the ISP end. It was just a coincidence that it started when it did. I'll close this for now.

closed 2022-02-10 by o-alquimista