openwrt/openwrt #5054: beagleboardblack fails to boot 21.02.1 sdcard image

efx, 2022-02-09:
I have observed the issue described in https://bugs.openwrt.org/index.php?do=details&task_id=4237.

## basic steps
1. download image https://downloads.openwrt.org/releases/21.02.1/targets/omap/generic/openwrt-21.02.1-omap-ti_am335x-bone-black-squashfs-sdcard.img.gz
2. write it to an sdcard (`sudo dd if=openwrt-21.02.1-omap-ti_am335x-bone-black-squashfs-sdcard.img bs=1M of=/dev/mmcblk0`)
3. plug sdcard in and reboot bbb device

## logs

<details><summary>serial log</summary>

```
U-Boot SPL 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600)
Trying to boot from MMC2
Loading Environment from EXT4... 
** Unable to use mmc 0:1 for loading the env **

U-Boot 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600), Build: jenkins-github_Bootloader-Builder-137

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
No match for driver 'omap_hsmmc'
No match for driver 'omap_hsmmc'
Some drivers were not found
Reset Source: Global warm SW reset has occurred.
Reset Source: Power-on reset has occurred.
RTC 32KCLK Source: External.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Loading Environment from EXT4... 
** Unable to use mmc 0:1 for loading the env **
Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
BeagleBone Black:
BeagleBone: cape eeprom: i2c_probe: 0x54:
BeagleBone: cape eeprom: i2c_probe: 0x55:
BeagleBone: cape eeprom: i2c_probe: 0x56:
BeagleBone: cape eeprom: i2c_probe: 0x57:
Net:   eth0: MII MODE
cpsw, usb_ether
Press SPACE to abort autoboot in 0 seconds
board_name=[A335BNLT] ...
board_rev=[000C] ...
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
326 bytes read in 5 ms (63.5 KiB/s)
## Executing script at 80000000
3800360 bytes read in 336 ms (10.8 MiB/s)
58519 bytes read in 17 ms (3.3 MiB/s)
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Device Tree to 8ffee000, end 8ffff496 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.154 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16325-88151b8303)) #0 SMP Sun Oct 24 09:01:35 2021
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: TI AM335x BeagleBone Black
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 16 MiB at 0x9ec00000
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon)
[    0.000000] percpu: Embedded 15 pages/cpu s31116 r8192 d22132 u61440
[... 170 lines cut ...]
[    1.766448] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    1.775009] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[    1.783580] Function entered at [<c06b4294>] from [<c06a4968>]
[    1.789684] Function entered at [<c06a4968>] from [<c05adf3c>]
[    1.795786] Function entered at [<c05adf3c>] from [<c05ac51c>]
[    1.801889] Function entered at [<c05ac51c>] from [<c05ac800>]
[    1.807991] Function entered at [<c05ac800>] from [<c05aaa54>]
[    1.814092] Function entered at [<c05aaa54>] from [<c05ac2b4>]
[    1.820194] Function entered at [<c05ac2b4>] from [<c05ab668>]
[    1.826296] Function entered at [<c05ab668>] from [<c05abaf8>]
[    1.832397] Function entered at [<c05abaf8>] from [<c0145dd0>]
[    1.838500] Function entered at [<c0145dd0>] from [<c01462d4>]
[    1.844601] Function entered at [<c01462d4>] from [<c014b7c0>]
[    1.850705] Function entered at [<c014b7c0>] from [<c01010e8>]
[    1.856807] Exception stack(0xdd9edfb0 to 0xdd9edff8)
[    1.862093] dfa0:                                     00000000 00000000 00000000 00000000
[    1.870654] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    1.879213] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    1.886140] Code: e34c209e e5953024 e2430010 e5937040 (e5978008) 
[    1.892559] ---[ end trace b1ea14783d63706c ]---
[    1.897394] Kernel panic - not syncing: Fatal exception
[    1.902870] Rebooting in 1 seconds..
```

</details>

alexeys85, 2022-02-10:
Have you tried to boot from internal eMMC?

ynezz, 2022-02-10:
> Unable to handle kernel NULL pointer dereference at virtual
 > address 00000008
Nice one. Are you able to reproduce with current development snapshot?

 > PC is at 0xc06b4294
 > LR is at 0xc06a4968

Resolving those addresses to source code locations would help tracking 
down the issue.

efx, 2022-02-11:
> Have you tried to boot from internal eMMC?

Good question; I have not. This was only from the sdcard.

> Nice one. Are you able to reproduce with current development snapshot?

I shall give that a whirl and report back!

ynezz, 2022-02-12:
> PC is at 0xc06b4294

0xc06b4294 is in dsps_musb_init (drivers/usb/musb/musb_dsps.c:430)

> LR is at 0xc06a4968

0xc06a4968 is in musb_probe (drivers/usb/musb/musb_core.c:2217)

ynezz, 2022-02-12:
Looks like it should be fixed in 
https://github.com/torvalds/linux/commit/9ab5d539bc975b8dcde86eca1b58d836b657732e 
which was first fixed in 5.4.155, so should be fixed in the next 21.02.2 
release. In order to confirm that, try to use 
[21.02-SNAPSHOT](https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/omap/generic/) 
and report back.

efx, 2022-02-12:
@ynezz wow, precise diagnosis and clear instructions! 
I will try that snapshot this week and report back.

ynezz, 2022-02-21:
21.02.2 is out, try to reproduce and if it's not fixed, then report back and I'll reopen this.

closed 2022-02-21 by ynezz