openwrt/openwrt #5058: odhcpd: DHCPv6 issue with Windows clients

tiagogaspar8, 2022-02-09:
Hi all,

I've been having a few issues with windows clients and odhcpd's dhcpv6.
The issue is that sometimes duplicate addresses appear on the dhcpv6 messages, and windows stops recognizing almos tall of them... it's a mess...
I'm attaching here a pcapng of some requests I have made so it might help someone diagnose the issue.
[dhcpv6_issue.zip](https://github.com/openwrt/openwrt/files/8036797/dhcpv6_issue.zip)
In the pcapng (which is inside the zip) what I did was:

1. After the issue was occurring I renewed the lease (first two packets) and you can see the issue I'm mentioning
2. I released the lease
3. Then renewed it, now the addresses are fine as you see.

I have no idea of what this issue is.

Openwrt: latest master
End device: Windows 11 with the latest updates (this has been happening for the last months)

Thanks.

dedeckeh, 2022-02-10:
Looking into the pcap traces the Windows client includes duplicate addresses in the renew messages even with an identical IAID while odhcpd acknowledges these addresses in the DHCPv6 reply. This seems like an issue in the Windows DHCPv6 client; the only improvement which can be made in odhcpd is either considering this DHCPv6 renew as invalid (due to identical IAIDs) or ignore the identical IAIDs

tiagogaspar8, 2022-02-10:
I have no ideia how this happens though... I'll put a capture running to see if I can catch when this happens...
But odhcpd should refuse the request the laptop made right? Because it's not conformant

tiagogaspar8, 2022-02-17:
Have you found anything @dedeckeh ? I have read the RFC on dhcpv6 and found Nothing in regards to this issue