openwrt/openwrt #5061: firewall4: Not able to limit by device and direction

dave14305, 2022-02-10:
Using OpenWrt SNAPSHOT r18779-64125ed1d0 on bcm2711, I am trying to configure firewall4 to mangle packets in the postrouting chain, but only when going out the WAN interface eth1. The device and direction options do not seem to be supported in the firewall4 code. Intentional or not?
```
config rule
        option name 'Zoom AF41'
        list proto 'udp'
        option src '*'
        option dest 'wan'
        option dest_port '8801-8810'
        option target 'DSCP'
        option set_dscp 'AF41'
        option direction 'out'
        option device 'eth1'
```
Resulting rules without `oifname "eth1"`:
```
meta nfproto ipv4 udp dport 8801-8810 counter packets 0 bytes 0 ip dscp set af41 comment "!fw4: Zoom AF41" 
meta nfproto ipv6 udp dport 8801-8810 counter packets 0 bytes 0 ip6 dscp set af41 comment "!fw4: Zoom AF41"
```

jow-, 2022-02-10:
https://git.openwrt.org/?p=project/firewall4.git;a=commitdiff;h=5c21714dc503d9cee6a638cb8ab06fb40ae473cc;hp=3eacc97415f877f3a20d1fd9771318041c6d6f8a

jow-, 2022-02-13:
Fixed via 0d1220acdfdc0b73d3d00429175468703f6f7d36

closed 2022-02-13 by jow-