openwrt/openwrt #5065: script in /etc/hotplug.d/block are NOT called when USB storage is plugged in

forever8938, 2022-02-10:
I've installed required packages to mount USB storage. I can mount the device using these commands:
```
  block detect | uci import fstab
  uci set fstab.@mount[-1].enabled='1'
  uci set fstab.@mount[-1].target='/mnt/sda1'
  uci commit fstab
  service fstab boot
```
Now I want to run these commands whenever USB storage in plugged in. So I created a script in /etc/hotplug.d/block to do this. Aside from my script, there are already some scripts /etc/hotplug.d/block, but NONE of them is called when USB storage is plugged in. I've confirmed that by printing log in each of the scripts.

wulfy23, 2022-02-11:
was there an issue template you could use?

you have not;
+ stated your openwrt version
+ stated your device / filesystem type or whether an overlay is in use

were you able to label your bug report?

forever8938, 2022-02-11:
@wulfy23 Sorry about that. I've providec additional information, please confirm.

wulfy23, 2022-02-11:
much better... thankyou for supplying those details...

all I can add to this is I have a block hotplug script also on master/ext4-nooverlay and it runs perfectly ok

I will have a chance to retest it when 21.02.2 comes out... if there are issues, I will report them here

forever8938, 2022-02-11:
I tested again and it works on both versions mentioned above. Could be my mistake. So let me close the issue for now. Sorry for inconvenience.

BTW, /etc/hotplug.d/block scripts are also called during boot time. In that case, logs are not seen, probably because it happens even before logger service is started. I confirmed that by printing logs to a file.

closed 2022-02-11 by forever8938