openwrt/openwrt #5077: Flashing OpenWrt 21.02 on AVM 3000 no longer possible

MichiMunich, 2022-02-12:
I am currently trying to flash OpenWRT 21.02.1 to an AVM 3000 repeater. The instructions are described here: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=148d29d47b1d555d0058dcc5dda450b74c4628ec

Up to step 7 everything works as well.
The initramfs image is also loaded and access via http and SSH are possible.
However, the ubi volumes cannot be removed because there is no /dev/ubi0. There is only /dev/ubi_crtl

I tried with AVM firmware 7.20 and 7.29.
How to find out where the AVM filesystem is now to remove it?

MichiMunich, 2022-02-13:
I have tried to gather some more information.
While the initramfs image is loaded, no UBI volumes are available. 
In the kernel messages I also see that the attempt to attach mtd10 failed.
Maybe someone can give me some guidance based on the information.

root@OpenWrt:~# ubinfo 
UBI version:                    1
Count of UBI devices:           0
UBI control device major/minor: 10:62

root@OpenWrt:~# dmesg | grep ubi
[    1.269661] 0x000001300000-0x000008000000 : "ubi"
[  594.840161] ubi0: attaching mtd10
[  594.841373] ubi0 warning: 0xc067d7fc: error -524 while reading 64 bytes from PEB 0:0, read only 0 bytes, retry
[  594.843318] ubi0 warning: 0xc067d7fc: error -524 while reading 64 bytes from PEB 0:0, read only 0 bytes, retry
[  594.853320] ubi0 warning: 0xc067d7fc: error -524 while reading 64 bytes from PEB 0:0, read only 0 bytes, retry
[  594.863277] ubi0 error: 0xc067d848: error -524 while reading 64 bytes from PEB 0:0, read 0 bytes
[  594.872374] CPU: 2 PID: 2590 Comm: ubiattach Not tainted 5.4.154 #0
[  594.971784] ubi0 error: 0xc06780d8: failed to attach mtd10, error -524
[  609.509754] ubi0: attaching mtd10
[  609.510829] ubi0 warning: 0xc067d7fc: error -524 while reading 64 bytes from PEB 0:0, read only 0 bytes, retry
[  609.512807] ubi0 warning: 0xc067d7fc: error -524 while reading 64 bytes from PEB 0:0, read only 0 bytes, retry
[  609.522769] ubi0 warning: 0xc067d7fc: error -524 while reading 64 bytes from PEB 0:0, read only 0 bytes, retry
[  609.532799] ubi0 error: 0xc067d848: error -524 while reading 64 bytes from PEB 0:0, read 0 bytes
[  609.541984] CPU: 1 PID: 2592 Comm: ubiattach Not tainted 5.4.154 #0
[  609.641362] ubi0 error: 0xc06780d8: failed to attach mtd10, error -524

root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "SBL1"
mtd1: 00080000 00020000 "MIBIB"
mtd2: 00080000 00020000 "QSEE"
mtd3: 00040000 00020000 "CDT"
mtd4: 00080000 00020000 "QSEE_B"
mtd5: 00040000 00020000 "urlader0"
mtd6: 00040000 00020000 "urlader1"
mtd7: 00840000 00020000 "nand-tffs"
mtd8: 00400000 00020000 "uboot0"
mtd9: 00400000 00020000 "uboot1"
mtd10: 06d00000 00020000 "ubi"

jmv2009, 2022-04-30:
Ref: https://forum.openwrt.org/t/fritz-repeater-3000-ubirmvol/119513/20?page=2

MichiMunich, 2022-05-03:
It seems to be a similar problem to #9704 
According to https://wikidevi.wi-cat.ru/AVM_FRITZ!Repeater_3000 the AVM 3000 has a "Macronix MX30LF1G18AC-XKI" NAND chip. However, my boot log gives me a "Macronix MX30LF1G28AD". There have been changes on the part of AVM.
I have attached a current boot log.
[AVM_3000_Boot.log](https://github.com/openwrt/openwrt/files/8608576/AVM_3000_Boot.log)

andyboeh, 2022-05-06:
I think the problem is the hard-coded ECC strength in the ipq4019 dtsi file. The flash IC seems to be properly detected so there is no need for an additional patch.

MichiMunich, 2023-08-13:
Fixed by https://github.com/openwrt/openwrt/pull/13101

closed 2023-08-13 by MichiMunich