openwrt/openwrt #5078: WARNING: CPU: 0 PID: 0 at backports-5.10.68-1/net/mac80211/rate.c

jbourne27, 2022-02-12:
Almost the same described here: https://bugs.openwrt.org/index.php?do=details&task_id=4248

*** SYSTEM INFO ***
Model: TP-LINK TD-W8970
Architecture: xRX200 rev 1.2
Firmware Version | OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-21.357.58218-b3cd473
Kernel: 5.4.154

*** KERNEL LOG ***
[1795353.955866] ------------[ cut here ]------------
[1795353.959294] WARNING: CPU: 0 PID: 0 at backports-5.10.68-1/net/mac80211/rate.c:677 0x8291ebf0 [mac80211@94271347+0x7dad0]
[1795353.970358] Modules linked in: ath9k ath9k_common iptable_nat ath9k_hw ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppoe nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mac80211 iptable_mangle iptable_filter ipt_REJECT ip6_tables ip_tables cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG x_tables usblp pppox ppp_async nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 crc_ccitt compat drv_dsl_cpe_api uhci_hcd ledtrig_usbport drv_mei_cpe pppoatm ppp_generic slhc br2684 atm drv_ifxos usb_storage fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd dwc2 gpio_button_hotplug ext4 mbcache jbd2 crc32c_generic
[1795354.034988] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W         5.4.154 #0
[1795354.042419] Stack : 00000165 00000166 82980000 80081d1c 00000000 00000000 00000000 00000000
[1795354.050944]         00000000 00000000 00000000 00000000 00000000 00000001 83c0bbb8 a1ce6f73
[1795354.059468]         83c0bc50 00000000 00000000 00000166 00000038 80615d04 2e342e31 35342023
[1795354.067998]         00000166 56000000 00000000 000088ac 00000000 83c0bb98 00000000 8291ebf0
[1795354.076528]         00000009 82977450 829d81a4 80730000 00000001 00000010 00000000 80890000
[1795354.085058]         ...
[1795354.087666] Call Trace:
[1795354.087697] [<82980000>] 0x82980000 [ath9k_hw@0bb7245c+0x555c0]
[1795354.096380] [<80081d1c>] 0x80081d1c
[1795354.100069] [<80615d04>] 0x80615d04
[1795354.103712] [<8291ebf0>] 0x8291ebf0 [mac80211@94271347+0x7dad0]
[1795354.109808] [<82980000>] 0x82980000 [ath9k_hw@0bb7245c+0x555c0]
[1795354.115877] [<8000ef60>] 0x8000ef60
[1795354.119534] [<8000ef68>] 0x8000ef68
[1795354.123189] [<80760000>] 0x80760000
[1795354.126844] [<805fb8fc>] 0x805fb8fc
[1795354.130500] [<8007e7c4>] 0x8007e7c4
[1795354.134158] [<80030b7c>] 0x80030b7c
[1795354.137826] [<8291ebf0>] 0x8291ebf0 [mac80211@94271347+0x7dad0]
[1795354.143907] [<80030c24>] 0x80030c24
[1795354.147585] [<8291ebf0>] 0x8291ebf0 [mac80211@94271347+0x7dad0]
[1795354.153681] [<829e8918>] 0x829e8918 [ath9k@6a74de93+0x18210]
[1795354.159492] [<80097b20>] 0x80097b20
[1795354.163168] [<829e9b8c>] 0x829e9b8c [ath9k@6a74de93+0x18210]
[1795354.169013] [<829ecf4c>] 0x829ecf4c [ath9k@6a74de93+0x18210]
[1795354.174858] [<829e48dc>] 0x829e48dc [ath9k@6a74de93+0x18210]
[1795354.180698] [<829315e4>] 0x829315e4 [mac80211@94271347+0x7dad0]
[1795354.186765] [<82931760>] 0x82931760 [mac80211@94271347+0x7dad0]
[1795354.192822] [<80082f14>] 0x80082f14
[1795354.196507] [<829395c8>] 0x829395c8 [mac80211@94271347+0x7dad0]
[1795354.202634] [<8293b288>] 0x8293b288 [mac80211@94271347+0x7dad0]
[1795354.208697] [<829e3fe0>] 0x829e3fe0 [ath9k@6a74de93+0x18210]
[1795354.214503] [<80034c28>] 0x80034c28
[1795354.218159] [<8061bdbc>] 0x8061bdbc
[1795354.221805] [<80082f14>] 0x80082f14
[1795354.225466] [<800351ac>] 0x800351ac
[1795354.229117] [<803297dc>] 0x803297dc
[1795354.232774] [<8000aa08>] 0x8000aa08
[1795354.236422] 
[1795354.238161] ---[ end trace 0654955756bba11e ]---

ynezz, 2022-02-13:
On 12. 02. 22 12:19, jbourne27 wrote:
> [1795353.959294] WARNING: CPU: 0 PID: 0 at backports-5.10.68-1/net/mac80211/rate.c:677 0x8291ebf0 ***@***.***+0x7dad0]

this translates to the following:

```c
if (rates[i].flags & IEEE80211_TX_RC_VHT_MCS) {
     WARN_ON(ieee80211_rate_get_vht_mcs(&rates[i]) > 9);
     continue;
}
```

which I read as invalid VHT MCS rate index is being used by the other 
end client/AP or it might be some bug in radio firmware. I don't know 
how frequent are these warnings, but if it's something you would like 
removed/fixed, then I would recommend reporting it upstream, which is 
***@***.***

jbourne27, 2022-02-13:
Thanks @ynezz for your reply. I don't know if it's important, but the router is configured as dumb AP, current uptime is 24 days and the event occured a few days ago, 2 "call traces" registered...

rathann, 2022-02-22:
> Almost the same described here: https://bugs.openwrt.org/index.php?do=details&task_id=4248

This is actually reposted here: #9228 by @openwrt-bot . I can see 4 call traces logged on a device with 21 days uptime.

ynezz, 2022-02-26:
We've similar one in other reports:

> WARNING: CPU: 0 PID: 10928 at backports-5.10.68-1/net/mac80211/rate.c:667

```c
if (rates[i].flags & IEEE80211_TX_RC_MCS) {
                        WARN_ON(rates[i].idx > 76); 
```

which I read as invalid HT rate index is being used by the other end client/AP or it might be some bug in radio firmware. I don't know how frequent are these warnings, but if it's something you would like removed/fixed, then I would recommend reporting it upstream, which is `linux-wireless@vger.kernel.org`