openwrt/openwrt #5083: 21.02.1: iw-full conflicts with iw when selected both

nickbash11, 2022-02-12:
```
Collected errors:
 * check_data_file_clashes: Package iw-full wants to install file /home/lede-firmware/openwrt-21.02.1/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/iw
	But that file is already provided by package  * iw
 * opkg_install_cmd: Cannot install package iw-full.
```

It can be resolved by fully uncheck **iw** when you use sources, but there is no the way to fix that when you use _sdk + imagebuilder_, then _imagebuilder_ always wants **iw** first!

mkresin, 2022-02-13:
> but there is no the way to fix that when you use sdk + imagebuilder, then imagebuilder always wants iw first!

Please show the full command line you are using to build an image with the imagebuilder! I'm quite sure it will work similar to:

`make image PROFILE="rpi" PACKAGES="-dnsmasq dnsmasq-full"`

jow-, 2022-02-13:
Works fine here:

```
jow@j7:~$ cd /tmp/
jow@j7:/tmp$ wget https://downloads.openwrt.org/releases/21.02.1/targets/ramips/mt7621/openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64.tar.xz
--2022-02-14 00:16:43--  https://downloads.openwrt.org/releases/21.02.1/targets/ramips/mt7621/openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64.tar.xz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 2a01:4f8:251:321::2, 168.119.138.211
Connecting to downloads.openwrt.org (downloads.openwrt.org)|2a01:4f8:251:321::2|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 98310732 (94M) [application/octet-stream]
Saving to: ‘openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64.tar.xz’

openwrt-imagebuilde 100%[===================>]  93.76M  1.19MB/s    in 83s     

2022-02-14 00:18:06 (1.13 MB/s) - ‘openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64.tar.xz’ saved [98310732/98310732]

jow@j7:/tmp$ tar -xf openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64.tar.xz 
jow@j7:/tmp$ cd openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64/
jow@j7:/tmp/openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64$ make image PACKAGES="iw-full"
Generate local signing keys...
Generate local certificate...
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'perl-data-dumper'... ok.
Checking 'perl-findbin'... ok.
Checking 'perl-file-copy'... ok.
Checking 'perl-file-compare'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'xargs'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'egrep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'python3-distutils'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'rsync'... ok.
Checking 'which'... ok.
Checking 'ldconfig-stub'... ok.
Building images for ramips - ADSLR G7
Packages: iw-full base-files busybox ca-bundle dnsmasq dropbear firewall fstools ip6tables iptables kernel kmod-gpio-button-hotplug kmod-ipt-offload kmod-leds-gpio kmod-mt7615-firmware kmod-mt7615e libc libgcc libustream-wolfssl logd mtd netifd odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe procd uci uclient-fetch urandom-seed urngd wpad-basic-wolfssl

Downloading https://downloads.openwrt.org/releases/21.02.1/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /tmp/openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../tmp/openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/21.02.1/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /tmp/openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../tmp/openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /tmp/openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../tmp/openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
[... 309 lines cut ...]
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 73
Number of ids (unique uids + gids) 1
Number of uids 1
	unknown (0)
Number of gids 1
	unknown (0)
4645+1 records in
4645+1 records out
2378481 bytes (2.4 MB, 2.3 MiB) copied, 0.00693473 s, 343 MB/s
6432+1 records in
6432+1 records out
3293202 bytes (3.3 MB, 3.1 MiB) copied, 0.00711681 s, 463 MB/s
padding image to 00569000
padding image to 0056a000
padding image to 0056c000
padding image to 00570000
padding image to 00580000

Calculating checksums...
jow@j7:/tmp/openwrt-imagebuilder-21.02.1-ramips-mt7621.Linux-x86_64$ 
```

Make sure to place "iw-full" early in your `PACKAGES=...` list, before any package depending on iw.

closed 2022-02-13 by jow-

nickbash11, 2022-02-14:
```
$ make image PROFILE=xiaomi_mi-router-4a-100m PACKAGES="meshsetup mesh-maclist munin-mesh wpad -wpad-basic -kmod-ipt-offload -ppp-mod-pppoe -ppp -firewall -wpad-basic-wolfssl"
Building images for ramips - Xiaomi Mi Router 4A 100M Edition
Packages: meshsetup mesh-maclist munin-mesh wpad base-files busybox ca-bundle dnsmasq dropbear fstools ip6tables iptables kernel kmod-gpio-button-hotplug kmod-leds-gpio kmod-mt7603 kmod-mt76x2 libc libgcc libustream-wolfssl logd mtd netifd odhcp6c odhcpd-ipv6only opkg procd swconfig uci uclient-fetch urandom-seed urngd

Downloading https://downloads.openwrt.org/releases/21.02.1/targets/ramips/mt76x8/packages/Packages.gz
Updated list of available packages in /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/openwrt_telephony
Downloading file:../openwrt-sdk-21.02.1-ramips-mt76x8_gcc-8.4.0_musl.Linux-x86_64/bin/packages/mipsel_24kc/batadv_ospf3_mesh/Packages
Updated list of available packages in /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/batadv_ospf3_mesh

Installing packages...
Installing libc (1.1.24-3) to root...
Installing libgcc1 (8.4.0-3) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/libgcc1_8.4.0-3_mipsel_24kc.ipk.
Configuring libgcc1.
Configuring libc.
Installing kernel (5.4.154-1-81b5fa8a3bdde9109c65df5b7a2d086e) to root...
Configuring kernel.
Installing meshsetup (0.1-1) to root...
Downloading file:../openwrt-sdk-21.02.1-ramips-mt76x8_gcc-8.4.0_musl.Linux-x86_64/bin/packages/mipsel_24kc/batadv_ospf3_mesh/meshsetup_0.1-1_mipsel_24kc.ipk
Installing libpthread (1.1.24-3) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/libpthread_1.1.24-3_mipsel_24kc.ipk.
Installing bird2 (2.0.8-2) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/bird2_2.0.8-2_mipsel_24kc.ipk.
Installing libnl-tiny1 (2020-08-05-c291088f-2) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/libnl-tiny1_2020-08-05-c291088f-2_mipsel_24kc.ipk.
Installing librt (1.1.24-3) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/librt_1.1.24-3_mipsel_24kc.ipk.
Installing batctl-default (2021.1-1) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/batctl-default_2021.1-1_mipsel_24kc.ipk.
Installing alfred (2021.1-1) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/alfred_2021.1-1_mipsel_24kc.ipk.
Installing liblua5.1.5 (5.1.5-9) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/liblua5.1.5_5.1.5-9_mipsel_24kc.ipk.
Installing lua (5.1.5-9) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/lua_5.1.5-9_mipsel_24kc.ipk.
Installing socat (1.7.3.4-2) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/socat_1.7.3.4-2_mipsel_24kc.ipk.
Installing libubox20210516 (2021-05-16-b14c4688-2) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/libubox20210516_2021-05-16-b14c4688-2_mipsel_24kc.ipk.
Installing libuci20130104 (2021-04-14-4b3db117-5) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/libuci20130104_2021-04-14-4b3db117-5_mipsel_24kc.ipk.
Installing libubus20210630 (2021-06-30-4fc532c8-2) to root...
Copying /tmp/openwrt-imagebuilder-21.02.1-ramips-mt76x8.Linux-x86_64/dl/libubus20210630_2021-06-30-4fc532c8-2_mipsel_24kc.ipk.
[... cut ...]
include $(INCLUDE_DIR)/package.mk

define Package/meshsetup
  SECTION=batadv-ospf3-mesh
  CATEGORY:=batadv-ospf3-mesh
  DEPENDS:=+bird2 +batctl-default +alfred +lua +socat +iwinfo +inotifywait +netperf +kmod-ipt-nat +kmod-batman-adv
  TITLE:=Scripts for quick deploy mesh network
endef

define Package/meshsetup/description
  Scripts for quick deploy mesh network
endef

define Build/Compile
endef

define Package/meshsetup/install
  $(CP) ./files/* $(1)/
endef

$(eval $(call BuildPackage,meshsetup))

```

jow-, 2022-02-14:
You forgot to include `iw-full` as first item in the `PACKGES=...` list

nickbash11, 2022-02-15:
> You forgot to include `iw-full` as first item in the `PACKGES=...` list

You are right, I did not know.