openwrt/openwrt #5086: add support for Dlink DIR-816

sakthimaan, 2022-02-13:
[Device Info:](https://www.techeia.com/2018/12/d-link-dir-816-tech-pictures-specs-and.html)

 D-Link DIR-816 A1
Part number: OIR816IEU-A1E
Availability: India
Manuf/OEM/ODM SGE AP699E8C .CW125A
Series: AC750
CPU1: MediaTek MT7620A (580 MHz)
Update: RAM1: 512Mb winbond w9751g6kb-25 8M * 4 Banks* 16Bit DDR2 SDRAM
FLA1: ? MiB (Brand? Model?)
Expansion IFs: none specified
WI1 chip1: MediaTek MT7620A
WI1 802dot11 protocols: bgn
WI1 MIMO config: 2x2:2
WI1 antenna connector: none
WI2 chip1: MediaTek MT7610EN
WI2 802dot11 protocols: an+ac
WI2 MIMO config: 1x1:1
WI2 antenna connector: none
ETH chip1: MediaTek MT7620A
Switch: MediaTek MT7620A
LAN speed: 10/100
LAN ports: 4
WAN speed: 10/100
WAN ports: 1 

[image]

Boot logs are given in the attachment. 

[DIR_816.log](https://github.com/openwrt/openwrt/files/8055410/DIR_816.log)

jospezial, 2022-02-13:
Nearest supported model I could find was
https://openwrt.org/toh/d-link/dir-810l
Compare with you device. Ask for to write a page into openwrt wiki.
Have a look into
https://github.com/openwrt/openwrt/tree/master/target/linux/ramips
Backup your device full flash and single partitions. Dump partitions (mtd) layout.
Try to boot from RAM
https://downloads.openwrt.org/releases/21.02.1/targets/ramips/mt7620/openwrt-21.02.1-ramips-mt7620-dlink_dir-810l-initramfs-kernel.bin
if that works compare mtd layout and flash https://downloads.openwrt.org/releases/21.02.1/targets/ramips/mt7620/openwrt-21.02.1-ramips-mt7620-dlink_dir-810l-squashfs-sysupgrade.bin
Or compile your own fitting openwrt image.
Then you can add a pull request.

pkgadd, 2022-02-13:
This would be a feature request, not a bug - most of all nothing actionable within an issue report. Only someone with the device on their desk (regional availability doesn't help) can work on this, and the only way to add support for a new device would be submitting a pull request against the OpenWrt source.