openwrt/openwrt #5090: Kernel too large for Ubiquiti EdgeRouter X

cyounkins, 2022-02-14:
Thanks for OpenWRT!

OpenWRT 21.02.1 cannot be installed on a stock EdgeRouter X because the factory image is not generated. The factory image is not generated because the kernel has grown past a defined `KERNEL_SIZE`.

Code determining if factory image is built: https://github.com/openwrt/openwrt/blob/v21.02.1/target/linux/ramips/image/mt7621.mk#L62

KERNEL_SIZE definition: https://github.com/openwrt/openwrt/blob/v21.02.1/target/linux/ramips/image/mt7621.mk#L1308

I think the `KERNEL_SIZE` is due to the size of the Kernel1/Kernel2 MTD devices:

```
root@EdgeRouter-X-5-Port:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 10000000 00020000 "ALL"
mtd1: 10000000 00020000 "ALL"
mtd2: 00080000 00020000 "Bootloader"
mtd3: 00060000 00020000 "Config"
mtd4: 00060000 00020000 "eeprom"
mtd5: 00300000 00020000 "Kernel1"
mtd6: 00300000 00020000 "Kernel2"
mtd7: 0f7c0000 00020000 "RootFS"
```

`0x00300000 = 3145728`

Github user stman has a repo with instructions for producing the factory file: https://github.com/stman/OpenWRT-19.07.2-factory-tar-file-for-Ubiquiti-EdgeRouter-x

Unfortunately even with the suggested modifications, the kernel is too big.

* Target size for `openwrt-21.02.1-ramips-mt7621-ubnt_edgerouter-x-initramfs-kernel.bin` is 3145728 bytes
* Current buildbot size is 4942307 bytes
* stman modifications results in 3921794 bytes
* stman + CONFIG_KERNEL_CC_OPTIMIZE_FOR_SIZE results in 3637591 bytes

For the stman + CONFIG_KERNEL_CC_OPTIMIZE_FOR_SIZE configuration:
```
[nix-shell:/fast/craig/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.154]$ size vmlinux
   text	   data	    bss	    dec	    hex	filename
4867540	2548124	 235760	7651424	 74c060	vmlinux

[nix-shell:/fast/craig/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.154]$ size */built-in.a | sort -n -r -k 4 | head -n20
1818658	      0	      0	1818658	 1bc022	usr/initramfs_data.o (ex usr/built-in.a)
  20733	    248	 142620	 163601	  27f11	kernel/printk/printk.o (ex kernel/built-in.a)
  75475	     34	      0	  75509	  126f5	kernel/bpf/verifier.o (ex kernel/built-in.a)
  71474	   1870	    292	  73636	  11fa4	net/core/dev.o (ex net/built-in.a)
  63401	    546	      4	  63951	   f9cf	net/core/filter.o (ex net/built-in.a)
  53096	    216	      0	  53312	   d040	net/core/devlink.o (ex net/built-in.a)
   2988	  49184	     32	  52204	   cbec	kernel/irq/irqdesc.o (ex kernel/built-in.a)
  47820	    566	   1184	  49570	   c1a2	net/ipv6/addrconf.o (ex net/built-in.a)
  39441	   7748	   1567	  48756	   be74	kernel/cgroup/cgroup.o (ex kernel/built-in.a)
  43839	    823	      5	  44667	   ae7b	net/ipv6/route.o (ex net/built-in.a)
  38131	   6289	     26	  44446	   ad9e	mm/memcontrol.o (ex mm/built-in.a)
  43891	     15	      8	  43914	   ab8a	net/ipv4/tcp_input.o (ex net/built-in.a)
  43169	    318	      0	  43487	   a9df	net/core/skbuff.o (ex net/built-in.a)
  36717	     79	    524	  37320	   91c8	net/core/rtnetlink.o (ex net/built-in.a)
  36202	    381	     48	  36631	   8f17	mm/page_alloc.o (ex mm/built-in.a)
  35662	    233	     42	  35937	   8c61	drivers/pci/pci.o (ex drivers/built-in.a)
  34041	    156	      0	  34197	   8595	drivers/mtd/spi-nor/spi-nor.o (ex drivers/built-in.a)
  34156	      3	      0	  34159	   856f	drivers/mtd/nand/raw/nand_base.o (ex drivers/built-in.a)
  33670	    109	      8	  33787	   83fb	kernel/sched/fair.o (ex kernel/built-in.a)
  29183	   3689	    492	  33364	   8254	kernel/sched/core.o (ex kernel/built-in.a)
```

I disabled PPP, IPv6 (I know...), printk, BPF for cgroups, and everything else I could find that didn't seem absolutely essential for bootstrapping and got it to 3138903 bytes. The factory is built as expected but I'm not able to test if it even boots right now. Here's the config: https://gist.github.com/cyounkins/07df94620057d6876b4ee0b0a19a5e52

Questions
1. If a factory image is produced with a stripped-down kernel, installed, and the system is upgraded, will a larger kernel be usable? Are changes to the MTD made to remove the 3MB limit?
2. Are other devices having this same issue?
3. Is it just me or does the output size vary slightly from build to build?

Using the bootloader-based TFTP might work if the 3MB limit is not an issue after base installation: https://help.ui.com/hc/en-us/articles/360019289113 But it worries me that the filenames used in that document end in `.signed`

Hardware wiki: https://openwrt.org/toh/ubiquiti/edgerouter_x_er-x_ka

Commit adding support for ER-X: https://github.com/openwrt/openwrt/commit/4e31b2e8697c21f9bd6493bb3d78a4a41c6a9a52

Old bug manager: https://bugs.openwrt.org/index.php?do=details&task_id=1955

Forum thread: https://forum.openwrt.org/t/ubiquiti-edgerouter-x-unable-to-create-initramfs-factory-tar/39739

namidairo, 2022-02-14:
> stman + CONFIG_KERNEL_CC_OPTIMIZE_FOR_SIZE results in 3637591 bytes

I notice you still have wolfssl and the ca bundle enabled, compared to their config. I believe those are somewhat large? I'm thinking that may be what is stopping you from getting under the metaphorical limbo bar.

lucize, 2022-02-14:
tftp method works, but if you don't want to open it, use 19 branch for the first flash

cyounkins, 2022-02-14:
stman + no CA bundle results in 3807336 bytes which is decent but not nearly enough.

Wolfssl is not so easily disabled in menuconfig as it is added as a device package. It's not clear to me if it may be needed for SSH. menuconfig search:

```
  │ Symbol: DEFAULT_libustream-wolfssl [=y]                                                                                       │
  │ Type  : bool                                                                                                                  │
  │ Defined at tmp/.config-target.in:199893                                                                                       │
  │ Selected by [y]:                                                                                                              │
  │   - TARGET_ramips_mt7621_DEVICE_ubnt_edgerouter-x [=y] && <choice> && TARGET_ramips_mt7621 [=y]
...
  │ Symbol: PACKAGE_libwolfssl [=y]                                                                                               │
  │ Type  : tristate                                                                                                              │
  │ Defined at tmp/.config-package.in:50812                                                                                       │
  │   Prompt: libwolfssl............................................... wolfSSL library                                           │
  │   Location:                                                                                                                   │
  │     -> Libraries                                                                                                              │
  │ (2)   -> SSL                                                                                                                  │
  │ Selects: PACKAGE_librt [=n] && PACKAGE_kmod-crypto-user [=n] && PACKAGE_libc [=y] && PACKAGE_kmod-cryptodev [=n] && PACKAGE_l │
  │ Selected by [y]:                                                                                                              │
  │   - PACKAGE_libustream-wolfssl [=y]
```

The `define Device/ubnt_edgerouter_common` block in `mt7621.mk` has `DEVICE_PACKAGES += -wpad-basic-wolfssl`.

I'm unable to find any factory builds on `downloads.openwrt.org` so I can't use "official" v19 for the first flash.
```
$ for v in 17.01.0 17.01.0-rc1 17.01.0-rc2 17.01.1 17.01.2 17.01.3 17.01.4 17.01.5 17.01.6 17.01.7 18.06.0 18.06.0-rc1 18.06.0-rc2 18.06.1 18.06.2 18.06.3 18.06.4 18.06.5 18.06.6 18.06.7 18.06.8 18.06.9 19.07.0 19.07.0-rc1 19.07.0-rc2 19.07.1 19.07.2 19.07.3 19.07.4 19.07.5 19.07.6 19.07.7 19.07.8 21.02.0 21.02.0-rc1 21.02.0-rc2 21.02.0-rc3 21.02.0-rc4 21.02.1; curl -o "$v.html" "https://downloads.openwrt.org/releases/$v/targets/ramips/mt7621/"; end
...
$ grep 'ubnt' *
... results ...
$ grep 'ubnt' * | grep 'factory'
<nothing>
```

I know others have built factory tars that they claim work, but telling folks to install a 3rd party binary isn't a great user experience.

cyounkins, 2022-02-14:
Unfortunately the TFTP recovery mode that is baked into the Ubiquiti bootloader that operates without a serial cable (https://help.ui.com/hc/en-us/articles/360019289113) did not work for me. 

Before attempting to flash OpenWRT, the system had been upgraded to v2.0.9-hotfix.1. This means it has the recent bootloader which includes the TFTP recovery mode.

```
$ shasum -a 256 openwrt-21.02.1-ramips-mt7621-ubnt_edgerouter-x-initramfs-kernel.bin
619b40ed12106e608efea79c4cbe1aafd048e73be542dacd713c8286d3c989a5  openwrt-21.02.1-ramips-mt7621-ubnt_edgerouter-x-initramfs-kernel.bin

$ tftp
tftp> connect 192.168.1.20
tftp> binary
tftp> put openwrt-21.02.1-ramips-mt7621-ubnt_edgerouter-x-initramfs-kernel.bin
Error code 512: Firmware check failed
Sent 4941824 bytes in 5.0 seconds
tftp> put openwrt-21.02.1-ramips-mt7621-ubnt_edgerouter-x-initramfs-kernel.bin
Error code 512: Firmware check failed
Sent 4941824 bytes in 4.6 seconds
tftp> put ER-e50.recovery.v2.0.6.5208541.190708.0508.16de5fdde.img.signed
Sent 107022404 bytes in 109.1 seconds
tftp>
```

Some differences in the files:
```
$ file ER-e50.recovery.v2.0.6.5208541.190708.0508.16de5fdde.img.signed
ER-e50.recovery.v2.0.6.5208541.190708.0508.16de5fdde.img.signed: HIT archive data

$ file openwrt-21.02.1-ramips-mt7621-ubnt_edgerouter-x-initramfs-kernel.bin
openwrt-21.02.1-ramips-mt7621-ubnt_edgerouter-x-initramfs-kernel.bin: u-boot legacy uImage, MIPS OpenWrt Linux-5.4.154, Linux/MIPS, OS Kernel Image (Not compressed), 4942243 bytes, Sun Oct 24 09:01:35 2021, Load Address: 0x80001000, Entry Point: 0x80001000, Header CRC: 0xB02C71FE, Data CRC: 0xE20649DC
```

And differences in the v2.0.6 regular download and the 'recovery' version:
```
$ ls -al ER-e50.recovery.v2.0.6.5208541.190708.0508.16de5fdde.img.signed ER-e50.v2.0.6.5208541.tar
-rw-r--r--@ 1 craig  staff  107022404 Feb 13 13:39 ER-e50.recovery.v2.0.6.5208541.190708.0508.16de5fdde.img.signed
-rw-r--r--@ 1 craig  staff   83783680 Feb 14 15:58 ER-e50.v2.0.6.5208541.tar

$ file ER-e50.recovery.v2.0.6.5208541.190708.0508.16de5fdde.img.signed ER-e50.v2.0.6.5208541.tar
ER-e50.recovery.v2.0.6.5208541.190708.0508.16de5fdde.img.signed: HIT archive data
ER-e50.v2.0.6.5208541.tar:                                       POSIX tar archive (GNU)

$ hexdump -C -n128 ER-e50.recovery.v2.0.6.5208541.190708.0508.16de5fdde.img.signed
00000000  55 42 4e 54 2e 45 44 47  45 4f 53 00 00 00 00 00  |UBNT.EDGEOS.....|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000020  00 00 00 01 00 00 00 00  06 61 08 44 1c a9 de f3  |.........a.D....|
00000030  00 00 00 04 00 00 0e 50  00 00 0e 51 00 00 0e 52  |.......P...Q...R|
00000040  00 00 0e 55 27 05 19 56  9f fa f9 44 5d 35 81 ad  |...U'..V...D]5..|
00000050  06 61 07 c0 81 00 10 00  81 38 41 c0 6c 27 28 dd  |.a.......8A.l'(.|
00000060  05 05 02 00 4c 69 6e 75  78 20 4b 65 72 6e 65 6c  |....Linux Kernel|
00000070  20 49 6d 61 67 65 00 00  00 00 00 00 00 00 00 00  | Image..........|
```

Is the `initramfs-kernel.bin` file not in the format the TFTP daemon is expecting? Or is there a signature being checked?

cyounkins, 2022-02-16:
I can do it manually, so of course it's possible.

Because I'm not well-versed in `make`, a top-level build process that solves this is:
1. Checkout the tree
2. Patch the kernel config for super tiny kernel
3. make
4. Unpatch the kernel config
5. make

The second `make` will replace most outputs but since the `-factory.tar` file will not be built with the larger kernel it will remain. This is almost certainly not the best way of accomplishing this, but it is one way. 

The project already builds many versions of the kernel for different architectures. My reading of `mt7621.mk` is that `DEVICE_PACKAGES` adds packages to the config per-device. Can we write a new device `make` target that removes packages / config flags, builds a separate kernel along with the factory image? Then we'd link users to the `-factory.tar` image from this new target and `-sysupgrade.bin` from the current target.

cyounkins, 2022-02-25:
So I have stock OpenWRT working on the EdgeRouter X, and as I previously noted the MTD sizes for "kernel1" and "kernel2" remain at 3MB. But the stock kernel is ~4.8MB. `head` shows that `/dev/mtd3ro` and `/dev/mtd4ro` have the start of OpenWRT kernels. The bootloader is Ralink u-boot.

Can someone clue me in as to how this is working? I can only think of a few possibilities:

1. The kernels are complete because they over-run the 3MB allotment, potentially destroying data
2. The MTD kernels are incomplete but enough to load a complete kernel from somewhere else (squashfs?)
3. The kernels are incomplete and somehow still work

cyounkins, 2022-03-19:
I built factory files for the Edgerouters based on 21.02.2. You can find them and the configuration options used here: https://github.com/cyounkins/edgerouter-factory

I've run out of steam getting all this to work. I pass the torch to someone else to:
1. Determine if the >3MB kernels are problematic (last comment above)
2. Update the wiki with links to my factory files
3. Update the wiki with info from earlier in this report - the bootloader TFTP mode cannot be used, etc

ynezz, 2022-05-23:
As stated above, there is one kernel for each (sub)target and AFAIK there is currently no plan to support and produce device specific kernels. 

> I built factory files for the Edgerouters based on 21.02.2.

Indeed, such community builds are probably the only way currently to support such use cases. In order to reach broader use base, feel free to create a post on [Community Builds, Projects & Packages](https://forum.openwrt.org/c/community-builds-projects-packages) and add this information/link to the device's wiki page.

closed 2022-05-23 by ynezz

darkxst, 2024-04-17:
> But the stock kernel is ~4.8MB.

The kernel in 23.05.3 is actually only about ~2.7MB. Factory images are larger since they also include rootfs/initramfs. Looks like linux 6.1 does breach the 3MB limit though. I am looking into this over at:
https://forum.openwrt.org/t/resizing-kernel-images-for-edgerouter-x/194968