openwrt/openwrt #5091: RUTX10 nand startup problem

simonastamosaitis, 2022-02-14:
Hi!
I found issue with RUTX10. Looks like that 5.10 kernel can't properly read NAND ID.
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.100 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 11.2.0 r18802-5205010a54) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Sun Feb 13 23:31:27 2022
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Teltonika RUTX10
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x000000008fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000087dfffff]
[    0.000000]   node   0: [mem 0x0000000087e00000-0x0000000087ffffff]
[    0.000000]   node   0: [mem 0x0000000088000000-0x000000008fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000008fffffff]
[    0.000000] percpu: Embedded 15 pages/cpu s30732 r8192 d22516 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: ubi.mtd=rootfs ubi.block=0,1 root=/dev/ubiblock0_1 rootfstype=squashfs console=ttyMSM0,115200n8 rootwait
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 246148K/262144K available (6430K kernel code, 605K rwdata, 1604K rodata, 1024K init, 245K bss, 15996K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x360/0x50c with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000008] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000025] Switching to timer-based delay loop, resolution 20ns
[    0.000262] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000286] pid_max: default: 32768 minimum: 301
[    0.000467] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000486] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001517] CPU: Testing write buffer coherency: ok
[    0.001836] qcom_scm: convention: smc legacy
[    0.002572] Setting up static identity map for 0x80300000 - 0x8030003c
[    0.002704] rcu: Hierarchical SRCU implementation.
[    0.002909] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.003169] smp: Bringing up secondary CPUs ...
[    0.006285] smp: Brought up 1 node, 4 CPUs
[    0.006310] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.006320] CPU: All CPU(s) started in SVC mode.
[    0.011557] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.011720] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.011749] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.011997] pinctrl core: initialized pinctrl subsystem
[    0.013188] NET: Registered protocol family 16
[    0.013585] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.014714] thermal_sys: Registered thermal governor 'step_wise'
[    0.015121] cpuidle: using governor ladder
[    0.015230] cpuidle: using governor menu
[    0.041177] cryptd: max_cpu_qlen set to 1000
[    0.044792] usbcore: registered new interface driver usbfs
[    0.044874] usbcore: registered new interface driver hub
[    0.044935] usbcore: registered new device driver usb
[    0.044987] pps_core: LinuxPPS API ver. 1 registered
[    0.044999] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.045028] PTP clock support registered
[    0.046990] clocksource: Switched to clocksource arch_sys_counter
[    0.047954] NET: Registered protocol family 2
[    0.048113] IP idents hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.048963] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.049021] TCP established hash table entries: 2048 (order: 1, 8192 bytes, linear)
[... 44 lines cut ...]
[    0.873559] spi-nand spi0.1: unknown raw ID ffffffff
[    0.873617] spi-nand: probe of spi0.1 failed with error -524
[    0.911716] ESS reset ok!
[    0.936010] ESS reset ok!
[    1.344829] PHY3 test see issue!
[    1.348933] PHY4 test see issue!
[    1.373017] ESS reset ok!
[    1.871843] i2c /dev entries driver
[    1.874947] sdhci: Secure Digital Host Controller Interface driver
[    1.874985] sdhci: Copyright(c) Pierre Ossman
[    1.880484] sdhci-pltfm: SDHCI platform and OF driver helper
[    1.888791] NET: Registered protocol family 10
[    1.891965] Segment Routing with IPv6
[    1.894896] NET: Registered protocol family 17
[    1.898647] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[    1.902891] 8021q: 802.1Q VLAN Support v1.8
[    1.916131] Registering SWP/SWPB emulation handler
�[    1.936476] Waiting for root device /dev/ubiblock0_1...

```
`[    0.873559] spi-nand spi0.1: unknown raw ID ffffffff`  
`[    0.873617] spi-nand: probe of spi0.1 failed with error -524`  
`Linux version 5.10.100 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 11.2.0 r18802-5205010a54) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Sun Feb 13 23:31:27 2022`
I tried to change chip select values, but with no success.  
5.4 kernel boots fine.

0xFelix, 2022-02-14:
Maybe this patch [1] is faulty?

From what I remember it worked fine with 5.10 but unfortunately I do not have a device for testing currently.

[1]: https://github.com/openwrt/openwrt/blob/master/target/linux/generic/pending-5.10/483-mtd-spinand-add-support-for-xtx-xt26g0xa.patch

simonastamosaitis, 2022-02-15:
Looks like I found issue. I changed `SPINAND_READID_METHOD_OPCODE_DUMMY` to `SPINAND_READID_METHOD_OPCODE_ADDR` in your patch and now router boots fine :)
```
[    0.871021] spi-nand spi0.1: XTX SPI NAND was found.
[    0.871067] spi-nand spi0.1: 256 MiB, block size: 128 KiB, page size: 2048, OOB size: 64
[    0.875267] 2 fixed-partitions partitions found on MTD device spi0.1
[    0.883181] Creating 2 MTD partitions on "spi0.1":
[    0.889492] 0x000000000000-0x000008000000 : "rootfs_1"
[    0.899550] random: fast init done
[    1.226106] random: crng init done
[    1.513714] 0x000008000000-0x000010000000 : "rootfs"
[    2.226724] mtd: device 17 (rootfs) set to be root filesystem
[    2.228226] mtdsplit: no squashfs found in "rootfs"
```

jow-, 2022-02-20:
@0xFelix - will you follow up with a fix for this?

0xFelix, 2022-02-21:
@jow- Yes, I want to. Unfortunately I did not have time yet.

0xFelix, 2022-04-12:
@simonastamosaitis Can you check again if it works now?

simonastamosaitis, 2022-04-13:
> @simonastamosaitis Can you check again if it works now?
Yes, it's fine now. Thanks.

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.110 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 11.2.0 r19427-5d5afd5177) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Tue Apr 12 07:33:22 2022
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Teltonika RUTX10
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x000000008fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000087dfffff]
[    0.000000]   node   0: [mem 0x0000000087e00000-0x0000000087ffffff]
[    0.000000]   node   0: [mem 0x0000000088000000-0x000000008fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000008fffffff]
[    0.000000] percpu: Embedded 15 pages/cpu s30732 r8192 d22516 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: ubi.mtd=rootfs ubi.block=0,1 root=/dev/ubiblock0_1 rootfstype=squashfs console=ttyMSM0,115200n8 rootwait
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 246148K/262144K available (6446K kernel code, 606K rwdata, 1604K rodata, 1024K init, 244K bss, 15996K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x360/0x50c with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000009] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000025] Switching to timer-based delay loop, resolution 20ns
[    0.000261] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000286] pid_max: default: 32768 minimum: 301
[    0.000467] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000488] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001516] CPU: Testing write buffer coherency: ok
[    0.001845] qcom_scm: convention: smc legacy
[    0.002586] Setting up static identity map for 0x80300000 - 0x8030003c
[    0.002719] rcu: Hierarchical SRCU implementation.
[    0.002922] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.003175] smp: Bringing up secondary CPUs ...
[    0.006299] smp: Brought up 1 node, 4 CPUs
[    0.006325] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.006335] CPU: All CPU(s) started in SVC mode.
[    0.011651] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.011822] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.011849] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.012100] pinctrl core: initialized pinctrl subsystem
[    0.013279] NET: Registered protocol family 16
[    0.013677] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.014803] thermal_sys: Registered thermal governor 'step_wise'
[    0.015249] cpuidle: using governor ladder
[    0.015309] cpuidle: using governor menu
[    0.041435] cryptd: max_cpu_qlen set to 1000
[    0.045084] usbcore: registered new interface driver usbfs
[    0.045157] usbcore: registered new interface driver hub
[    0.045219] usbcore: registered new device driver usb
[    0.045272] pps_core: LinuxPPS API ver. 1 registered
[    0.045285] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.045313] PTP clock support registered
[    0.047269] clocksource: Switched to clocksource arch_sys_counter
[    0.048240] NET: Registered protocol family 2
[    0.048396] IP idents hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.049248] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[... 185 lines cut ...]
[   20.137356] PPP generic driver version 2.4.2
[   20.138713] NET: Registered protocol family 24
[   22.393815] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   22.393872] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   22.405758] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi [... cut ...]
[   23.819645] kmodloader: done loading kernel modules from /etc/modules.d/*
[   48.716115] br-lan: port 1(eth0) entered blocking state
[   48.716182] br-lan: port 1(eth0) entered disabled state
[   48.720696] device eth0 entered promiscuous mode
[   48.829210] ess_edma c080000.edma eth1: Link is Down
[   49.767389] br-lan: port 1(eth0) entered blocking state
[   49.767465] br-lan: port 1(eth0) entered forwarding state
[   49.772255] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
```

closed 2022-04-13 by simonastamosaitis