openwrt/openwrt #5094: high latency / no packet transmitted on WRT1900ACS (88W8864)

zorxd, 2022-02-14:
Hi

On Linksys WRT1900ACS (and possibly other devices using marvel 88W8864 WiFi), WiFi connectivity is intermittently unreliable. On my Galaxy S10e smartphone, the symptom is that the device reports being connected to WiFi, but when trying to load a web page, it takes forever. The device switches to cellular connectivity after some time, and reconnect to WiFi (possibly on the other band) and then it works for a while.

As a couple users have reported, it is necessary to disable TX AMSDU to fix this bug.
This can be accomplished by issuing the following commands:

echo 0 > /sys/kernel/debug/ieee80211/phy1/mwlwifi/tx_amsdu
echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu

This should be done by default.

Related discussions:
https://forum.openwrt.org/t/openwrt-21-02-0-first-stable-release/105395/209 https://github.com/kaloz/mwlwifi/issues/207
https://github.com/kaloz/mwlwifi/issues/207
https://bugs.openwrt.org/index.php?do=details&task_id=4064

olimpiumarius, 2022-02-14:
take a look here can help you with something

https://forum.openwrt.org/t/wrt1900acs-v2-wifi-problems/120332

zorxd, 2022-02-15:
well the AMSDU fix works well for me. I just think it should be enabled by default in OpenWRT so that others do not experience the same bug as I did.
I don't think your forum link has anything to do with this issue however.

olimpiumarius, 2022-02-15:
I also tried the above settings, it seems to be causing me problems, 
How can I disable them ?

[image]

zorxd, 2022-02-15:
Not sure what your problem is, but to revert just reboot your rooter. These settings are not permanent (unless added to a boot script).
Or you can issue a "echo 1" instead of "echo 0" in the above commands.

zorxd, 2022-02-22:
No, I wasn't aware, thanks. Is it supposed to fix this problem?

mmortal03, 2022-02-22:
> No, I wasn't aware, thanks. Is it supposed to fix this problem?

You'll have to test it and report back. A major bug that was affecting some WRT series routers *was* fixed in 21.02.2, but I don't know if this AMSDU issue is still present or not. 

For reference, if it wasn't fixed, people have claimed your aforementioned commands can be included in /etc/rc.local:

echo 0 > /sys/kernel/debug/ieee80211/phy1/mwlwifi/tx_amsdu
echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu

zorxd, 2022-03-09:
so far so good after a couple days, but I'll continue testing.

ThunderMikey, 2022-09-28:
I am having similar problems with my WRT3200ACM running OpenWRT 22.03. The connected Google Home Max cannot do anything (it just says "there's some problems with your Wi-Fi") most of the time. However, the Nest Home Mini is immune to this issue.

Another thing I observed was some apps in my phone took a long time to load, whereas they loaded fine on 4G and on another router Wi-Fi network.

closed 2023-06-22 by zorxd