openwrt/openwrt #5095: Ethernet not working on Ubiquity UniFi AP

Immanuel84, 2022-02-14:
Hi,
i just tried to update my Ubiquity Unifi AP to 21.02.1 from older version (19.07.5) but i'm having issues
with ethernet port: it just doesn't work.
I'm able to connect using serial so i tried to:
- reset configuration (using firstboot)
- re-flash again (wifi works so i was able to scp the firmware)
- boot in failsafe and reset again
- re-flash via TFTP using openwrt factory .bin

No matter what i do, ethernet just doesn't go in up state.
I also tried to stop the br-lan (via /etc/init.d/network stop) to check if the bridge was the problem,
i was able to set a fixed ip address on eth0 but it doesn't go up.
I'm reverting to older OpenWrt version.
Does anyone else have a similar issue?
I'm attacching the serial output log.
I just noticed that MTD partitions are different.

OpenWrt 19.07.5
```shell
[    0.630160] 5 cmdlinepart partitions found on MTD device spi0.0
[    0.636158] Creating 5 MTD partitions on "spi0.0":
[    0.640978] 0x000000000000-0x000000040000 : "u-boot"
[    0.647939] 0x000000040000-0x000000050000 : "u-boot-env"
[    0.656168] 0x000000050000-0x0000007b0000 : "firmware"
[    0.672597] 2 uimage-fw partitions found on MTD device firmware
[    0.678595] 0x000000050000-0x0000001e0000 : "kernel"
[    0.684641] 0x0000001e0000-0x0000007b0000 : "rootfs"
[    0.690629] mtd: device 4 (rootfs) set to be root filesystem
[    0.696422] 1 squashfs-split partitions found on MTD device rootfs
[    0.702632] 0x000000720000-0x0000007b0000 : "rootfs_data"
[    0.712318] 0x0000007b0000-0x0000007f0000 : "cfg"
[    0.719032] 0x0000007f0000-0x000000800000 : "EEPROM"
```

OpenWrt 21.02.1
```shell
[    0.374608] 6 fixed-partitions partitions found on MTD device spi0.0
[    0.380996] Creating 6 MTD partitions on "spi0.0":
[    0.385831] 0x000000000000-0x000000040000 : "u-boot"
[    0.392336] 0x000000040000-0x000000050000 : "u-boot-env"
[    0.399280] 0x000000050000-0x0000007a0000 : "firmware"
[    0.411042] 2 uimage-fw partitions found on MTD device firmware
[    0.417033] Creating 2 MTD partitions on "firmware":
[    0.422040] 0x000000000000-0x000000200000 : "kernel"
[    0.428407] 0x000000200000-0x000000750000 : "rootfs"
[    0.434899] mtd: device 4 (rootfs) set to be root filesystem
[    0.443437] 1 squashfs-split partitions found on MTD device rootfs
[    0.449739] 0x000000520000-0x000000750000 : "rootfs_data"
[    0.456619] 0x0000007a0000-0x0000007b0000 : "board_config"
[    0.463720] 0x0000007b0000-0x0000007f0000 : "cfg"
[    0.469946] 0x0000007f0000-0x000000800000 : "art"
```

OpenWrt 21.02.1 Boot Log
````shell
U-Boot 1.1.4.2-s446 (Sep  6 2010 - 06:00:24)                                                        
                                                                                                    
Board: Ubiquiti Networks XM board (rev 0.2 e502)                                                    
DRAM:  64 MB                                                                                        
Flash:  8 MB                                                                                        
Net:   eth0, eth1                                                                                   
Hit any key to stop autoboot:  0                                                                    
## Booting image at 9f050000 ...                                                                    
   Image Name:   MIPS OpenWrt Linux-5.4.154                                                         
   Created:      2021-10-24   9:01:35 UTC                                                           
   Image Type:   MIPS Linux Kernel Image (lzma compressed)                                          
   Data Size:    2047384 Bytes =  2 MB                                                              
   Load Address: 80060000                                                                           
   Entry Point:  80060000                                                                           
   Verifying Checksum ... OK                                                                        
[... 173 lines cut ...]
 OpenWrt 21.02.1, r16325-88151b8303
 -----------------------------------------------------
root@OpenWrt:/# p[   46.816845] br-lan: port 2(wlan0) entered blocking state

root@OpenWrt:/# cat /proc/cpuinfo 
system type             : Atheros AR7240 rev 2
machine                 : Ubiquiti UniFi
processor               : 0
cpu model               : MIPS 24Kc V7.4
BogoMIPS                : 259.58
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 16
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16
Options implemented     : tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit perf
shadow register sets    : 1
kscratch registers      : 0
package                 : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available
````

RussellSenior, 2022-03-03:
Try running: swconfig dev switch0 show | grep port

Immanuel84, 2022-03-05:
On OpenWrt 19.07.5:
```
root@OpenWrt:~# swconfig dev switch0 | grep sort
root@OpenWrt:~# swconfig dev switch0
swconfig list
swconfig dev <dev> [port <port>|vlan <vlan>] (help|set <key> <value>|get <key>|load <config>|show)
root@OpenWrt:~# swconfig list
```
As soon as possibile i will re-flash 21.02 and paste the output.

RussellSenior, 2022-03-05:
Can you paste /etc/config/network on 19.07.5 and 21.02.x?

Immanuel84, 2022-03-05:
Yep, on OpenWrt 19.07.5:
```
root@OpenWrt:~# cat /etc/config/network 

config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd41:93e6:89ad::/48'

config interface 'lan'
	option type 'bridge'
	option ifname 'eth0'
	option proto 'static'
	option ipaddr '192.168.23.2'
	option netmask '255.255.255.0'
	option gateway '192.168.23.4'
	option ip6assign '60'
	list dns '8.8.8.8'
	option ip6addr '2001:xxxxxxxxx'
	option ip6gw '2001:xxxxxxxxx'
```
It's pretty default, except for static IPv4 and IPv6.
I didn't copy 21.02.x config, but it was the default one, i didn't keep the configuration after flashing on purpose.

RussellSenior, 2022-03-05:
I'm curious what the 21.02.x config looks like because of the different switch configuration between it and 19.07. I don't have your device, and the fact that it's not working implies something is wrong. Seeing it might help guess about what that is.

Immanuel84, 2022-03-05:
I will retry tomorrow, this time i'll copy /etc/ folder.

Immanuel84, 2022-03-06:
Here is the /etc/config/network from OpenWrt 21.02:
```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdf1:dad8:64b6::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
```
I alson noticed the following lines from boot, but they appear only on first boot, probably the system configuration is still in progress:

```
[   15.688213] ath9k 0000:00:00.0: Direct firmware load for ath9k-eeprom-pci-0000:00:00.0.bin failed with error -2
[   15.698419] ath9k 0000:00:00.0: Falling back to sysfs fallback for: ath9k-eeprom-pci-0000:00:00.0.bin
```

`swconfig dev switch0 show | grep port` it's the same, i also noticed that the listed ports are *always* 10baseT half duplex, no matter if the computer is set to Automatic or 100baseT.

RussellSenior, 2022-03-07:
I think I spotted your problem.

`[    0.000000] SoC: Atheros AR7240 rev 2`

What image file did you flash?

RussellSenior, 2022-03-07:
This looks like maybe a problem with two hardware variants. The current ath79 assuming ar7241 in the device tree, and you have an ar7240. The drivers in ar71xx could accommodate either, but ath79 needs to be specific. See the two ubiquiti bullet variants as a way it was worked around previously.

Immanuel84, 2022-03-07:
I flashed "openwrt-21.02.2-ath79-generic-ubnt_unifi-squashfs-factory.bin"; yes, older (working) version was ar71xx, exactly this image "openwrt-ar71xx-generic-ubnt-unifi-squashfs-factory.bin".
Maybe i understand what you mean: i should look at .dts for differences between "ar7240_ubnt_bullet-m-ar7240.dts" and "ar7241_ubnt_bullet-m-ar7241.dts"?
Well, they are different, in ar2740 there is this part that is missing in ar2741:
```
&eth0 {
        fixed-link {
                speed = <100>;
                full-duplex;
        };
};  
```
Weird thing is that according to [wiki](https://openwrt.org/toh/ubiquiti/unifi) there are no AR7241 version, only AR7240, maybe there is a mistake in the new .dts.

Immanuel84, 2022-03-08:
Ok, maybe i fixed, in file **target/linux/ath79/dts/ar7241_ubnt_unifi.dtsi** just change `#include "ar7241.dtsi"` to `#include "ar7240.dtsi"`.
Actual output of `swconfig dev switch0 show`:
```
root@OpenWrt:/# swconfig dev switch0 show
Global attributes:
        enable_vlan: 0
        ar8xxx_mib_poll_interval: 500
        ar8xxx_mib_type: 0
        enable_mirror_rx: 0
        enable_mirror_tx: 0
        mirror_monitor_port: 0
        mirror_source_port: 0
        arl_table: address resolution table

Port 0:
        mib: No MIB data
        pvid: 0
        link: port:0 link:up speed:1000baseT full-duplex txflow rxflow 
Port 1:
        mib: No MIB data
        pvid: 0
        link: port:1 link:down
Port 2:
        mib: No MIB data
        pvid: 0
        link: port:2 link:down
Port 3:
        mib: No MIB data
        pvid: 0
        link: port:3 link:down
Port 4:
        mib: No MIB data
        pvid: 0
        link: port:4 link:down
```

Ping works, speed is maybe incorrect but if i remember correctly it was always incorrect (actually eth is 100Mbps).

Console log:
```
U-Boot 1.1.4.2-s446 (Sep  6 2010 - 06:00:24)

Board: Ubiquiti Networks XM board (rev 0.2 e502)
DRAM:  64 MB
Flash:  8 MB
Net:   eth0, eth1
Hit any key to stop autoboot:  0 
## Booting image at 9f050000 ...
   Image Name:   MIPS OpenWrt Linux-5.4.154
   Created:      2021-10-24   9:01:35 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2055908 Bytes =  2 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.4.154 (emanuele@debian) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16325-88151b8303)) #0 Sun Oct 24 09:011
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is Ubiquiti UniFi
[    0.000000] SoC: Atheros AR7240 rev 2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[... 126 lines cut ...]
[   14.786642] Loading modules backported from Linux version v5.10.68-0-g4d8524048a35
[   14.794337] Backport generated by backports.git v5.10.68-1-0-ga4f9ba32
[   14.880867] xt_time: kernel timezone is -0000
[   15.093373] PPP generic driver version 2.4.2
[   15.110958] NET: Registered protocol family 24
[   15.255914] ath9k 0000:00:00.0: Direct firmware load for ath9k-eeprom-pci-0000:00:00.0.bin failed with error -2
[   15.266136] ath9k 0000:00:00.0: Falling back to sysfs fallback for: ath9k-eeprom-pci-0000:00:00.0.bin
[   15.401078] urngd: v1.0.2 started.
[   15.947674] random: crng init done
[   16.184733] ath: phy0: Ignoring endianness difference in EEPROM magic bytes.
[   16.213155] ieee80211 phy0: Atheros AR9287 Rev:2 mem=0xb0000000, irq=13
[   16.303693] kmodloader: done loading kernel modules from /etc/modules.d/*
[   81.135653] jffs2_scan_eraseblock(): End of filesystem marker found at 0x0
[   81.157573] jffs2_build_filesystem(): unlocking the mtd device... 
[   81.157583] done.
[   81.165754] jffs2_build_filesystem(): erasing all blocks after the end marker... 
[   87.673928] br-lan: port 1(eth0) entered blocking state
[   87.686772] br-lan: port 1(eth0) entered disabled state
[   87.692482] device eth0 entered promiscuous mode
[   89.735212] eth0: link up (10Mbps/Half duplex)
[   89.750682] br-lan: port 1(eth0) entered blocking state
[   89.755970] br-lan: port 1(eth0) entered forwarding state
[   89.784579] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
```
I'll test it just to be sure tha everythin works.