openwrt/openwrt #9246: Constant null pointer stacktrace (mac80211.h:2721) with following Oops & reboot (kworker/u8) on RasPi 3B+

gmanic, 2022-02-15:
Since updating my Raspberry Pi 3 (B+ Rev 1.3) to 21.02.1, r16325-88151b8303 my openwrt router keeps rebooting after crashing with NULL pointer dereference error.
Aside from "factory install" (sysupgrade, which flashed SD) and growing overlayfs to SD-card size, not much additionally on it: openvpn, travelmate, mqtt (currently not in use) plus a couple (unused) bluetooth and pyhton3 packages (no further programs running).

It throws the following error, randomly between 5 mins uptime to approx 3-4h uptime. The device is remote, has another WiFi/GSM as uplink and one more Raspi as client (providing net connectivity to it). The whole setup is connected (vpn) to my network.

    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.679243] ------------[ cut here ]------------
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.685992] WARNING: CPU: 3 PID: 7 at backports-5.10.68-1/include/net/mac80211.h:2721 0xbf336898 [rt2x00lib@9c498de5+0xa000]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.701406] Modules linked in: rt2800usb rt2800lib pppoe ppp_async iptable_nat brcmfmac xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT rt2x00usb rt2x00lib pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbhid slhc rfcomm nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ipt_ECN ip_tables hidp hid_generic hci_uart crc_ccitt compat btusb btintel brcmutil bnep bluetooth snd_bcm2835(C) hid evdev nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore nls_utf8 ecdh_generic ecc kpp cmac vfat fat nls_iso8859_1 nls_cp437 uhci_hcd ohci_platform
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.701623]  ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd 
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.816055] CPU: 3 PID: 7 Comm: kworker/u8:0 Tainted: G         C        5.4.154 #0
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.826167] Hardware name: BCM2835
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.830796] Workqueue: phy1 0xbf33e610 [rt2x00usb@5a3e45ed+0x4000]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.838252] Function entered at [<c020fda4>] from [<c020b280>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.845344] Function entered at [<c020b280>] from [<c0832f98>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.852389] Function entered at [<c0832f98>] from [<c021e72c>] 
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.859391] Function entered at [<c021e72c>] from [<c021e798>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.866346] Function entered at [<c021e798>] from [<bf336898>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.873275] Function entered at [<bf336898>] from [<bf337560>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.880133] Function entered at [<bf337560>] from [<bf335010>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.886945] Function entered at [<bf335010>] from [<bf2aa404>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.893712] Function entered at [<bf2aa404>] from [<bf2aa52c>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.900428] Function entered at [<bf2aa52c>] from [<bf2b0ff0>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.907088] Function entered at [<bf2b0ff0>] from [<bf2b280c>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.913708] Function entered at [<bf2b280c>] from [<c0222fcc>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.920276] Function entered at [<c0222fcc>] from [<c02028d8>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.926801] Function entered at [<c02028d8>] from [<c0223324>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.933278] Function entered at [<c0223324>] from [<c02233ec>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.939704] Function entered at [<c02233ec>] from [<bf333900>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.946093] Function entered at [<bf333900>] from [<bf33e660>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.952451] Function entered at [<bf33e660>] from [<c0236860>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.958772] Function entered at [<c0236860>] from [<c0236afc>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.965073] Function entered at [<c0236afc>] from [<c023c61c>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.971339] Function entered at [<c023c61c>] from [<c02010d8>]
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.977578] Exception stack(0xef099fb0 to 0xef099ff8)
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.983044] 9fa0:                                     00000000 00000000 00000000 00000000
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.992109] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  622.001341] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  622.008613] ---[ end trace 4db318219d192a09 ]---

Followed by 

    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  622.013876] 8<--- cut here ---
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  622.017550] Unable to handle kernel NULL pointer dereference at virtual address 00000006
    Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  622.027062] pgd = 7f4e30b3
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.030489] [00000006] *pgd=80000000004003, *pmd=00000000
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.036683] Internal error: Oops: 206 [#1] SMP ARM
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.042270] Modules linked in: rt2800usb rt2800lib pppoe ppp_async iptable_nat brcmfmac xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT rt2x00usb rt2x00lib pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbhid slhc rfcomm nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ipt_ECN ip_tables hidp hid_generic hci_uart crc_ccitt compat btusb btintel brcmutil bnep bluetooth snd_bcm2835(C) hid evdev nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore nls_utf8 ecdh_generic ecc kpp cmac vfat fat nls_iso8859_1 nls_cp437 uhci_hcd ohci_platform
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.042366]  ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.151707] CPU: 3 PID: 7 Comm: kworker/u8:0 Tainted: G        WC        5.4.154 #0
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.161747] Hardware name: BCM2835
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.166304] Workqueue: phy1 0xbf33e610 [rt2x00usb@5a3e45ed+0x4000]
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.173710] PC is at 0xbf336818 [rt2x00lib@9c498de5+0xa000]
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.180508] LR is at 0xbf336898 [rt2x00lib@9c498de5+0xa000]
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.187275] pc : [<bf336818>]    lr : [<bf336898>]    psr: 60000013
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.194746] sp : ef099c50  ip : 00000000  fp : 00000001
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.201135] r10: ef099d60  r9 : 00000002  r8 : ee08f748
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.207477] r7 : 00000000  r6 : eda01fa0  r5 : ec75c840  r4 : ef099c74
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.215131] r3 : 00000000  r2 : 00000600  r1 : c0c3c8e0  r0 : 00000000
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.222745] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.230983] Control: 30c5383d  Table: 2da11d00  DAC: 41005015
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.237783] Process kworker/u8:0 (pid: 7, stack limit = 0x92510cef)
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.245104] Stack: (0xef099c50 to 0xef09a000)
[... cut ...]
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.677671] Exception stack(0xef099fb0 to 0xef099ff8)
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.692582] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.701847] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.709106] Code: e0833101 e59338d4 e5933004 e0833102 (e5d31006)
    Feb 15 15:52:32 OpenWrt_Fozzy kernel: [  622.715883] ---[ end trace 4db318219d192a0a ]---

wulfy23, 2022-02-25:
layman opinion... this is a driver/arch issue more than a device related issue...

```
Feb 15 15:52:31 OpenWrt_Fozzy kernel: [  621.830796] Workqueue: phy1 0xbf33e610 [rt2x00usb@5a3e45ed+0x4000]
```

consider providing your /etc/config/wireless and switching out the usb-wifi dongle for something else (or giving it more power, cooling or alternate cable etc. etc.)

gmanic, 2022-02-25:
Sure, here's my /etc/config/wireless. I doubt, though, it being a device issue, as Openwrt was running just fine with 19.xx with the exact same hardware (well, the sd card is different, I admit). Only thing changed being sysupgrading to 21.02 and adjusting the path definitions to match current layout. Power supply is plenty and also unchanged.

```
config wifi-device 'radio0'
	option type 'mac80211'
	option channel '11'
	option hwmode '11g'
	option path 'platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
	option htmode 'HT20'
	option country 'DE'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0'
	option hwmode '11g'
	option channel 'auto'
	option country 'DE'
	option cell_density '0'

config wifi-iface 'radio0_xxxxxx_'
	option network 'trm_wwan'
	option device 'radio0'
	option mode 'sta'
	option ssid 'xxxxx'
	option key 'xxxxxx'
	option encryption 'psk-mixed'
	option disassoc_low_ack '0'
	option disabled '0'

config wifi-device 'radio2'
	option type 'mac80211'
	option path 'platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0'
	option htmode 'HT20'
	option hwmode '11g'
	option channel 'auto'
	option country 'DE'
	option disabled '0'
	option cell_density '0'

config wifi-iface 'wifinet6'
	option network 'lan'
	option ssid 'xxxxx'
	option device 'radio2'
	option mode 'ap'
	option key 'xxxxxx'
	option wmm '0'
	option disabled '0'
	option encryption 'sae-mixed'

config wifi-device 'radio3'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
	option htmode 'VHT80'
	option cell_density '0'
	option country 'DE'
	option disabled '0'

config wifi-iface 'wifinet7'
	option ssid 'xxxxxx'
	option device 'radio3'
	option mode 'ap'
	option network 'lan'
	option key 'xxxxxx'
	option encryption 'sae-mixed'
	option disabled '0'

config wifi-iface 'trm_uplink5'
	option ssid 'xxxxxxx'
	option encryption 'psk2+ccmp'
	option device 'radio0'
	option mode 'sta'
	option key 'xxxxxxx'
	option network 'trm_wwan'
	option disabled '1'

```
(SSIDs and keys of course removed)

ynezz, 2022-02-26:
Probably related https://github.com/openwrt/openwrt/issues/9060

ynezz, 2022-02-26:
> It throws the following error, randomly between 5 mins uptime to approx 3-4h uptime.

If it's that reproducible, would it be possible to try latest development snapshot? If it's not possible, would you be willing to try 21.02.2 with kernel debug symbols compiled in, so we get better stack trace?

gmanic, 2022-02-27:
I completely failed with the dev-snapshot.

Installed a 21.02.2 r16495-bf0c965af0, re-setup everything (still there, phew...), but - no internal Wlan. Device is somehow recognized, driver (kmod-rt2x00usb/lib loaded), but no wlan-device.
Hence, I cannot report any update on this issue. :(

No hint in dmesg (full log attached)
```
[    2.734969] usb 1-1.2: New USB device found, idVendor=148f, idProduct=5572, bcdDevice= 1.01
[    2.745879] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.754574] usb 1-1.2: Product: 802.11 n WLAN
[    2.760253] usb 1-1.2: Manufacturer: Ralink
[    2.765682] usb 1-1.2: SerialNumber: 1.0
```
```
[    9.768217] ieee80211 phy1: rt2x00_set_rt: Info - RT chipset 5592, rev 0222 detected
[    9.788125] ieee80211 phy1: rt2x00_set_rf: Info - RF chipset 000f detected
[    9.796739] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[    9.797933] usbcore: registered new interface driver rt2800usb
```

```
[   12.418765] ieee80211 phy1: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[   12.430732] ieee80211 phy1: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
[   12.821640] br-lan: port 2(wlan1) entered blocking state
[   12.828964] br-lan: port 2(wlan1) entered disabled state
[   12.836443] device wlan1 entered promiscuous mode
[   12.911827] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   12.920053] br-lan: port 2(wlan1) entered blocking state
[   12.926823] br-lan: port 2(wlan1) entered forwarding state

```

On Syslog, also nothing I would consider helpful (full log attached)

```
Sun Feb 27 18:10:22 2022 daemon.notice netifd: radio1 (1326): Could not find PHY for device 'radio1'
Sun Feb 27 18:10:22 2022 daemon.notice netifd: Wireless device 'radio1' set retry=0
Sun Feb 27 18:10:22 2022 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Sun Feb 27 18:10:22 2022 daemon.notice netifd: radio2 (1327): Could not find PHY for device 'radio2'
Sun Feb 27 18:10:22 2022 daemon.notice netifd: Wireless device 'radio2' set retry=0
Sun Feb 27 18:10:22 2022 daemon.crit netifd: Wireless device 'radio2' setup failed, retry=0
```

```
Sun Feb 27 18:10:22 2022 daemon.notice netifd: radio1 (1416): WARNING: Variable 'data' does not exist or is not an array/object
Sun Feb 27 18:10:22 2022 daemon.notice netifd: radio1 (1416): Bug: PHY is undefined for device 'radio1'
Sun Feb 27 18:10:22 2022 daemon.notice netifd: Wireless device 'radio1' is now down
Sun Feb 27 18:10:22 2022 daemon.notice netifd: radio2 (1439): WARNING: Variable 'data' does not exist or is not an array/object
Sun Feb 27 18:10:22 2022 daemon.notice netifd: radio2 (1439): Bug: PHY is undefined for device 'radio2'
Sun Feb 27 18:10:22 2022 daemon.notice netifd: Wireless device 'radio2' is now down
Sun Feb 27 18:10:22 2022 daemon.notice netifd: radio3 (1328): command failed: No error information (-524)
Sun Feb 27 18:10:22 2022 daemon.notice netifd: radio0 (1325): command failed: No error information (-524)

```

Unfortunately, there's a new error, very early on: an exception on swapper (I'll check other issues and open a new one, if none shows this).

I'll also check on enabling debugging symbols. Might take some time.

Seems like the files are to big, so here they are:

[https://nedoc.gecius.de/dmesg-20220227.txt](https://nedoc.gecius.de/dmesg-20220227.txt)
[https://nedoc.gecius.de/syslog-20220227.txt](https://nedoc.gecius.de/syslog-20220227.txt)

gmanic, 2022-05-04:
I can confirm that 21.02.3 has resolved this issue on my setup with a Raspi 3B+ and the rt2x00usb WLAN adapter.

System running fine.

closed 2022-05-04 by gmanic