openwrt/openwrt #9247: TLS Server Hello provides no cipher suites at startup

cyounkins, 2022-02-15:
When the router is first booted after losing power, Firefox will frequently show me the error "Cannot communicate securely with peer: no common encryption algorithm(s). Error code: SSL_ERROR_NO_CYPHER_OVERLAP". Chrome says something unspecific about "malformed data".

The issue fixes itself after some time (2-3 minutes). It seems to fix itself faster when the WAN cable is connected, but is not necessary.

Because the error message is unusual and the state can persist for minutes, I thought the device was totally broken.

Initially the system clock is off by ~7 hours and without WAN `ntpd` can't time sync, but I don't think time sync is the problem. Resetting the time back after NTP does not reproduce the issue.

Packet capture reveals that initially no cipher suites are selected in the TLS Server Hello:

[image]

And when it fixes itself, a suite is selected:

[image]

I believe what's happening is wolfSSL is not selecting a cipher due to low system entropy. After some time of entropy collection, wolfSSL selects a cipher and a connection can be established.

Todo: Verify kernel entropy estimate (`/proc/sys/kernel/random/entropy_avail`) is a good indicator of whether ciphers will be presented 

`urngd` is installed and starts 16 seconds into boot:
```
[   16.662840] urngd: v1.0.2 started.
```

There are things we can do to mitigate this issue:
1. Delay starting the HTTP daemon until sufficient entropy has been collected. Entropy can be estimated from `/proc/sys/kernel/random/entropy_avail` but I don't know what threshold to use.
2. I believe some systems save ~1KB of entropy to storage and feed it into the entropy pool at boot.

Hardware: Ubiquiti EdgeRouter X
Firmware: OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-21.295.67054-13df80d

Forum discussion on this issue: https://forum.openwrt.org/t/firefox-ssl-error-no-cypher-overlap-wolfssl-update-strategy/113210

ynezz, 2022-02-16:
> Firmware: OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-21.295.67054-13df80d

Would it be possible to reproduce it with [21.02-SNAPSHOT](https://downloads.openwrt.org/releases/21.02-SNAPSHOT/) which is almost 21.02.2 and should contain wolfssl 5.1.1 (you're using 4.8.1).

cyounkins, 2022-02-17:
I was not able to reproduce it with the linked snapshot "OpenWrt 21.02-SNAPSHOT r16485-59e7ae8d65" so I'll close this for now. If someone is able to reproduce with 21.02.2 or later, please re-open.

closed 2022-02-17 by cyounkins