openwrt/openwrt #9248: Upcoming release requirements

aparcar, 2022-02-15:
This is a meta issue to keep track of what's needed for the next release.

- [ ] All targets must be on Kernel 5.10
  - #5048 
- [ ]  `firewall4` must work
  - https://github.com/openwrt/packages/issues/16818
  - https://github.com/openwrt/luci/issues/5409
  - https://github.com/openwrt/routing/issues/731
- [ ] more

All project members may edit this issue and add missing features/bugs

nicklowe, 2022-02-21:
With WPA3, I suggest enabling H2E for SAE before forking: https://github.com/openwrt/openwrt/pull/9300

Qosifyck, 2022-02-25:
Please @aparcar, can you migrate OpenWrt to [odhcpd + unbound](https://openwrt.org/docs/guide-user/services/dns/dot_unbound#command-line_instructions) by default and remove dnsmasq (because it's garbage) and if you do it would be the biggest change (for better) in OpenWrt since its founding.

@aparcar you once asked it in the forum:
- [Odhcpd by default? - OpenWrt Forum](https://forum.openwrt.org/t/odhcpd-by-default/92704)

jow-, 2022-02-25:
We will not replace dnsmasq with odhcpd + unbound at this point in time.

Qosify, 2022-02-27:
@aparcar I don't understand why you don't want to replace dnsmasq with [odhcpd + unbound](https://openwrt.org/docs/guide-user/base-system/dhcp_configuration#replacing_dnsmasq_with_odhcpd_and_unbound), if Unbound is much better and also encrypts DNS by default without having to install external packages and offers better performance than dnsmasq and all the web pages will load super fast 100% verified.

PolynomialDivision, 2022-02-27:
Would have been nice to have https://github.com/openwrt/openwrt/pull/4721 included. (cc @robimarko) However, with a feature freeze on 1 March, we would need to be very quick converting the last boards. ;)

PolynomialDivision, 2022-02-27:
Sysupgrade is broken on ath79 ubnt-xm, due to oom-killer killing sysupgrade.

We have two options:
1) Finding more processes to kill before doing sysupgrade
2) Move ath79 ubnt-xm to tiny like I did in the PR: https://github.com/openwrt/openwrt/pull/4879

champtar, 2022-02-27:
@PolynomialDivision any manual workaround ? I still have a bunch of them in the field :(

robimarko, 2022-02-27:
> Would have been nice to have #4721 included. (cc @robimarko) However, with a feature freeze on 1 March, we would need to be very quick converting the last boards. ;)

Unfortunately, I don't think this is viable before the feature freeze and it should not be rushed.

reusu, 2022-02-28:
fullcone nat support？

varoudis, 2022-03-01:
#4996 maybe? arm64 with UEFI.

jow-, 2022-03-01:
> fullcone nat support？
Just add static SNAT and DNAT rules?

nicklowe, 2022-03-01:
We ought to also cover enabling H2E SAE in the case that wpa_supplicant is used as a client. I will create a patch.

DarknessOpen, 2022-03-01:
> Just add static SNAT and DNAT rules?

Is not the same.

I would also like Full cone NAT support.

In stock firmware TOTOLINK A8000RU has Full cone NAT support:
[image]

* [Gaming with Full Cone vs Symmetric NAT Routers](https://web.archive.org/web/20210506124657/https://badmodems.com/Forum/viewtopic.php?t=21)
> For some voice and video concurrent services or games, there might be some connection issue with symmetric NAT. You might switch to full-cone NAT for a better connection. However, with full-cone all external host can send packets to intranet.

> "A full cone NAT is one where all requests from the same internal IP address and port are mapped to the same external IP address and port. Furthermore, any external host can send a packet to the internal host, by sending a packet to the mapped external address."

> Single NAT is where either a modem/router device or a modem only and a external router, where either the modem/router or external router handles all routing and NAT traffic and is the main host router for all connected devices including game consoles and PCs. The ISP public IP address is translated from the ISP modem to the gaming device seamlessly. WAN to LAN. WAN side ISP public IP address ##.##.###.### to a LAN side IP address of 192.168.#.#. The WAN side ISP public IP address ##.##.###.### is correctly translated to game consoles/PCs by just the external router handling the NAT traffic.
This configuration is ideal and what works best for gaming configurations and should be configured as such and is the "golden" configuration for general home networking and gaming environments.

> MS and Sony may have not fully tested two or more game console/PC and same or different game environments. Current marketing perspective maybe thinking of one household, one game console. uPnP code handling between the consoles and games are problematic. Router Mfr use of Symmetric NAT type only inhibits good NAT handling for two or more game consoles/PCs, where FULL CONE NAT handles two or more game consoles/PCs playing the same game at the same time correctly. Also games may have specific problems in handling of NAT when being played on routers with Symmetric NAT as well which should be looked at for multiple game environments by game developers. There also could be a problem with Symmetric NAT in itself as well which should be looked at. Can Symmetric NAT handle two or more game consoles? Again, all these considerations are only seen when using two or more game consoles. A mixed of consoles would also be a consideration as well, i.e. Xbox and Play station on the same router online at the same time.
>
>Game Mfrs also have a role to play in this and from what i've seen, some games are not handling uPnP ports and NAT kinds correctly with two or more consoles running same game, which needs to be addressed by game Mfrs which use and give NAT status in there game. I'm seeing that recently that two consoles running the same game are achieving OPEN NAT on Port Address Restricted NAT routers on the consoles network settings with two consoles, however after loading the game, one game console reports OPEN NAT and the other console reports MODERATE NAT. This is with the new MS xbox one port address change feature that was released fully on Oct 17th when it hit my xbox one console. When using a FULL CONE NAT router with two same game running consoles, OPEN NAT is seen across the board and is not an issue.

champtar, 2022-03-01:
Many protocols figured out NAT traversal and full cone NAT is not present in OpenWrt since forever so this can't be a release blocker, this is a normal feature request.

jow-, 2022-03-01:
"A full cone NAT is one where all requests from the same internal IP address and port are mapped to the same external IP address and port. Furthermore, any external host can send a packet to the internal host, by sending a packet to the mapped external address."

Sounds like a static SNAT/DNAT setup to me.

DarknessOpen, 2022-03-02:
> Sounds like a static SNAT/DNAT setup to me.

I am sorry to tell you that it is not and due to your wrong thinking you have not implemented full cone NAT support in OpenWrt.

People have been asking/pleading for years to add Full cone NAT support by default in OpenWrt, but you guys have been ignoring them.

DarknessOpen, 2022-03-02:
> Sounds like a static SNAT/DNAT setup to me.

Sorry to tell you that you are wrong and due to your wrong thinking you have not implemented Full cone NAT support in OpenWrt.

People have been asking/pleading for years to add Full cone NAT support by default in OpenWrt, but you guys have been ignoring them.

jow-, 2022-03-02:
> I am sorry to tell you this, but you are wrong and due to your wrong thinking you have not implemented Full cone NAT support in OpenWrt.

What is wrong about the statement that a static SANT and DNAT rule achieve "full cone NAT" properties? What exactly makes "full cone NAT" different from a static SNAT and DNAT rule? Autoconfiguration?

DarknessOpen, 2022-03-02:
> What exactly makes "full cone NAT" different from a static SNAT and DNAT rule? Autoconfiguration?

I'm going to go to your house and show you the difference with my fists.

DarknessOpen, 2022-03-02:
> What exactly makes "full cone NAT" different from a static SNAT and DNAT rule? Autoconfiguration?

I'm going to go to your house and show you the difference with my fists.
[image]

reusu, 2022-03-02:
> What is wrong about the statement that a static SANT and DNAT rule achieve "full cone NAT" properties? What exactly makes "full cone NAT" different from a static SNAT and DNAT rule? Autoconfiguration?

i think 
masquerade = symmetric nat
snat + dnat = port forward != fullcone nat
that's all

jow-, 2022-03-02:
> snat + dnat = port forward != fullcone nat

That still does not explain what "fullcone NAT" *is*. From what I could gather in the net, it is basically a form of NAT where a specific internal port of a specific internal host is mapped to a unique external port. A traditional port forward ensures that a given external port is forwarded to a specific internal host and port. Adding an additional SNAT rule to explicitly rewrite *outbound* traffic from the internal host and port to the specific external port without relying on MASQUERADE egress port auto allocation will achieve exactly that. What else is needed to achieve "full cone NAT" ?

aparcar, 2022-03-02:
@jow- using my detective skills I found the following [code fragment](https://github.com/x-wrt/com.x-wrt/blob/09c93fd1ced8d477b9b0086593b3295799306382/natcap/files/natcapd.sh#L275):

```shell
		ipset create cone_wan_ip iphash hashsize 32 maxelem 256 >/dev/null 2>&1
		ipset flush cone_wan_ip
		devs=`ip r | grep default | grep -o "dev ".* | awk '{print $2}'`
		for dev in $devs; do
			ips=`ip addr list dev $dev | grep -o inet" ".* | awk '{print $2}' | cut -d/ -f1`
			for ip in $ips; do
				ipset add cone_wan_ip $ip >/dev/null 2>&1
			done
		done
```

jow-, 2022-03-02:
@aparcar - and how is the `cone_wan_ip` set used?

WinLinux1028, 2022-03-02:
There is a patch to use fullcone nat:
https://github.com/llccd/openwrt-fullconenat/tree/dev
However, it doesn't seem to be compatible with nftables.

LA-CUCUCCUCURACHA, 2022-03-02:
In Asus routers you can also change the nat type:
[image]

LA-CUCUCCUCURACHA, 2022-03-02:
> There is a patch to use fullcone nat:
> https://github.com/llccd/openwrt-fullconenat/tree/dev
> https://github.com/LGA1150/openwrt-fullconenat
 > @llccd @LGA1150

In Asus routers you can also change the nat type:
[image]

LA-CUCUCCUCURACHA, 2022-03-02:
> There is a patch to use fullcone nat:
> https://github.com/llccd/openwrt-fullconenat/tree/dev
> https://github.com/LGA1150/openwrt-fullconenat

In Asus routers you can also change the nat type:
[image]

@llccd @LGA1150 @jow- @aparcar

LA-CUCUCCUCURACHA, 2022-03-02:
#7021

The default OpenWrt netfilter implements "Port Restricted Cone" NAT. Sometimes it's useful to have less restricted NAT.

Someone recently created a netfilter extension that implements "Full Cone" NAT. Please include it in the package feeds.

Full Cone NAT description:
https://tools.ietf.org/html/rfc3489#section-5

Netfilter extension:
https://github.com/Chion82/netfilter-full-cone-nat

OpenWrt package:

https://github.com/LGA1150/openwrt-fullconenat
https://github.com/LGA1150/fullconenat-fw3-patch

mkresin, 2022-03-02:
Instead of posting script, links or insults one should go back and provide an answer to jows question

> What exactly makes "full cone NAT" different from a static SNAT and DNAT rule?"

With SNAT you do not know in advance which src port is used by the gateway to send the request (only high ports are used as src port and so on). Hence, you can not add a (static) DNAT rule for incoming traffic matching the gateways used src port.

Speaking in terms of SNAT and DNAT, with full cone NAT a DNAT rule is automatically added for the src port used by the gateway. DNAT dest port is the internal clients src port. I wouldn't be surprised if the pasted script mimics the behaviour.

I'm not entirely sure if with full cone NAT the gateway needs to use/uses the same src port as the internal client.

https://en.wikipedia.org/wiki/Network_address_translation#Methods_of_translation has an okayish illustration about the differences.

champtar, 2022-03-02:
Another solution to the full cone NAT problem is switching to IPv6 ;)

LA-CUCUCCUCURACHA, 2022-03-02:
IPv6 🤢🤮

LA-CUCUCCUCURACHA, 2022-03-03:
> Instead of posting script, links or insults one should go back and provide an answer to jows question

The fools do not want to understand with words because they are deceived with their own lies due to their lack of knowledge and arrogance not recognizing that they are wrong and the only answer I can give to those lazy ones who do not want to learn are my fists.

[image]

predators46, 2022-03-03:
I can't wait for the 5.10.x kernel to be on the stable version of openwrt. i need musl version in snapshoot

SorryJow, 2022-03-03:
I'm sorry @jow- for insulting you and threatening you.

[image]

SorryJow, 2022-03-03:
Es tut mir leid @jow- 

https://user-images.githubusercontent.com/100845128/156558713-8ff9b62c-3188-4cb0-8b73-2e9867085248.mp4

dfateyev, 2022-03-03:
Could we reduce the spam amount, and discuss here real blockers and requirements for the upcoming release.
This thread needs some moderation. Also, part of the messages can be wiped out entirely — they have no useful info and just spam.

aparcar, 2022-03-04:
From my understanding we'll need the new dnsmasq version since it support nftables, could people please test the PR I linked?

BabochkaAce, 2022-03-04:
[odhcpd](https://github.com/libremesh/lime-packages/issues/294) support nftables?

Headcrabed, 2022-03-05:
> Another solution to the full cone NAT problem is switching to IPv6 ;)

Sadly, not every software supports ipv6, especially games. Besides, bug that ipv6 package loss when using flow offload haven
t been fixed yet.

Borromini, 2022-03-05:
@aparcar I think https://github.com/openwrt/openwrt/issues/9377 'deserves' a place on this list as well, since it's a regression compared to 21.02.

dfateyev, 2022-03-06:
@nicklowe you have mentioned H2E SAE support in AP client mode, do you have any plans to provide it?

stangri, 2022-03-07:
I've asked this before and I'm sorry if I asked in the wrong issues/places.

My `vpn-policy-routing` and `vpnbypass` packages heavily depend on the `dnsmasq-full`'s ipset support. What's the plan for `dnsmasq-full` going forward, would it still include only `ipset` support in the next release or only `nftset` support or would it be split into `dnsmasq-full-ipset` and `dnsmasq-full-nftset`?

nicefile, 2022-03-08:
@aparcar I would add regression [#3832](https://github.com/openwrt/openwrt/issues/8833) regarding switch ports with 10/100 Mbit PHY found in fritzbox 7360 7360v2 7362sl .lantiq xrx200 target

BabochkaAce, 2022-03-09:
[odhcpd](https://github.com/libremesh/lime-packages/issues/294) or unbound support nftables?

aparcar, 2022-03-09:
@BabochkaAce Please stop cross posting and repeating the same question again and again. It's unrelated to the release requirements. Please use the forum and open a new topic or issue here on GitHub.

BabochkaAce, 2022-03-09:
Someone answered my question elsewhere.

As my Mother Russian says, "Я люблю спать, есть и какать" which means that Ukraine and all of Europe belong to Russia.

https://www.youtube.com/watch?v=_lJPr-L6gNo

Ansuel, 2022-03-11:
@aparcar should we add this to the known issue? (it's like that from ages... considering wps push button is provided in luci, i think it should be noted)

https://github.com/openwrt/openwrt/issues/9369

allhailCdosdude, 2022-03-13:
Is it possible, please to add regression[ 3832]( https://github.com/openwrt/openwrt/issues/8833)

obsy, 2022-03-19:
Bug in qmi script: https://github.com/openwrt/openwrt/issues/9501

robertfoss, 2022-03-19:
Ubiquiti Nanostations can no longer be flashed due to ram size limitations, a PR improving this situation has been submitted #4879.

mpratt14, 2022-03-19:
I opened #9507 

probably just something to add to "known bugs"

I don't know if it can be fixed, but if so, it can wait until after branching anyway

MaxSKash, 2022-03-24:
Enable `management` in openvpn

mans0n, 2022-03-28:
I've just got to know that bcm53xx had a breaking change (convert to DSA) but did not bumped `DEVICE_COMPAT_VERSION`.
https://github.com/openwrt/openwrt/commit/e9672b1a8fa4714cbc35d5964caf704b2e571f35#commitcomment-58577826
The commit did add a simple migration script but I don't think that is enough... (especially for the stable release)
Ref: 47b531de3250 b1df48caac1b

kofec, 2022-03-29:
One small issue: https://github.com/openwrt/openwrt/issues/9580

howl, 2022-04-07:
#9676 should be considerer as the issue makes the connection to work very bad and is because a feature debuting in this release.

Regloom, 2022-04-18:
Please cosider I2Pd update to 2.41 (mipselsf-k3.4 arch). Here is a link to issue [783](https://github.com/Entware/Entware/issues/783) on Entware.

WinLinux1028, 2022-05-13:
#9881 I found a problem of 22.03(latest snapshot)

Tabiskabis, 2022-05-21:
I used uci interfaces to map devices to firewall zones, and have OpenVPN configure those devices. This worked fine with fw3. 
fw4 however does not assign these devices to their zones. I worked around the issue by switching the interfaces protocol from "none" to "static" with no additional settings.
This behaviour seems faulty to me. In case that this is not considered a bug, it should at least get well documented.

howl, 2022-05-22:
Another fw4 nftables issue #9923.

[38 more replies]