openwrt/openwrt #9249: SFP+ Strange Issue

cdrag45, 2022-02-16:
so had a strange issue today me and my roommate put a (Intel 82599EN-Based Ethernet Network Interface Card, 10G Single-Port SFP+, PCIe 2.0 x 8, Tall&Short Bracket) in a dell poweredge r610 and for all intensive purposes it nuked the install and internet didnt work right even after removing the card
after booting with the card the devices on a unmanaged switch had Internet but the router didn't have Internet and nothing we could do was able to restore function so we had to reinstall and restore a configuration

when booting with the card it takes eth0 instead of adding a new interface which doesn't happen for any other type of card we have used Ethernet wise so we don't know what is going on our how to troubleshoot farther
we have checked the bios any settings seem to be ok

going to include the system log and the kernel log of a testing environment we have reproduced the problem consistently 
as a side note the card is 82599en which i checked with the card to make sure its not a driver conflict since the driver for 82599 is included in the image for openwrt release and the driver that is distributed by intel for both is the same driver so i would assume that is the file that is prepackaged inside openwrt(correct me if im wrong as i cant find anything to say that it packages a different version)
Kernel
[https://www.mediafire.com/file/p1t504jla7w5xq9/kernal/file](url)
syslog
[https://www.mediafire.com/file/7b8vy7yzqq82zlk/sys/file](url)

aparcar, 2022-02-17:
Please try to find a better title for this issue. Also post logs in the issue surrounded by `<details></details>`

cdrag45, 2022-02-18:
> Please try to find a better title for this issue. Also post logs in the issue surrounded by `<details></details>`

the logs were to big to include in the post which is why i uploaded txt file to mediafire i tried using the formating to post them but it wouldnt let me for the size

as for the title i really wasnt sure what to choose since i really dont entirely know what the issue is since everything ive been reading says it should work