openwrt/openwrt #9251: [Openwrt21.02] enable hostapd & wpa_cli report error

heyg, 2022-02-16:
when build hostapd,
build_dir/target-aarch64_cortex-a53_glibc/hostapd-full-internal/hostapd-2020-06-08-5a8b3662/hostapd'

===> report the follow issue,
staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_glibc/lib/gcc/aarch64-openwrt-linux-gnueabi/8.4.0/../../../../aarch64-openwrt-linux-gnueabi/bin/ld: <artificial>:(.text+0xb094): undefined reference to `sta_track_claim_taxonomy_info'

cyounkins, 2022-02-16:
More details are needed. Is this happening when you are installing the `hostapd` package from within the web UI? Please put the full build output in a GitHub Gist and link it here. What hardware are you using?

heyg, 2022-02-17:
HI Craig,
Thanks your response,

I did this,
1) config,
==============================>
#
# WirelessAPD
#
# CONFIG_PACKAGE_eapol-test is not set
# CONFIG_PACKAGE_eapol-test-openssl is not set
# CONFIG_PACKAGE_eapol-test-wolfssl is not set
CONFIG_PACKAGE_hostapd=y                                                                                                                                                                     
# CONFIG_PACKAGE_hostapd-basic is not set
# CONFIG_PACKAGE_hostapd-basic-openssl is not set
# CONFIG_PACKAGE_hostapd-basic-wolfssl is not set
CONFIG_PACKAGE_hostapd-common=y
# CONFIG_PACKAGE_hostapd-mini is not set
# CONFIG_PACKAGE_hostapd-openssl is not set
# CONFIG_PACKAGE_hostapd-utils is not set
# CONFIG_PACKAGE_hostapd-wolfssl is not set
# CONFIG_PACKAGE_hs20-client is not set
# CONFIG_PACKAGE_hs20-common is not set
# CONFIG_PACKAGE_hs20-server is not set
CONFIG_PACKAGE_wpa-cli=y
CONFIG_PACKAGE_wpa-supplicant=y
CONFIG_WPA_RFKILL_SUPPORT=y
CONFIG_WPA_MSG_MIN_PRIORITY=3
<==============================
2) kernel use 5.4, and buildin cfg80211

3) HW config
CONFIG_TARGET_ARCH_PACKAGES="aarch64_cortex-a53"

4) gcc,
CONFIG_GDB=y
CONFIG_USE_GLIBC=y
CONFIG_LIBC_USE_GLIBC=y
CONFIG_SSP_SUPPORT=y
CONFIG_BINUTILS_VERSION_2_34=y
CONFIG_BINUTILS_VERSION="2.34"
CONFIG_GCC_VERSION="8.4.0"
# CONFIG_GCC_USE_IREMAP is not set
CONFIG_LIBC="glibc"
CONFIG_TARGET_SUFFIX="gnueabi"

error log,
===================================>
make[4]: Entering directory '/user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85'
make[5]: Entering directory '/user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/src'
make[5]: 'dnsmasq' is up to date.
make[5]: Leaving directory '/user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/src'
install -d /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/ipkg-install/usr/sbin
install -d /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/ipkg-install/usr/share/man/man8
install -m 644 man/dnsmasq.8 /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/ipkg-install/usr/share/man/man8 
install -m 755 src/dnsmasq /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/ipkg-install/usr/sbin
make[4]: Leaving directory '/user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85'
touch /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.built
rm -rf /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq.installed /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq
mkdir -p /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq
install -d -m0755 /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq/usr/sbin
cp -fpR /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/ipkg-install/usr/sbin/dnsmasq /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq/usr/sbin/
install -d -m0755 /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq/etc/config
install -m0600 ./files/dhcp.conf /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq/etc/config/dhcp
install -m0600 ./files/dnsmasq.conf /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq/etc/dnsmasq.conf
install -d -m0755 /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq/etc/init.d
install -d -m0755 /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq/etc/hotplug.d/dhcp
install -d -m0755 /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq/etc/hotplug.d/neigh
install -d -m0755 /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq/etc/hotplug.d/ntp
install -d -m0755 /user/bill_he/lede-source/build_dir/target-aarch64_cortex-a53_glibc/dnsmasq-nodhcpv6/dnsmasq-2.85/.pkgdir/dnsmasq/etc/hotplug.d/tftp
[... 520 lines cut ...]
  CC  ../src/ap/ieee802_11_ht.c
  CC  ../src/common/gas.c
  CC  ../src/ap/gas_serv.c
  CC  ../src/drivers/driver_common.c
/user/bill_he/lede-source/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_glibc/lib/gcc/aarch64-openwrt-linux-gnueabi/8.4.0/../../../../aarch64-openwrt-linux-gnueabi/bin/ld: /user/bill_he/lede-source/tmp/cc9cARQV.ltrans7.ltrans.o: in function `ap_sta_add':
<artificial>:(.text+0xb088): undefined reference to `sta_track_claim_taxonomy_info'
/user/bill_he/lede-source/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_glibc/lib/gcc/aarch64-openwrt-linux-gnueabi/8.4.0/../../../../aarch64-openwrt-linux-gnueabi/bin/ld: <artificial>:(.text+0xb094): undefined reference to `sta_track_claim_taxonomy_info'
collect2: error: ld returned 1 exit status
[... cut ...]
make[2]: Leaving directory '/user/bill_he/lede-source'
package/Makefile:110: recipe for target '/user/bill_he/lede-source/staging_dir/target-aarch64_cortex-a53_glibc/stamp/.package_compile' failed
make[1]: *** [/user/bill_he/lede-source/staging_dir/target-aarch64_cortex-a53_glibc/stamp/.package_compile] Error 2
make[1]: Leaving directory '/user/bill_he/lede-source'
/user/bill_he/lede-source/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2
<===================================

heyg, 2022-02-17:
if only disable"CONFIG_PACKAGE_hostapd", use the follow config build ok,
===========================================>
#
# WirelessAPD
#
# CONFIG_PACKAGE_eapol-test is not set
# CONFIG_PACKAGE_eapol-test-openssl is not set
# CONFIG_PACKAGE_eapol-test-wolfssl is not set
# CONFIG_PACKAGE_hostapd is not set                                                                                                                                                          
# CONFIG_PACKAGE_hostapd-basic is not set
# CONFIG_PACKAGE_hostapd-basic-openssl is not set
# CONFIG_PACKAGE_hostapd-basic-wolfssl is not set
CONFIG_PACKAGE_hostapd-common=y
# CONFIG_PACKAGE_hostapd-mini is not set
# CONFIG_PACKAGE_hostapd-openssl is not set
# CONFIG_PACKAGE_hostapd-utils is not set
# CONFIG_PACKAGE_hostapd-wolfssl is not set
# CONFIG_PACKAGE_hs20-client is not set
# CONFIG_PACKAGE_hs20-common is not set
# CONFIG_PACKAGE_hs20-server is not set
CONFIG_PACKAGE_wpa-cli=y
CONFIG_PACKAGE_wpa-supplicant=y
CONFIG_WPA_RFKILL_SUPPORT=y
CONFIG_WPA_MSG_MIN_PRIORITY=3
<===========================================

cyounkins, 2022-02-17:
Please put the build log in a Gist (gist.github.com/) and wrap the config in a code block (three backtick ` characters, check the preview). Additionally, can you explain more generally what you are trying to do? Build a custom image?

heyg, 2022-02-17:
can you explain more generally what you are trying to do? Build a custom image?
A: yes, I am trying to build a new image (include hostapd)

heyg, 2022-02-18:
Hi Guys,
I found must enable "CONFIG_PACKAGE_kmod-cfg80211"  will fixed this issue.
That means kernel cannot build in cfg80211. So i think we can close this issue.