openwrt/openwrt #9256: LED fails on new revisions of Ubiquiti UniFi 6 LR

jrosink, 2022-02-16:
The Unify 6 LR originally shipped with a RGB LED, newer revisions of this AP are shipped with a "classic" Ubiquiti Blue/White LED. Quote from current datasheet:

> *Due to chip supply shortages, the U6LR's LED has been limited to only white and blue color modes.
https://dl.ui.com/ds/u6-lr_ds.pdf

After power on a recent U6-LR model with OpenWRT the LED is flashing a few times but never lights up again during and after boot. The boot itself is succesfull, only the visual feedback from the LED is missing. Trying to set the LED on a running system also fails. 
This issue is purely cosmetic and doesn't affect the working of the device. Only after first flash to OpenWRT one might think the device isn't functional (bricked), but ssh and/or luci is available.

Log from current snapshot build (2022-02-16 - r18809-5a0975f7ef):

Linux version 5.10.100

`[   19.845162] ubnt-ledbar 0-0030: Error initializing LED transaction: 00

[   38.565153] ubnt-ledbar 0-0030: Error initializing LED transaction: 00

[   38.571773] leds blue: Setting an LED's brightness failed (-22)

[   57.285153] ubnt-ledbar 0-0030: Error initializing LED transaction: 00

[   57.291727] leds green: Setting an LED's brightness failed (-22)

[   76.005154] ubnt-ledbar 0-0030: Error initializing LED transaction: 00

[   76.011727] leds red: Setting an LED's brightness failed (-22)

[   76.017715] ubnt-ledbar: probe of 0-0030 failed with error -22`

blocktrron, 2022-02-19:
Can you share the content of `/sys/kernel/debug/gpio` in Ubiquitis stock firmware?

jrosink, 2022-02-19:
```
GPIOs 409-511, platform/10005000.pinctrl, 10005000.pinctrl:
 gpio-471 (                    |reset               ) in  hi IRQ
```

HRio, 2022-05-25:
Just got a new LR model that is still on stock firmware, any checks I can do before flashing OpenWRT?

HRio, 2022-05-26:
From the boot log with stock firmware
```
...
[    4.034863] input: gpio-keys as /devices/platform/gpio-keys/input/input0
...
[    5.297367] creating /proc/gpio/
[    5.297389]          /proc/gpio/ledbar_control
[    5.297395]          /proc/gpio/led_pattern
[    5.297400]          /proc/gpio/led_tempo
[    5.297405]          /proc/gpio/poe_passthrough
[    5.297411]          /proc/gpio/gps_reset
[    5.297417]             /proc/gpio/ble_reset
[    5.297421]             /proc/gpio/ble_normalmode
```

`/proc/gpio/led_pattern` is used to control the leds

| **Value** | **Blue LED** | **White LED** |
| ---          | --- | --- |
|   0         | off | off |
|   1         | on | off |
|   2         | off | on |
|   3         | on | on |

HRio, 2022-05-28:
Getting out a `dts `from the data in `/sys/firmware/devicetree/base` gives
```
        gpio-leds {
                compatible = "gpio-leds";

                power {
                        label = "ubnt:white:personality";
                        linux,default-trigger = "external0";
                        gpios = <0x20 0x44 0x00>;
                };

                wifi {
                        label = "ubnt:blue:personality";
                        linux,default-trigger = "external1";
                        gpios = <0x20 0x43 0x00>;
                };
        };
```

HRio, 2022-05-28:
@dangowrt & @blocktrron whats the best way to handle that there are now two different versions of UniFi 6 LR, one original with RGB LED and one with just two LEDs controlled with gpio?

dangowrt, 2022-05-31:
As they are effectively different devices they will need different device tree and images (in `target/linux/mediatek/image/mt7622.mk`) built for each of them.
So I would remove the LED related parts from `mt7622-ubnt-unifi-6-lr.dtsi` and then include that file in two new to-be-created `mt7622-ubnt-unifi-6-lr-v1.dtsi` and `mt7622-ubnt-unifi-6-lr-v2.dtsi`, where the `v1` declares the I2C controlled LEDs and the `v2` declares the more simple GPIO connected LEDs.
Then rename `mt7622-ubnt-unifi-6-lr.dts` to `mt7622-ubnt-unifi-6-lr-v1.dts` and include the new `mt7622-ubnt-unifi-6-lr-v1.dtsi` in that.
Have another copy as `mt7622-ubnt-unifi-6-lr-v2.dts` and there include `mt7622-ubnt-unifi-6-lr-v2.dtsi`.
Same for the `mt7622-ubnt-unifi-6-lr-ubootmod.dts` which should become `mt7622-ubnt-unifi-6-lr-v1-ubootmod.dts` and `mt7622-ubnt-unifi-6-lr-v2-ubootmod.dts`.

HRio, 2022-06-18:
There are now snapshot builds available for this device: https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/

jrosink, 2022-06-28:
@HRio You Rock!

Confirmed working on three different LR6 v2 AP's. Both static and dynamic (netdev + tx/rx) LED configuration are working for both White & Blue Leds.

closed 2022-06-28 by jrosink