openwrt/openwrt #9265: Today's realtek changes cause crashes on Netgear GS108T v3

hardfalcon, 2022-02-17:
Somewhere between 923cc869a6eec7dd6fdf728b2adb05824e4b9ac0 and a93dfff10e067c152d4872719ec7837fcd8c94a3 a bug/regression/whatever was introduced that makes my Netgear GS108T v3 crash after a few minutes.

https://github.com/openwrt/openwrt/compare/923cc869a6eec7dd6fdf728b2adb05824e4b9ac0...a93dfff10e067c152d4872719ec7837fcd8c94a3

One symptom appears to be error messages like the following (disregard the incorrect timestamps, these error messages are roughly 20-30 minutes old):
```
Thu Feb 17 17:19:05 2022 kern.warn kernel: [  423.013092] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800220, cur a2800224
Thu Feb 17 17:19:06 2022 kern.warn kernel: [  423.723713] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800224, cur a2800228
Thu Feb 17 17:19:06 2022 kern.warn kernel: [  423.733896] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800224, cur a2800228
Thu Feb 17 17:19:06 2022 kern.warn kernel: [  423.746053] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800224, cur a2800228
Thu Feb 17 17:19:06 2022 kern.warn kernel: [  423.757488] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800224, cur a2800228
Thu Feb 17 17:19:14 2022 kern.warn kernel: [  432.047639] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800220, cur a2800228
Thu Feb 17 17:19:14 2022 kern.warn kernel: [  432.240976] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800224, cur a2800228
Thu Feb 17 17:19:15 2022 kern.warn kernel: [  432.412697] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800220, cur a2800228
Thu Feb 17 17:19:15 2022 kern.warn kernel: [  432.773592] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800224, cur a280022c
Thu Feb 17 17:19:15 2022 kern.warn kernel: [  432.842995] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800224, cur a280022c
Thu Feb 17 17:19:15 2022 kern.warn kernel: [  433.185728] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800224, cur a280022c
Thu Feb 17 17:19:16 2022 kern.warn kernel: [  433.945251] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800228, cur a2800230
Thu Feb 17 17:19:21 2022 kern.warn kernel: [  439.250012] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800220, cur a2800230
Thu Feb 17 17:19:21 2022 kern.warn kernel: [  439.260167] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800220, cur a2800230
Thu Feb 17 17:19:21 2022 kern.warn kernel: [  439.271469] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800228, cur a2800230
```

They do not *constantly* flood `logread`, but show up in bursts of perhaps 1-2 dozen lines, they seem to occur only after a few minutes of operation, leaving the router crashed/locked up after a few minutes (at least that's what it looks like on SSH - I haven't disassembled the device yet to connect a UART adapter and check the serial console).

I do use a slightly "non-standard" configuration/build - for example, both the build from last night without the issue and today's build use kernel 5.10.101, I use SELinux (but with `enforcing=0` patched into the cmdline), and I use all the available compile-time hardening flags turned to "maximum". However, I'm reasonably confident that the issue I'm seeing is not related to these "non-standard" changes, since I've been using this configuration for 1-2 months now without any problems, and with a rebuilt from the latest git master more or less every time kernel.org released a new kernel in the 5.10 branch. I'm also pretty confident that the issue is not caused by the switch from `firewall3`/`iptables` to `firewall4`/`nftables` since I've been using *that* for at least 1-2 weeks now without any problems whatsoever.

svanheule, 2022-02-17:
This issue is currently also being discussed and debugged in the PR https://github.com/openwrt/openwrt/pull/4973#issuecomment-1043308864

dangowrt, 2022-02-18:
I've applied https://github.com/openwrt/openwrt/commit/4d8020072ee291fe4e7b751e2c1490830932bcc4 and hopefully that should fix this.

hardfalcon, 2022-03-13:
This appears to have fixed it indeed - thanks! :)

closed 2022-03-13 by hardfalcon