openwrt/openwrt #9275: AVM Fritzbox LAN ports

allhailCdosdude, 2022-02-18:
First time doing this, so I hope I'm doing it right.

I have two AVM Fritzboxes, and when installed the latest version of Openwrt, only the first two LAN hardware ports work. That means, when I connect an Ethernet cable to them, I can use the browser to login to LuCi. On both devices, the ports are LAN 1 and LAN 2. LAN 3 and LAN 4 do not work. However, on 7362SL, I reverted back to OpenWrt 19.07.7 r11306-c4a6851c72 / LuCI openwrt-19.07 branch git-21.044.30835-34e0d65 on the 7362SL, and all ports work, as described when I had 21.02 installed.

The devices are:
7362 SL
7360

My threads on both are here at the forum.

Thanks a lot.

abajk, 2022-02-19:
Duplicate: https://github.com/openwrt/openwrt/issues/5064 and https://github.com/openwrt/openwrt/issues/8843.

abajk, 2022-02-19:
I think you can close this issue and continue discussions in earlier threads.

abajk, 2022-02-19:
@allhailCdosdude

closed 2022-02-20 by allhailCdosdude