openwrt/openwrt #9280: TP-Link Archer C7 v5 / ath79/generic: Current git very unreliable

xosevp, 2022-02-19:
TP-Link Archer C7 v5 / ath79/generic
OpenWrt SNAPSHOT r18806-2120cad38d / LuCI Master git-22.025.79016-22e2bfb

Lot of errors:
ath10k_pci 0000:00:00.0: failed to parse phyerr tlv payload at byte 0
ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon

```
[    0.000000] Linux version 5.10.100 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r18806-2120cad38d) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Mon Feb 14 13:40:34 2022
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link Archer C7 v5
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 120576K/131072K available (5844K kernel code, 602K rwdata, 1276K rodata, 1228K init, 210K bss, 10496K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] random: get_random_bytes called from start_kernel+0x3c0/0x5b0 with crng_init=0
[    0.000000] CPU clock: 775.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000008] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008234] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.074768] pid_max: default: 32768 minimum: 301
[    0.079766] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.087477] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.097389] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.107955] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.118356] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.125619] pinctrl core: initialized pinctrl subsystem
[    0.134556] NET: Registered protocol family 16
[    0.139962] thermal_sys: Registered thermal governor 'step_wise'
[    0.190491] clocksource: Switched to clocksource MIPS
[    0.203277] NET: Registered protocol family 2
[    0.208005] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.216321] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.225222] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.233321] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.240766] TCP: Hash tables configured (established 1024 bind 1024)
[    0.247582] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.254530] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.262120] NET: Registered protocol family 1
[    0.266733] PCI: CLS 0 bytes, default 32
[    0.276285] workingset: timestamp_bits=14 max_order=15 bucket_order=1
[    0.286659] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.292868] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.304808] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 251)
[    0.315174] pinctrl-single 1804002c.pinmux: 544 pins, size 68
[    0.321778] gpio-export gpio-export: 1 gpio(s) exported
[    0.328016] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[... 306 lines cut ...]
[233307.356190] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[233307.458658] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[254497.733232] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is down
[295963.496455] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is up
[295973.896724] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is down
[295975.976414] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is up
[295982.216711] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is down
[295988.456397] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is up
[296280.696280] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is down
[296294.216094] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is up
[296300.456273] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is down
[296305.656033] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is up
[298939.971074] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is down
[298953.490728] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is up
[298958.691030] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is down
[298963.890687] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 5 is up
[309285.601266] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[309285.703662] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[312646.349488] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[312646.451873] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[314326.673444] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[317087.244039] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[317087.346432] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[317207.260602] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
```

jkhsjdhjs, 2022-08-26:
I also see these errors with 19.07.10 sometimes, after the device has been running for a few months.

xosevp, 2022-08-26:
> I also see these errors with 19.07.10 sometimes, after the device has been running for a few months.

They also appear in 21.02.3. But it's just a warning.
With the latest git-tree or 22.03.0-rcX the net connections are **very unstable**.

akaessens, 2022-12-28:
There are no such issues in 22.03.2 with 50d uptime.

Djfe, 2023-02-05:
@xosevp are you still experiencing these issues?

xosevp, 2023-02-11:
With r22036-1a145ccb0a "Hardware flow offloading" is still broken in this platform/router.
So, I can not test it properly.

akaessens, 2023-02-11:
> With r22036-1a145ccb0a "Hardware flow offloading" is still broken in this platform/router. So, I can not test it properly.

Hardware Flow Offloading is not broken , it's unsupported on this platform

xosevp, 2023-02-11:
> Hardware Flow Offloading is not broken , it's unsupported on this platform

600 Mbps GPON-FTTH:

- with "21.02.5, r16688-fa9a932fdb" down/up: 70/ 77 MB/s
- with "git r22036-1a145ccb0a ": 24/50 MB/s

akaessens, 2023-02-11:
> > Hardware Flow Offloading is not broken , it's unsupported on this platform
> 
> 600 Mbps GPON-FTTH:
> 
>     * with "21.02.5, r16688-fa9a932fdb" down/up: 70/ 77 MB/s
> 
>     * with "git r22036-1a145ccb0a ": 24/50 MB/s

Yes, these numbers are possible with software flow offloading