openwrt/openwrt #9281: MWAN3-LUCI add button is broken.

dragon2611, 2022-02-19:
On the MWAN3 LUCI interface, the add button appears to be greyed out and is not clickable on interfaces,members,policies,rules.etc

Typing in the text box and pressing enter on the keyboard does actually go ahead and create a new entry but the button never activates.

Tested on Edge, Brave, Firefox, Tested with a private tab on Edge incase of a browser caching issue.

Luci openwrt-21.02 branch (git-21.295.67054-13df80d)] /  OpenWrt 21.02.1 r16325-88151b8303
luci-app-mwan3 git-21.126.37401-0ddb72d
x86-64 arch.

aphorise, 2022-08-25:
@dragon2611 it would be nice if you can retest but including a screenshot of the developer / inspect view of the browser for any noticeable errors. I typically find UI issues are more easily conveyed and relatable with a picture :-)

Tutislav, 2023-12-22:
I have same problem, the add button is always disabled.
[image]
It throws this error in console:
[image]

LuCI openwrt-21.02 branch (git-21.231.26241-422c175) / OpenWrt 21.02.0 r16279-5cc0535800