openwrt/openwrt #9283: Problem with libustream-wolfssl20201210

obsy, 2022-02-19:
```
root@OpenWrt:/# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02-SNAPSHOT'
DISTRIB_REVISION='r16497-d1c15c41d9'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 21.02-SNAPSHOT r16497-d1c15c41d9'
DISTRIB_TAINTS=''
```

Lets try download site https://letsencrypt.org/:

```
root@OpenWrt:/# wget -O /dev/null https://letsencrypt.org/
Downloading 'https://letsencrypt.org/'
Connecting to 3.125.16.34:443
Connection error: Invalid SSL certificate
```
Installed version:
```
root@OpenWrt:/# ls -n /usr/bin/wget 
lrwxrwxrwx    1 0        0               18 Feb 17 18:00 /usr/bin/wget -> /bin/uclient-fetch
root@OpenWrt:/# opkg list-installed | grep wolfssl
libustream-wolfssl20201210 - 2022-01-16-868fd881-1
libwolfssl5.1.1.99a5b54a - 5.1.1-stable-1
px5g-wolfssl - 3
root@OpenWrt:/# 
```

curl compiled with this libwolfssl works ok, so it's a problem with libustream-wolfssl20201210 which uclient-fetch is using

```
root@OpenWrt:/# curl -o /dev/null https://letsencrypt.org/
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 33667  100 33667    0     0  18490      0  0:00:01  0:00:01 --:--:-- 20354
root@OpenWrt:/# opkg list-installed | grep curl
curl - 7.80.0-1
libcurl4 - 7.80.0-1
root@OpenWrt:/# opkg info libcurl4
Package: libcurl4
Version: 7.80.0-1
Depends: libc, libwolfssl5.1.1.99a5b54a, libnghttp2-14, ca-bundle
Provides: libcurl
Status: install ok installed
Section: libs
Architecture: x86_64
Size: 159205
Filename: libcurl4_7.80.0-1_x86_64.ipk
Description: A client-side URL transfer library
Installed-Time: 1645294757

```

ynezz, 2022-02-21:
I'm able to reproduce it:

```
docker run -it --rm openwrt/rootfs:x86_64-21.02.2 sh -c "wget https://letsencrypt.org"
Downloading 'https://letsencrypt.org'
Connecting to 18.159.128.50:443
Connection error: Invalid SSL certificate
```

ynezz, 2022-02-21:
It's not present in 21.02.1:

```
docker run -it --rm openwrt/rootfs:x86_64-21.02.1 sh -c "wget https://letsencrypt.org"
Downloading 'https://letsencrypt.org'
Connecting to 3.125.16.34:443
Writing to 'index.html'
index.html           100% |*******************************| 33667   0:00:00 ETA
Download completed (33667 bytes)
```

csrutil, 2022-02-22:
IPv6 is ok

```
root@main:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.2'
DISTRIB_REVISION='r16495-bf0c965af0'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 21.02.2 r16495-bf0c965af0'
DISTRIB_TAINTS=''
root@main:~# wget -O /dev/null https://letsencrypt.org/
--2022-02-22 05:07:21--  https://letsencrypt.org/
Resolving letsencrypt.org... 2400:6180:0:d1::650:7001, 2406:da18:880:3801:52c7:4593:210d:6aae, 138.68.235.89, ...
Connecting to letsencrypt.org|2400:6180:0:d1::650:7001|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 33667 (33K) [text/html]
Saving to: '/dev/null'

/dev/null                100%[==================================>]  32.88K  --.-KB/s    in 0.1s

2022-02-22 05:07:22 (226 KB/s) - '/dev/null' saved [33667/33667]
```

obsy, 2022-02-22:
But this output from "normal" wget, right? Not uclient-fetch.

csrutil, 2022-02-22:
yes

```bash
root@main:~# /usr/bin/wget --version
GNU Wget 1.21.1 built on linux-gnu.

-cares +digest -gpgme +https +ipv6 -iri +large-file -metalink -nls
+ntlm +opie -psl +ssl/openssl

Wgetrc:
    /etc/wgetrc (system)
```

obsy, 2022-02-22:
Curl and "normal" wget works, yes. Problem is with build-in uclient-fetch, based on wolfssl (by default, wget is link to uclient-fetch), not normal wget.

ynezz, 2022-02-22:
The [culprit seems to be](https://gitlab.com/openwrt/project/ustream-ssl/-/blob/master/ustream-openssl.c#L271) the [API breakage]() introduced in 5.1.1-stable:

```
        res = SSL_get_verify_result(ssl);
	if (res != X509_V_OK) {
		if (us->notify_verify_error)
			us->notify_verify_error(us, res, X509_verify_cert_error_string(res));
		return;
	}
```

Where in 4.8.1 it returns `X509_V_OK` but in 5.1.1 it returns some uknown error `39`.

I've tried to call `SSL_get_peer_certificate()` before `SSL_get_verify_result()` and it seems to work as expected, but need to check if the behavior is same with OpenSSL backend as well.

commit b9251e3b (2022-02-22): wolfssl: fix API breakage of SSL_get_verify_result

Backport fix for API breakage of SSL_get_verify_result() introduced in
v5.1.1-stable.  In v4.8.1-stable SSL_get_verify_result() used to return
X509_V_OK when used on LE powered sites or other sites utilizing
relaxed/alternative cert chain validation feature. After an update to
v5.1.1-stable that API calls started returning X509_V_ERR_INVALID_CA
error and thus rendered all such connection attempts imposible:

 $ docker run -it openwrt/rootfs:x86_64-21.02.2 sh -c "wget https://letsencrypt.org"
 Downloading 'https://letsencrypt.org'
 Connecting  [... cut]
commit f49eec63 (2022-02-22): wolfssl: fix API breakage of SSL_get_verify_result

Backport fix for API breakage of SSL_get_verify_result() introduced in
v5.1.1-stable.  In v4.8.1-stable SSL_get_verify_result() used to return
X509_V_OK when used on LE powered sites or other sites utilizing
relaxed/alternative cert chain validation feature. After an update to
v5.1.1-stable that API calls started returning X509_V_ERR_INVALID_CA
error and thus rendered all such connection attempts imposible:

 $ docker run -it openwrt/rootfs:x86_64-21.02.2 sh -c "wget https://letsencrypt.org"
 Downloading 'https://letsencrypt.org'
 Connecting  [... cut]
closed 2022-02-22 by jow-