openwrt/openwrt #9284: broken rt5350 ethernet after ralink reset controller etc patchset

aireagoir8, 2022-02-19:
I have a patchset for a rt5350 router with 8mb/16mb flash and 64mb ram. So far it works well on the 21.02.1 and earlier branches but for the master branch ethernet does not work at all.

A while ago six patches were added for the ralink ethernet drivers:

https://git.openwrt.org/?p=openwrt%2Fopenwrt.git&a=search&h=HEAD&st=commit&s=ramips%3A+ethernet%3A+ralink

To get ethernet working again I reversed the last three patches:

2021-06-25	Alexander Couzens	ramips: ethernet: ralink: rewrite esw_rt3050 to support...
2021-06-25	Alexander Couzens	ramips: ethernet: ralink: allow to return EPROBE_DEFER...
2021-06-25	Alexander Couzens	ramips: ethernet: ralink: move reset of the esw into...

The problem patch makes use of the reset controller to separately reset the frame engine and the switch. Previously the fe and esw reset occurred at the same time:

fe_reset(RT305X_RESET_FE | RT305X_RESET_ESW);

I found the following comment in the [original ethernet patchset](https://git.openwrt.org/?p=openwrt/svn-archive/archive.git;a=blob;f=target/linux/mediatek/patches/0063-arm-mediatek-add-SDK-ethernet.patch;h=7263bb21816b36733b561b749de589d4b010d718;hb=d8804dd309b0a1701c8ebe0a26d700745df24f11) from long ago:
// RT5350 need to reset ESW and FE at the same to avoid PDMA panic //	

Serial console log:
```[    0.000000] Linux version 5.10.100 (alan@debian1) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r18771-ae75541594) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Mon Feb 14 13:40:34 2022
[    0.000000] SoC Type: Ralink RT5350 id:1 rev:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001964c (MIPS 24KEc)
[    0.000000] MIPS: machine is Proroute H820 (8M)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 16kB, 4-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=0007b230
[    0.000000] Readback ErrCtl register=0007b230
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 54460K/65536K available (4037K kernel code, 582K rwdata, 1096K rodata, 4516K init, 200K bss, 11076K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x3bc/0x588 with crng_init=0
[    0.000000] CPU Clock: 360MHz
[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
[    0.000000] systick: running - mult: 214748, shift: 32
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 10618113593 ns
[    0.000019] sched_clock: 32 bits at 180MHz, resolution 5ns, wraps every 11930464253ns
[    0.015762] Calibrating delay loop... 239.61 BogoMIPS (lpj=1198080)
[    0.098098] pid_max: default: 32768 minimum: 301
[    0.107604] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.122121] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.141848] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.162467] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.182053] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.195621] pinctrl core: initialized pinctrl subsystem
[    0.210136] NET: Registered protocol family 16
[    0.354300] rt2880_gpio 10000600.gpio: registering 22 gpios
[    0.365495] rt2880_gpio 10000600.gpio: registering 22 irq handlers
[    0.395631] clocksource: Switched to clocksource MIPS
[    0.408481] NET: Registered protocol family 2
[    0.417589] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.433773] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.450496] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.465733] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.479722] TCP: Hash tables configured (established 1024 bind 1024)
[    0.492758] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.505852] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[... 157 lines cut ...]
[  107.743189] [<8002237c>] warn_slowpath_fmt+0x74/0x94
[  107.753212] [<802d7c84>] dev_watchdog+0x124/0x1a4
[  107.762728] [<8005bf34>] call_timer_fn.constprop.0+0x1c/0x80
[  107.774118] [<8005c234>] run_timer_softirq+0x29c/0x328
[  107.784496] [<803f08cc>] __do_softirq+0xf4/0x268
[  107.793813] [<80002838>] handle_int+0x138/0x144
[  107.802940] [<800026c0>] __r4k_wait+0x20/0x40
[  107.811707]
[  107.814690] ---[ end trace 2763870877d15361 ]---
[  107.823980] mtk_soc_eth 10100000.ethernet eth0: transmit timed out
[  107.836395] mtk_soc_eth 10100000.ethernet eth0: dma_cfg:00000057
[  107.848469] mtk_soc_eth 10100000.ethernet eth0: tx_ring=0, base=018cc000, max=1024, ctx=16, dtx=0, fdx=0, next=16
[  107.869024] mtk_soc_eth 10100000.ethernet eth0: rx_ring=0, base=02080000, max=1024, calc=293, drx=295
[  118.155682] mtk_soc_eth 10100000.ethernet eth0: transmit timed out
[  118.168118] mtk_soc_eth 10100000.ethernet eth0: dma_cfg:00000057
[  118.180214] mtk_soc_eth 10100000.ethernet eth0: tx_ring=0, base=00f00000, max=1024, ctx=18, dtx=0, fdx=0, next=18
[  118.200774] mtk_soc_eth 10100000.ethernet eth0: rx_ring=0, base=026ec000, max=1024, calc=28, drx=29
[  128.155680] mtk_soc_eth 10100000.ethernet eth0: transmit timed out
[  128.168127] mtk_soc_eth 10100000.ethernet eth0: dma_cfg:00000057
[  128.180223] mtk_soc_eth 10100000.ethernet eth0: tx_ring=0, base=00f00000, max=1024, ctx=21, dtx=0, fdx=0, next=21
[  128.200784] mtk_soc_eth 10100000.ethernet eth0: rx_ring=0, base=018cc000, max=1024, calc=21, drx=22
[  138.155682] mtk_soc_eth 10100000.ethernet eth0: transmit timed out
[  138.168116] mtk_soc_eth 10100000.ethernet eth0: dma_cfg:00000057
[  138.180211] mtk_soc_eth 10100000.ethernet eth0: tx_ring=0, base=026ec000, max=1024, ctx=8, dtx=0, fdx=0, next=8
[  138.200426] mtk_soc_eth 10100000.ethernet eth0: rx_ring=0, base=01890000, max=1024, calc=20, drx=21```

lynxis, 2022-02-21:
Hi @aireagoir8,

thanks for the good bug report.

I tested the commit with an asus rt n56u / rt3662 (and other mt76xx devices). It seems the rt305x is a little bit different from an rt305x.
I'm unsure if I've a device to test around.
So far I remember the dma problems only appear if there are packets which the other unit want to deliver. But the rt5350 could be different.

I would like to reproduce it on my table. If you have time you could try to add an additional reset to the new code using the reset api.

aireagoir8, 2022-02-21:
The following patch solved the issue as expected:
```
diff --git a/target/linux/ramips/files/drivers/net/ethernet/ralink/esw_rt3050.c b/target/linux/ramips/files/drivers/net/ethernet/ralink/esw_rt3050.c
index 75b6707be2..a75145d8a1 100644
--- a/target/linux/ramips/files/drivers/net/ethernet/ralink/esw_rt3050.c
+++ b/target/linux/ramips/files/drivers/net/ethernet/ralink/esw_rt3050.c
@@ -261,11 +261,14 @@ static inline void esw_rmw_raw(struct rt305x_esw *esw, unsigned reg,
 
 static void esw_reset(struct rt305x_esw *esw)
 {
-       if (!esw->rst_esw)
+       if (!esw->rst_esw || !esw->priv || !esw->priv->rst_fe)
                return;
 
+       /* RT5350 need to reset ESW and FE at the same to avoid PDMA panic */
+        reset_control_assert(esw->priv->rst_fe);
        reset_control_assert(esw->rst_esw);
        usleep_range(60, 120);
+        reset_control_deassert(esw->priv->rst_fe);
        reset_control_deassert(esw->rst_esw);
        /* the esw takes long to reset otherwise the board hang */
        msleep(10);
```
Is it necessary to validate esw->priv before esw->priv->rst_fe like I did?

The fe_reset_fe() function in mtk_eth_soc.c is still resetting only the fe. What if anything should be done about this?

I tested with debug serial console output and the fe_reset_fe() call occurs just before the esw code resets both at the same time.

aireagoir8, 2022-02-23:
@lynxis,

As I said on the OpenWrt forums, I'm not much of a programmer these days but I'm willing to learn.

Sadly the issue occurred when the device was powered on the first time today. It may have randomly occurred because fe_reset_fe() is still doing the separated frame engine reset.

lynxis, 2022-03-08:
@aireagoir8 You're resetting the frame engine. But you should also tell the frame engine driver about the reset.
I don't know, but it might also possible to pause the frame engine when resetting the esw to prevent a pdma panic and vice versa.

I forgot the asus rt n56u doesn't have the esw. it uses an external. So I don't have any board to test the pdma panic.

lynxis, 2022-03-25:
pepe2k gave me a test setup. I might able to fix this in the next week.

lynxis, 2022-07-31:
Not yet. I've tried out to synchronize the DMA engine with the ethernet switch. But it didn't worked. Even I found a sources which doesn't reset both units at the same. I'll implement now a reset for both at the same time.

lynxis, 2022-09-05:
Can somebody test this commit? https://github.com/lynxis/openwrt/commit/209612ceef34b8ef7d88fb5909305a98fa327da4

lynxis, 2022-09-09:
@badwater does everything else work? is the ethernet and switch working?

mans0n, 2022-09-12:
I rebased your two patches on top of the latest master, and the boot hangs on my device...

```
[    0.000000] Linux version 5.15.67 (localhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r20614-c27279dc26) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 Mon Sep 12 00:43:49 2022
[    0.000000] SoC Type: Ralink RT5350 id:1 rev:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001964c (MIPS 24KEc)
[    0.000000] MIPS: machine is Olimex RT5350F-OLinuXino-EVB
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 16kB, 4-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 8128
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000000] Writing ErrCtl register=00001f81
[    0.000000] Readback ErrCtl register=00001f81
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 25880K/32768K available (3977K kernel code, 562K rwdata, 596K rodata, 1192K init, 197K bss, 6888K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 360MHz
[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
[    0.000000] systick: running - mult: 214748, shift: 32
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 10618113593 ns
[    0.000004] sched_clock: 32 bits at 180MHz, resolution 5ns, wraps every 11930464253ns
[    0.015858] Calibrating delay loop... 239.61 BogoMIPS (lpj=1198080)
[    0.098206] pid_max: default: 32768 minimum: 301
[    0.107805] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.122376] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.143067] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.164799] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.184424] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.198017] pinctrl core: initialized pinctrl subsystem
[    0.210868] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.255002] rt2880_gpio 10000600.gpio: registering 22 gpios
[    0.266254] rt2880_gpio 10000600.gpio: registering 22 irq handlers
[    0.283764] clocksource: Switched to clocksource MIPS
[    0.296194] NET: Registered PF_INET protocol family
[    0.306548] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.322564] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.339327] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.354735] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.369983] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.383993] TCP: Hash tables configured (established 1024 bind 1024)
[    0.396957] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.410083] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.424725] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.436639] rt-timer 10000100.timer: maximum frequency is 3662Hz
[    0.459673] workingset: timestamp_bits=30 max_order=13 bucket_order=0
[    0.480822] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.492457] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.517807] gpio-export gpio_export: 3 gpio(s) exported
[    0.529856] Serial: 8250/16550 driver, 2 ports, IRQ sharing disabled
[    0.545233] 10000500.uart: ttyS1 at MMIO 0x10000500 (irq = 13, base_baud = 2500000) is a Palmchip BK-3103
[    0.566368] 10000c00.uartlite: ttyS0 at MMIO 0x10000c00 (irq = 20, base_baud = 2500000) is a Palmchip BK-3103
[    0.586241] printk: console [ttyS0] enabled
[    0.586241] printk: console [ttyS0] enabled
[    0.602834] printk: bootconsole [early0] disabled
[    0.602834] printk: bootconsole [early0] disabled
[    0.667930] spi spi0.0: force spi mode3
[    0.678701] spi-nor spi0.0: s25fl064k (8192 Kbytes)
[    0.688770] 4 fixed-partitions partitions found on MTD device spi0.0
[    0.701642] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
[    0.716412] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
[    0.732045] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
[    0.746882] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
[    0.762160] Creating 4 MTD partitions on "spi0.0":
[    0.771943] 0x000000000000-0x000000030000 : "u-boot"
[    0.790170] 0x000000030000-0x000000040000 : "u-boot-env"
[    0.803629] 0x000000040000-0x000000050000 : "factory"
[    0.821160] 0x000000050000-0x000000800000 : "firmware"
[... cut ...]
[    0.856869] 0x000000000000-0x00000018e61f : "kernel"
[    0.866873] mtd: partition "kernel" doesn't end on an erase/write block -- force read-only
[    0.890562] 0x00000018e61f-0x0000007b0000 : "rootfs"
[    0.900657] mtd: partition "rootfs" doesn't start on an erase/write block boundary -- force read-only
[    0.921731] mtd: device 5 (rootfs) set to be root filesystem
[    0.935784] 1 squashfs-split partitions found on MTD device rootfs
[    0.948332] 0x00000048d000-0x0000007b0000 : "rootfs_data"
## hangs here
```

Calchan, 2023-03-09:
In case that helps, I have what seems to be the same issue on a brand new Archer A6 v3 running 22.03.3:

```Wed Mar  8 20:09:27 2023 kern.warn kernel: [   48.138868] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:467 0x805602d8
Wed Mar  8 20:09:27 2023 kern.info kernel: [   48.145926] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
Wed Mar  8 20:09:27 2023 kern.warn kernel: [   48.152871] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
Wed Mar  8 20:09:27 2023 kern.warn kernel: [   48.216876] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.10.161 #0```

My router is simply connected to my Comcast modem in bridge mode, replacing a WDN3800 on OpenWRT which worked reliably for years before dying. The only changes I do to the default config are WiFi settings, a couple port-forwards and Google DDNS. On first boot, it worked correctly for a day or so, then the issue above showed up and I lost WAN. I couldn't find a way to fix it so I rebooted the router. After that it would fail again every time within minutes or even seconds of starting up.

I have downgraded to 21.02.5 and that's working so far. Let me know if I need to file a separate ticket.

Leo-PL, 2023-12-08:
I think one of issues I'm facing with ZTE MF283+ (RT3352) is the same as here. Just a heads-up.
I posted my logs on the forum: https://forum.openwrt.org/t/rt3352-ethernet-woes-after-22-03-release/180073

Linaro1985, 2023-12-08:
Hi @Leo-PL . Maybe this patch https://github.com/Linaro1985/openwrt/commit/60934cc12cf2946c8e22850b3fcc1887e63e392a resolves the issue.

Leo-PL, 2023-12-08:
Works for me. I cherry-picked it on top of current main, and did a series of reboots, booted flawlessly and connecting every time. Same is true for 23.05 branch.

This solution looks cleaner to me than initial proposal - but does essentially the same. Maybe you could try opening a PR with it?

Linaro1985, 2023-12-08:
I don't have enough free time. Maybe someone takes this as a basis and does PR. Personally, I only tested with MT7628. Now we have a test for RT3552. For PR it needs to be tested on the devices with a similar problem and another cpu. For a example, rt5350.

Leo-PL, 2023-12-10:
@Linaro1985 one question: what was the reason to increase reset duration in https://github.com/Linaro1985/openwrt/commit/60934cc12cf2946c8e22850b3fcc1887e63e392a#diff-3a1aa078d5f54e96288b5f87a21a051d9286f986e72e77a1356fa42b81cdc75fR152

I'd like to split the commit up into separate logical changes for upstreaming, so I'd like to know the rationale of this one. Changes in device tree are obvious - this is the crux of our problem, and the `devm_reset_control_array_get_optional_exclusive` changes are just refactoring, so they can be split out as well.

Linaro1985, 2023-12-10:
@Linaro1985 because of reset needs more time. Hardware hangs at boot without this change.

Leo-PL, 2023-12-10:
Okay, so this was the missing part from Lynxis' patches. Thanks!

Leo-PL, 2023-12-10:
I've put the cleaned-up commit here: https://github.com/Leo-PL/openwrt/commits/rt305x_ethernet_fixes - please take a look. @Linaro1985 please take a look if it's OK to submit this as a PR.

Linaro1985, 2023-12-10:
> please take a look if it's OK to submit this as a PR.

@Leo-PL I like it. Thank you very much!

commit fc92fecf (2024-01-02): ramips: dts: rt5350: reset FE and ESW cores together

Failing to do so will cause the DMA engine to not initialize properly
and fail to forward packets between them, and in some cases will cause
spurious transmission with size exceeding allowed packet size, causing a
kernel panic.

This is behaviour of downstream driver as well, however I
haven't observed bug reports about this SoC in the wild, so this
commit's purpose is to align this chip with all other SoC's - MT7620
were already using this arrangement.

Fixes: #9284
Fixes: 60fadae62b64 ("ramips: ethernet: ralink: move reset of the esw into [... cut]
PR #14194 (2024-01-02): ramips: fix "no traffic" problem and DMA kernel panic at startup on rt305x and mt7628 devices
This series fixes bootup and Ethernet support on rt305x subtarget. It combines the MAC core and switch core reset again, but using device tree properties, rather than reverting the changes introduced in 60fadae62b64b14faff818e4156d9c6eb3f96b65. However, reverting those changes or moving resets around in device tree wasn't enough - the wait duration after the reset needs increasing, to allow both cores to sync up after reset.

The final commit cleans up the way reset controller lines are accessed from  [... cut]
commit ee4a0424 (2024-01-05): ramips: dts: rt5350: reset FE and ESW cores together

Failing to do so will cause the DMA engine to not initialize properly
and fail to forward packets between them, and in some cases will cause
spurious transmission with size exceeding allowed packet size, causing a
kernel panic.

This is behaviour of downstream driver as well, however I
haven't observed bug reports about this SoC in the wild, so this
commit's purpose is to align this chip with all other SoC's - MT7620
were already using this arrangement.

Fixes: #9284
Fixes: 60fadae62b64 ("ramips: ethernet: ralink: move reset of the esw into [... cut]
commit b80c17b0 (2024-01-06): ramips: dts: rt5350: reset FE and ESW cores together

Failing to do so will cause the DMA engine to not initialize properly
and fail to forward packets between them, and in some cases will cause
spurious transmission with size exceeding allowed packet size, causing a
kernel panic.

This is behaviour of downstream driver as well, however I
haven't observed bug reports about this SoC in the wild, so this
commit's purpose is to align this chip with all other SoC's - MT7620
were already using this arrangement.

Fixes: #9284
Fixes: 60fadae62b64 ("ramips: ethernet: ralink: move reset of the esw into [... cut]
closed 2024-01-02 by openwrt-bot