openwrt/openwrt #9287: building a cmake project failed

jian-yu, 2022-02-20:
Makefile
```include $(TOPDIR)/rules.mk

PKG_NAME:=libigdgmm
PKG_VERSION:=22.0.2
PKG_RELEASE:=1

PKG_SOURCE_URL:=https://github.com/intel/gmmlib.git
PKG_SOURCE_PROTO:=git
PKG_SOURCE_VERSION:=9290546f493656d80a4aa05b63dfeaf9f8527b0a
PKG_MIRROR_HASH:=

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/cmake.mk 

define Package/$(PKG_NAME)
  SECTION:=Applications
  CATEGORY:=Intel VA-API
  TITLE:=Intel(R) Graphics Memory Management Library
  DEPENDS:=+libpthread
  URL:=https://github.com/intel/gmmlib.git
endef

define Package/$(PKG_NAME)/description
	The Intel(R) Graphics Memory Management Library provides device specific and buffer management for the Intel(R) Graphics Compute Runtime for OpenCL(TM) and the Intel(R) Media Driver for VAAPI.
endef

CMAKE_INSTALL:=1

CMAKE_OPTIONS += \
                -DCMAKE_BUILD_TYPE=Release \
                -DARCH=64 \

define Package/$(PKG_NAME)/install
	$(INSTALL_DIR) $(1)/usr/lib/
	$(CP) $(PKG_BUILD_DIR)/Source/Gmmlib/libigdgmm.so*  $(1)/usr/lib/
endef

$(eval $(call BuildPackage,$(PKG_NAME)))
```

error log:
```
MOCS table: Static
CMake Error at /work/nas/orico/linux-5.10.100/openwrt/staging_dir/host/share/cmake-3.22/Modules/CheckIncludeFile.cmake:57 (configure_file):
  configure_file attempted to configure a file:
  /work/nas/orico/linux-5.10.100/openwrt/build_dir/target-x86_64_glibc/libigdgmm-22.0.2/CMakeFiles/CMakeTmp/CheckIncludeFile.c
  into a source directory.
Call Stack (most recent call first):
  /work/nas/orico/linux-5.10.100/openwrt/staging_dir/host/share/cmake-3.22/Modules/FindThreads.cmake:146 (CHECK_INCLUDE_FILE)
  Source/GmmLib/CMakeLists.txt:442 (find_package)
-- Configuring incomplete, errors occurred!
```

error code location and line:
```
437 else()
438	      set_target_properties(${GMM_LIB_DLL_NAME} PROPERTIES OUTPUT_NAME "igdgmm")
439	      set_target_properties(${GMM_LIB_DLL_NAME} PROPERTIES VERSION 
 440    ${GMMLIB_API_MAJOR_VERSION}.${GMMLIB_API_MINOR_VERSION}.${GMMLIB_API_PATCH_VERSION})
441	      set_target_properties(${GMM_LIB_DLL_NAME} PROPERTIES SOVERSION ${GMMLIB_API_MAJOR_VERSION})
442        set(THREADS_PREFER_PTHREAD_FLAG ON)
443        find_package(Threads REQUIRED)
444        target_link_libraries(${GMM_LIB_DLL_NAME} Threads::Threads)
445
446 endif()
```

jian-yu, 2022-02-20:
it seem cmake can't found the Threads lib, so how can i fix Makefile?

jow-, 2022-02-20:
Please use the development section in the forum for cross compilation help.

closed 2022-02-20 by jow-