openwrt/openwrt #9293: 21.02.1: Enable vlan filtering in LuCI for wireless devices in bridge

PinkFreud, 2022-02-20:
I have a configuration where two openwrt devices are connected via 5 GHz radio, using WDS.  This connection passes several vlans (one pvid and untagged, the rest tagged).  It was something of a nightmare to configure in 19.07, but 21.02 resolves most of those with increased vlan support in LuCI.  The one place I'm missing a configuration option is via the vlan filtering page for the bridge itself: I can see both eth0 and eth1 (the device in question is a Netgear WNDR3800, but that shouldn't matter here), but the client wireless interface never appears on this page, despite being a member of the bridge:
[image]

[image]

I can set up filtering properly via cli, by installing the `ip-bridge` package and using the `bridge` util to enable filtering for the wireless interface:
```
# bridge vlan add vid 1 dev wlan1-trunk01 pvid untagged
# bridge vlan add vid 2 dev wlan1-trunk01 tagged
# bridge vlan show dev wlan1-trunk01
port              vlan-id  
wlan1-trunk01     1 PVID Egress Untagged
                  2
```

This works, and lets me set up VAPs using vlan interfaces off the bridge, like br-MW_Trunk.2 - but the downside is I either have to do it manually, or write a hotplug script to manage it whenever the trunked interface comes up.

Would it be possible to enable vlan filtering on wireless interfaces in LuCI instead?  This entire config could then be managed from LuCI, with no external hacks like hotplug scripts.  Thanks!

jow-, 2022-02-20:
Wireless interfaces are filtered intentionally in the LuCI view as the underlying uci configuration treats the specially. You can only indirectly attach wireless devices to bridge VLANs through `config interface`. 

Create a dummy interface, set its proto to unmanaged and its device to `br-MW_Trunk.1`. Then attach the wireless SSID to this dummy interface. It'll inherit the bridge-vlan membership.

closed 2022-02-20 by jow-

PinkFreud, 2022-02-21:
> Create a dummy interface, set its proto to unmanaged and its device to br-MW_Trunk.1. Then attach the wireless SSID to this dummy interface. It'll inherit the bridge-vlan membership.

I don't think that'll work in this case - br-MW_Trunk.1 is vlan 1 (the untagged vlan), while the wireless interface is the untagged vlan 1 + several tagged vlans.  What mechanism would allow the tagged vlans to be passed back to the bridge?

Attaching the wireless interface to the trunk bridge (br-MW_Trunk) doesn't work without vlan filtering, as the bridge doesn't know what to do with the vlan tags coming from the wireless interface.

jow-, 2022-02-21:
Wireless interfaces do not support tagged traffic, not sure where "+ several tagged vlans" would come from there? Netifd has special heursitics. If a wireless wifi-iface is atteched to an interface which is referencing a bridge-vlan-device (such as br-MW_Trunk.1), netifd will put the wireless netdev into the underlying bridge device (br-BW_Trunk.1), and will bridge VLAN 1 as PVID to the wireless netdev.

PinkFreud, 2022-02-21:
> Wireless interfaces do not support tagged traffic, not sure where "+ several tagged vlans" would come from there?

They do indeed support tagged traffic.  As I explained above, I have this working, and all I needed to do was to enable vlan filtering on the wireless interface.

Believe it or not, I've also had this working in 19.07 and earlier (I think I wrote my initial scripts for Barrier Breaker), though it took some convoluted scripts to achieve.

I'd be happy to provide detailed setup instructions if you'd like!

jow-, 2022-02-21:
The please show the exact configuration you're using. Ordinary wireless interfaces do not support tagged traffic. Maybe you're talking about AP/VLAN devices.

PinkFreud, 2022-02-22:
Certainly!

My configuration consists of three wireless APs, configured to be little more than wireless bridges - I have no need for routing on these.  One of these, a WNDR4300, is connected to a managed switch.  The other two, both WNDR3800s, are satellites, and not connected via ethernet cable - instead, they use their 5 GHz radio to connect via WDS to the WNDR4300.  This interface is, as I explained above, trunked.

```
                  WNDR3800 #1 -> wireless clients
                     /
Switch -> WNDR4300 -> wireless clients
                     \
                  WNDR3800 #2 -> wireless clients
```

Managed switch config for port connected to WNDR4300:
```
interface 0/2
description 'cablefree-4300'
switchport mode trunk
switchport trunk allowed vlan 1-254,256-4093
```

This part doesn't really matter, except to convince you that I am, indeed, passing vlans from the managed switch to the satellite WNDR3800s via wireless.  You can achieve the same thing by, say, using a Raspberry Pi configured for vlans connected to an AP running openwrt if you don't have a managed switch available.

Trunked bridge config on WNDR4300:
```
br-MW_trunk		7fff.4a5b9e71ef52	no		eth0
							wlan1
							wlan1.sta1
							wlan1.sta2
```

The WNDR4300 is running 19.07.4, and this is currently the oldest openwrt release in use here (one of the WNDR3800s has been updated to openwrt 19.07.7, and the other, of course, is now running 21.02.1).  I was never able to get vlan interfaces working directly off the bridges prior to 21.02.1 (though I don't remember the last time I tried - I've pretty much migrated this config from openwrt release to openwrt release with minor modifications as necessary, until completely re-doing it with 21.02.1!).  To show how I split the vlans on the WNDR4300, here's the lan bridge:
```
br-lan		7fff.4a5b9e71ef52	no		eth0.2
							wlan0
							wlan1-1
							wlan1-3
							wlan1.sta1.2
							wlan1.sta2.2
```

The wlan1.sta<x> interfaces are the WDS interfaces for both satellite WNDR3800s.

On the WNDR3800 running openwrt 21.02.1, I've been able to simplify this config quite a bit:
```
br-MW_Trunk		7fff.04a151bb3d45	no		eth0
							eth1
							wlan1-trunk01
```
On this device, `wlan1-trunk01` is the custom interface name I've assigned to the trunked wireless interface.  Secondary bridges utilizing vlans are much simpler to set up here:
```
bridge name	bridge id		STP enabled	interfaces
br-lan		7fff.02a151bb3d46	no		wlan0-1
							br-MW_Trunk.2
							wlan0-4
							wlan0-2
							wlan1-5
							wlan1
```

As I mentioned above, the wireless link passes multiple vlans.  I'm specifically only showing the trunked interface (which includes untagged vlan 1) and trunked vlan 2.  The other vlans are configured similarly, with a bridge containing MW_Trunk.<vlan #> and one or more wifi interfaces.  I do have several VAPs linked to the lan bridge - I have a couple on both the 2.4 GHz and 5 GHz radios, and I've started switching to WPA3 where possible - but some clients (*ahem*, Raspberry Pi Foundation...) don't play nicely on WPA3 or mixed WPA2/WPA3 as of yet, so I still have some legacy WPA2 VAPs configured here as well.

Most of my interfaces on the satellite APs are set up as Unmanaged.  There are two exceptions: vlan 1 and 2.  vlan 1 is used for management, and is unrouted outside the local network.  vlan 2 is the lan network, and is indeed routed outside of the network via NAT:
```
br-MW_Trunk.1@br-MW_Trunk: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 04:a1:51:bb:3d:45 brd ff:ff:ff:ff:ff:ff
    inet 10.42.12.90/24 brd 10.42.12.255 scope global br-MW_Trunk.1

br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 02:a1:51:bb:3d:46 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.90/24 brd 192.168.1.255 scope global br-lan
[... 69 lines cut ...]
                  3
                  8
                  9
                  10
                  11
                  12
wlan0-1           1 PVID Egress Untagged
wlan0-2           1 PVID Egress Untagged
wlan0-3           1 PVID Egress Untagged
wlan0-4           1 PVID Egress Untagged
wlan1             1 PVID Egress Untagged
wlan1-1           1 PVID Egress Untagged
wlan1-2           1 PVID Egress Untagged
wlan1-3           1 PVID Egress Untagged
wlan1-4           1 PVID Egress Untagged
wlan1-5           1 PVID Egress Untagged
```

vlan filtering for physical interfaces eth0 and eth1, as well as master bridge interface br-MW_Trunk, are handled neatly within LuCI.  vlan filtering for wlan1-trunk01 (the third member of br-MW_Trunk, along with eth0 + eth1) is not - I had to configure these via hotplug script, since LuCI currently filters out wireless interfaces from being eligible for vlan filtering.  It wasn't until I manually configured vlan filtering on this wireless interface that I was able to get the bridge to pass trunked traffic coming from wlan1-trunk01 to the other interfaces on the WNDR3800 running openwrt 21.02.1.  Before I did so, I was able to run tcpdump to see trunked traffic on the wireless interface, but not on the bridge or other bridge members.

Of course, this brings us to the purpose of this bug report: To ask that we be able to manage vlan filtering on wireless interfaces within LuCI.  :)

I've written this from my perspective - I know how my network is set up, and it's entirely possible I glazed over some things, or missed covering something that seems obvious to me.  If you have any further questions about this setup, please feel free to ask!  My goal is to convince you that this does indeed work, and that enabling vlan filtering on wireless interfaces in LuCI is worthwhile.

Thanks again.

jow-, 2022-02-22:
> Of course, this brings us to the purpose of this bug report: To ask that we be able to manage vlan filtering on wireless interfaces within LuCI. :)

As written above, this will not happen since the underlying uci configuration/netifd cpmfiguration model will not allow that. The only supported configuration is attaching wifi-ifaces (which might result in multiple wifi devices) indirectly to target bridges through logical interfaces utilizing either the target bridge device or a vlan on top of the target bridge device, in which case the wifi device(s) are enslaved into the target bridge and PVID is inherited from the bridge VLAN.

jow-, 2022-02-22:
@nbd168 - maybe you have some opinions here as this touches netifd's wireless<>network integration model

PinkFreud, 2022-02-23:
@jow- Can we re-open this issue?

Thanks!

jow-, 2022-02-23:
I won't reopen it as I see no way to implement this given the current constraints of the system. If some other developer thinks it is possible, he might reopen it.