openwrt/openwrt #9295: qosify: dscp os always overridden even if '+' is specified

tiagogaspar8, 2022-02-20:
Hi!

I've been trying to set up qosify but I ran into an issue where qosify always overrides the dscp, even if it was set on source and has a '+' specified.

I have found an issue where it doesn't classify ipv6 UDP packets and is dnsmas required? It isn't clear on the dependencies and on the readme file, and not providing an alternative for people that use for example odhcpd + unbound.

Thanks!

Qosifyck, 2022-02-25:
Please @nbd168, implement in Qosify that the "DNS hostname" (dns:) feature can be used by people who use odhcpd + unbound, because that feature ONLY works with dnsmasq (which is garbage).

[image]

Qosify, 2022-02-27:
Please @nbd168, you can add an option similar to Adblock where you can choose which DNS backend to use depending on whether you have dnsmasq or [odhcpd + Unbound](https://openwrt.org/docs/guide-user/base-system/dhcp_configuration#replacing_dnsmasq_with_odhcpd_and_unbound) installed for Adblock to work.

[image]

Please, you can add that same or similar option in Qosify to make it work in Unbound.

tiagogaspar8, 2022-03-21:
Hi @nbd168 

I don't know if you see this yet I'll record it here.
I've seen your changes in qosify and I've updated it to the latest version.
I'm glad to say the IPv6 UDP issue is gone.

I've seen you changed the way DNS is handled by snooping packets. I'm using DoT so that might be the reason it doesn't work.

Also, the '+' that should allow to specify not overriding a dscp if existent still doesn't work as intended.

Great work and thanks for everything!