openwrt/openwrt #9298: hostapd: handle_probe_req: send failed

ptpt52, 2022-02-21:
on latest master code https://github.com/openwrt/openwrt/commit/cbfce9236754700a343632fff8e035acbc1b1384
AP + STA never work on 5G wifi

mt76 driver tested on mt7915 or mt7612

config: /etc/config/wireless
```
config wifi-device 'radio0'
	option type 'mac80211'
	option band '2g'
	option htmode 'HE20'
	option phy 'phy0'
	option hwmode '11g'
	option disabled '0'
	option country 'CN'
	option channel 'auto'
	option cell_density '0'
	option txpower '20'

config wifi-device 'radio1'
	option type 'mac80211'
	option band '5g'
	option htmode 'HE80'
	option phy 'phy1'
	option hwmode '11a'
	option disabled '0'
	option country 'CN'
	option cell_density '0'
	option channel '44'

config wifi-iface 'wifinet0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'X-WRT_68FE'
	option encryption 'psk2'
	option skip_inactivity_poll '1'
	option wpa_group_rekey '0'
	option wpa_pair_rekey '0'
	option wpa_master_rekey '0'
	option disassoc_low_ack '0'
	option key '88888888'
	option ieee80211r '1'
	option ft_over_ds '1'
	option ft_psk_generate_local '1'

config wifi-iface 'wifinet1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option encryption 'psk2'
	option skip_inactivity_poll '1'
	option wpa_group_rekey '0'
	option wpa_pair_rekey '0'
	option wpa_master_rekey '0'
	option disassoc_low_ack '0'
	option key '88888888'
	option ieee80211r '1'
	option ft_over_ds '1'
	option ft_psk_generate_local '1'
	option ssid 'X-WRT_68FE_5G'

config wifi-iface 'wifinet2'
	option device 'radio1'
	option mode 'sta'
	option network 'wwan'
	option ssid 'Huawei_Wi-Fi5'
	option encryption 'psk2'
	option key '12345678'

```

log:
```
Mon Feb 21 19:29:18 2022 daemon.notice hostapd: wlan1-1: DFS-CAC-COMPLETED success=1 freq=5220 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5210 cf2=0
Mon Feb 21 19:29:18 2022 daemon.notice hostapd: wlan1-1: interface state DFS->ENABLED
Mon Feb 21 19:29:18 2022 daemon.notice hostapd: wlan1-1: AP-ENABLED
Mon Feb 21 19:29:20 2022 daemon.notice hostapd: wlan1-1: DFS-PRE-CAC-EXPIRED freq=5180 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5180 cf2=0
Mon Feb 21 19:29:20 2022 daemon.notice hostapd: wlan1-1: DFS-PRE-CAC-EXPIRED freq=5200 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5200 cf2=0
Mon Feb 21 19:29:20 2022 daemon.notice hostapd: wlan1-1: DFS-PRE-CAC-EXPIRED freq=5220 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5220 cf2=0
Mon Feb 21 19:29:20 2022 daemon.notice hostapd: wlan1-1: DFS-PRE-CAC-EXPIRED freq=5240 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5240 cf2=0
Mon Feb 21 19:34:35 2022 daemon.notice hostapd: handle_probe_req: send failed
Mon Feb 21 19:34:37 2022 daemon.notice hostapd: handle_probe_req: send failed
Mon Feb 21 19:34:40 2022 daemon.notice hostapd: handle_probe_req: send failed
Mon Feb 21 19:34:43 2022 daemon.notice hostapd: handle_probe_req: send failed
Mon Feb 21 19:34:55 2022 daemon.notice hostapd: handle_probe_req: send failed
Mon Feb 21 19:35:00 2022 daemon.notice hostapd: handle_probe_req: send failed
Mon Feb 21 19:35:18 2022 daemon.notice hostapd: handle_probe_req: send failed
Mon Feb 21 19:35:43 2022 daemon.notice hostapd: handle_probe_req: send failed
Mon Feb 21 19:35:44 2022 daemon.notice hostapd: handle_probe_req: send failed
Mon Feb 21 19:35:44 2022 daemon.notice hostapd: handle_probe_req: send failed

```

ptpt52, 2022-02-21:
but if the upstream router use channel 149, it seems work.

work:
client-AP+STA-------huawei-AP(channel 149)

not work:
client-AP+STA-------huawei-AP(channel 44)

ptpt52, 2022-02-21:
In the version a month ago, channel 44 was working fine, I am sure about this.

ynezz, 2022-02-21:
> In the version a month ago, channel 44 was working fine, I am sure about this.
That's a good news, now you just need to find out the offending commit 
yourself (git bisect-ing it). Once we know which commit broke it, it's 
much easier to fix it.

DazzyWalkman, 2022-02-22:
I come across similar problem, 2 mt7621 wr1200js, one as master, the other as STA+AP.   The master ap also failed to operate on the problematic build.
snapshot build 5G client-AP+STA working: 5a0975f7ef2561dd8bbc29e8a9a327f28e641a37
snapshot build 5G AP not working: cbfce92367
I take a look at the short log between them, 
Commits may be of interests:
2 about hostapd
e6df13d0e16dd394ca26a80feeb013234948e875
6f787239771044ed7eeaf22301c543b699f25cb4

and a couple of less suspecious ramips/realtek/wireless-regdb related ones

I will take 6f787239771044ed7eeaf22301c543b699f25cb4 as the first before-and-after starting point when I have time.

ptpt52, 2022-02-22:
I just do revert back to this https://github.com/openwrt/openwrt/commit/2a31e9ca975d99b6216b2979a825f07253c8947e
for hostapd
and it works

ptpt52, 2022-02-22:
ok, I found that this commit (https://github.com/openwrt/openwrt/commit/46e0eeb760c9d1da38660f9a02119a558d403625) cause the issue.

revert this single commit and it works again.

ping @nbd168 
maybe nbd168 know what/how it is

DazzyWalkman, 2022-02-24:
Yes. @ptpt52 may have found the offending commit.
And during my git bisect, I found that 19a90262df89f8419b058cf9a00dc9e8d465088b greatly manifested this issue.
I am using a similar wireless config with @ptpt52,  I don't experience problem on my device with 46e0eeb760c9d1da38660f9a02119a558d403625 applied, however, 19a90262df89f8419b058cf9a00dc9e8d465088b makes my 5g master ap inop and my log flooded with:

```Mon Feb 21 19:33:45 2022 daemon.notice wpa_supplicant[1646]: wlan1: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon Feb 21 19:33:45 2022 daemon.notice hostapd: wlan1-1: DFS-CAC-COMPLETED success=1 freq=5180 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5210 cf2=0
Mon Feb 21 19:33:47 2022 daemon.notice wpa_supplicant[1646]: wlan1: Unknown event 54
Mon Feb 21 19:33:47 2022 daemon.notice hostapd: wlan1-1: DFS-PRE-CAC-EXPIRED freq=5180 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5180 cf2=0
...
Wed Feb 23 17:26:54 2022 daemon.notice hostapd: handle_probe_req: send failed
...
Thu Feb 24 09:59:25 2022 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Thu Feb 24 09:59:25 2022 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
```
and on the problematic builds, iwinfo reads something like below:
````         Access Point: D6:5F:25:FC:XX:YY
          Mode: Master  Channel: 36 (5.180 GHz)
          Center Channel 1: unknown 2: unknown
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
````
I revert 19a90262df89f8419b058cf9a00dc9e8d465088b on the latest snapshot build (namely b9251e3b407592f3114e739231088c3d27663c4c), the issue is gone. 
Something do with hostapd channel selection?

brianjmurrell, 2022-12-14:
So last comment was almost a year ago.  Seeing this on latest 22.03.2.

Any progress on a fix?

trashhalo, 2022-12-17:
Having the same issue on 22.03.1

ptpt52, 2022-12-18:
scan wifi and then connect, and set the fixed-channel fix the issue.

brianjmurrell, 2022-12-19:
@ptpt52 Your suggestion, as much as it's appreciated (and thank-you for it) as a workaround, is still a work-around at best.  It should not be so cumbersome to simply connect to a WiFi network in STA mode.

Could one imagine having to tell one's grandma that in order to connect her phone to a WiFi network, first she has to scan it and then once she has figured out the channel of AP she wants to connect to she then has to go set a fixed channel connection?  And then if/when the AP changes channels because it's set to 'auto' (and perhaps it restarts), she has to go do that all over again.

I'm not actually even sure how I would do a fixed-channel configuration with [travelmate](https://github.com/openwrt/packages/blob/master/net/travelmate/files/README.mdhttps://github.com/openwrt/packages/blob/master/net/travelmate/files/README.md).

So my [previous question/comment](https://github.com/openwrt/openwrt/issues/9298#issuecomment-1352321204) was really more about the progress of finding a proper solution to this problem, rather than how can one jury rig around it.

dac2020, 2023-06-21:
Openwrt 22.03.3 still doing this. Any solution to this problem?

yulinbanshi, 2023-06-21:
这是来自鸿霞的假期自动回复邮件。
您好，你的来信已经收到，现无法马上回复您的邮件。我将尽快给您回复。
     谢谢！

mailinglists35, 2023-08-14:
I am constantly getting this message on FC38 with kernel 6.4.9-200.fc38.x86_64 
It works okay in Ubuntu 22.04 with kernel 6.2.0-26-generic amd64

in ubuntu I see this in modinfo, but nonetheless kernel does not print anything firmware-related when loading on neither distro:

````
firmware:       mediatek/mt7610u.bin
firmware:       mediatek/mt7610e.bin
````

mailinglists35, 2023-08-14:
````
$ sudo lsusb -d 0b05:17d1 -v

Bus 001 Device 013: ID 0b05:17d1 ASUSTek Computer, Inc. AC51 802.11a/b/g/n/ac Wireless Adapter [Mediatek MT7610U]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x17d1 AC51 802.11a/b/g/n/ac Wireless Adapter [Mediatek MT7610U]
  bcdDevice            1.00
  iManufacturer           1 MediaTek
  iProduct                2 WiFi
  iSerial                 3 1.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x004a
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              160mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           8
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      2
      bInterfaceProtocol    255
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x07  EP 7 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x09  EP 9 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x000c
  bNumDeviceCaps          1
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
Device Status:     0x0000
  (Bus Powered)
````

mailinglists35, 2023-08-14:
hostapd.conf

````
interface=wlp0s21f0u3
ssid=test
auth_algs=1
wpa=2
wpa_passphrase=tttttttt
wpa_key_mgmt=WPA-PSK-SHA256
wpa_pairwise=CCMP
country_code=RO
ieee80211d=1
ieee80211n=1
ieee80211ac=1
ieee80211h=1
hw_mode=a
channel=36
bridge=tor0
ht_capab=[HT40+]
wmm_enabled=1
logger_stdout_level=1
````

SmallAndSoft, 2024-09-10:
I am getting this error on OpenWrt 23.05.4 TP-Link Archer C7 v5 straight after a reboot.
2.4 Ghz wifi is not working because of this.
5 Ghz wifi works fine.

I have both AP and Client configured for 2.4Ghz

psparc82, 2024-09-23:
> I am getting this error on OpenWrt 23.05.4 TP-Link Archer C7 v5 straight after a reboot. 2.4 Ghz wifi is not working because of this. 5 Ghz wifi works fine.
> 
> I have both AP and Client configured for 2.4Ghz

did you find a fix?

SmallAndSoft, 2024-09-24:
> > I am getting this error on OpenWrt 23.05.4 TP-Link Archer C7 v5 straight after a reboot. 2.4 Ghz wifi is not working because of this. 5 Ghz wifi works fine.
> > I have both AP and Client configured for 2.4Ghz
> 
> did you find a fix?

No, only a work around. Manual stop-start of the interface configured as client.

grosjo, 2024-10-03:
Same problem here. Need to restart the wifi periodically

danpawlik, 2024-10-04:
Could someone set interface into [debug](https://openwrt.org/docs/guide-developer/debugging) mode and put the logs here?
e.g.:
```
uci set wireless.default_radio0.log_level=1
uci set wireless.default_radio1.log_level=1
uci commit wireless
reboot
```

TingyiChen, 2024-10-26:
Running a client and host ap on 5G radio, and all of my devices cannot connect to 5G ap.
I see hostapd spamming about errors. Can you check it? @danpawlik 
Device: Xiaomi Router AX3000T
OpenWrt version: OpenWrt 23.05.5 r24106-10cc5fcd00 / LuCI openwrt-23.05 branch git-24.264.56413-c7a3562
Logs after setting debug parameters:
`Sat Oct 26 02:53:07 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:53:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:53:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:53:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:53:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:53:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:53:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:53:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:53:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:53:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:53:13 2024 authpriv.info dropbear[5018]: Child connection from 192.168.1.212:50802
Sat Oct 26 02:53:13 2024 authpriv.notice dropbear[5018]: Password auth succeeded for 'root' from 192.168.1.212:50802
Sat Oct 26 02:53:13 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:53:19 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:53:25 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:53:26 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:53:26 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:53:27 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:53:31 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:53:37 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:53:43 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:53:49 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:53:55 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:54:01 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:54:06 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:54:07 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:54:07 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:54:07 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:54:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:54:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:54:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:54:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:54:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:54:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:54:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:54:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:54:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:54:13 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:54:19 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:54:25 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:54:31 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:54:37 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:54:43 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:54:49 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:54:55 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:55:01 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:55:07 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:55:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:55:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:55:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:55:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:55:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:55:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:55:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:55:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:55:10 2024 daemon.notice hostapd: handle_probe_req: send failed
Sat Oct 26 02:55:13 2024 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 26 02:55:16 2024 daemon.notice hostapd: Set new config for phy phy0:
Sat Oct 26 02:55:16 2024 daemon.notice hostapd: Remove interface 'phy0'
Sat Oct 26 02:55:16 2024 daemon.notice hostapd: phy0-ap0: interface state ENABLED->DISABLED
Sat Oct 26 02:55:16 2024 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED b0:78:39:91:4a:e5
Sat Oct 26 02:55:16 2024 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 54:1f:8d:8f:38:32
Sat Oct 26 02:55:16 2024 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 4c:c6:4c:91:13:bd
Sat Oct 26 02:55:16 2024 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 24:27:30:77:8d:5e
[... cut ...]
Sat Oct 26 02:55:23 2024 daemon.debug hostapd: phy0-ap0: STA 4c:c6:4c:91:13:bd WPA: received EAPOL-Key frame (4/4 Pairwise)
Sat Oct 26 02:55:23 2024 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED 4c:c6:4c:91:13:bd auth_alg=open
Sat Oct 26 02:55:23 2024 daemon.debug hostapd: phy0-ap0: STA 4c:c6:4c:91:13:bd IEEE 802.1X: authorizing port
Sat Oct 26 02:55:23 2024 daemon.info hostapd: phy0-ap0: STA 4c:c6:4c:91:13:bd WPA: pairwise key handshake completed (RSN)
Sat Oct 26 02:55:23 2024 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED 4c:c6:4c:91:13:bd`

Neustradamus, 2024-11-03:
With master code, the problem is always here?
hostapd has been updated...

TingyiChen, 2024-11-04:
I haven't tried snapshot version. Will try this weekend

wryun, 2025-01-20:
I'm seeing this message in the simpler case of AP/STA on the same radio and the STA failing to connect. I believe this is because the STA is still scanning, so hostapd is unable to send beacons.

The interesting thing is that start_disabled should be passed to hostapd when the STA hasn't come up so it doesn't try to send beacons, but it seems like the OpenWrt 23 introduction of ucode broke the logic here. Previously, if there was any STA involved, start_disabled would be set on bring up and when the STA associated it would set the channel and beacons would start. However, now it appears to set start_disabled ONLY if wpa_supplicant is COMPLETED when it comes up:
https://github.com/openwrt/openwrt/blob/42a253c7e8992e2274628b1770a0263ac433d0bc/package/network/services/hostapd/files/hostapd.uc#L198
In all other situations, it starts beaconing immediately:
https://github.com/openwrt/openwrt/blob/42a253c7e8992e2274628b1770a0263ac433d0bc/package/network/services/hostapd/files/hostapd.uc#L211

This seems wrong.