openwrt/openwrt #9305: Kernel error - r18878-a93dfff10e - raspberry pi 4B

AQPSal, 2022-02-21:
`[216151.642146] ------------[ cut here ]------------
[216151.648260] NETDEV WATCHDOG: eth0 (bcmgenet): transmit queue 1 timed out
[216151.656302] WARNING: CPU: 3 PID: 26 at net/sched/sch_generic.c:467 dev_watchdog+0x2c8/0x2d0
[216151.665944] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet brcmfmac wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack libchacha20poly1305 libblake2s iptable_mangle iptable_filter ipt_REJECT ip_tables chacha_neon cfg80211 ax88179_178a xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG x_tables usbnet usbhid slhc sch_cake poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha libblake2s_generic hid_generic crc_ccitt compat brcmutil cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route
[216151.666084]  cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact snd_bcm2835(C) hid evdev ifb ip6_udp_tunnel udp_tunnel snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore cifs nls_utf8 md5 md4 kpp des_generic libdes cmac arc4 vfat fat nls_iso8859_1 nls_cp437
[216151.793480] CPU: 3 PID: 26 Comm: ksoftirqd/3 Tainted: G         C        5.10.100 #0
[216151.802822] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[216151.810249] pstate: 40400005 (nZcv daif +PAN -UAO -TCO BTYPE=--)
[216151.817856] pc : dev_watchdog+0x2c8/0x2d0
[216151.823457] lr : dev_watchdog+0x2c8/0x2d0
[216151.829048] sp : ffffffc010bb3d70
[216151.833932] x29: ffffffc010bb3d70 x28: 0000000000000140 
[216151.840829] x27: 00000000ffffffff x26: 0000000000000000 
[216151.847719] x25: 0000000000000003 x24: 0000000000000000 
[216151.854583] x23: ffffff8040d40480 x22: 0000000000000001 
[216151.861467] x21: ffffffc010a26000 x20: ffffff8040d40000 
[216151.868353] x19: 0000000000000001 x18: 00000000000005d7 
[216151.875238] x17: ffffffc01595e218 x16: ffffffc01595e21c 
[216151.882140] x15: ffffffc010a378d0 x14: 0000000000001185 
[216151.889025] x13: 00000000000005d7 x12: ffffffc010bb3a48 
[216151.895925] x11: ffffffc010a8f8d0 x10: 00000000fffff000 
[216151.902822] x9 : ffffffc010a8f8d0 x8 : 0000000000000000 
[216151.909714] x7 : ffffffc010a378d0 x6 : 0000000000000001 
[216151.916599] x5 : 0000000000000000 x4 : 0000000000000000 
[216151.923487] x3 : 0000000000000000 x2 : ffffff80fb7f2028 
[216151.930371] x1 : ffffffc0eaddb000 x0 : 000000000000003c 
[216151.937251] Call trace:
[216151.941268]  dev_watchdog+0x2c8/0x2d0
[216151.946507]  call_timer_fn.constprop.0+0x24/0x80
[216151.952688]  __run_timers.part.0+0x240/0x280
[216151.958532]  run_timer_softirq+0x3c/0x74
[216151.964032]  _stext+0x124/0x290
[216151.968748]  irq_exit+0xa0/0xf0
[216151.973466]  __handle_domain_irq+0x80/0xe0
[216151.979141]  gic_handle_irq+0x78/0xa0
[216151.984362]  el1_irq+0xcc/0x180
[216151.989082]  finish_task_switch+0x58/0x140
[216151.994753]  __schedule+0x250/0x544
[216151.999806]  _cond_resched+0x3c/0x60
[216152.004950]  run_ksoftirqd+0x60/0x70
[216152.010092]  smpboot_thread_fn+0x154/0x1a0
[216152.015759]  kthread+0x124/0x12c
[216152.020550]  ret_from_fork+0x10/0x34
[216152.025696] ---[ end trace f4d16de4727760d0 ]---`

Kernel error occurred after several checksum errors reported - errors are either ICMP or UDP IPV6 packages - it seems IPV4 is unaffected. I have a suspicion it might have to do with Qosify and DSCP marking  but I am not technically savvy enough to verify that such is the case. (SQM before had some checksum errors as well, usually NTP packages but I no longer have the information - kernel errors did not occur then). 

After error, network is unresponsive. Using the console, and restarting network brings back connectivity.

Below is the immediately preceding chksum error reported (IPV6 address obscured)

`[216150.110639] CPU: 3 PID: 26 Comm: ksoftirqd/3 Tainted: G         C        5.10.100 #0
[216150.119544] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[216150.126541] Call trace:
[216150.130136]  dump_backtrace+0x0/0x15c
[216150.134941]  show_stack+0x18/0x30
[216150.139391]  dump_stack+0xd4/0x110
[216150.143917]  netdev_rx_csum_fault.part.0+0x48/0x58
[216150.149837]  netdev_rx_csum_fault+0x3c/0x40
[216150.155135]  __skb_checksum_complete+0xdc/0xe0
[216150.160690]  nf_ip6_checksum+0xbc/0x160
[216150.165639]  nf_nat_icmpv6_reply_translation+0xa0/0x5b0 [nf_nat]
[216150.172760]  nf_nat_icmpv6_reply_translation+0x298/0x5b0 [nf_nat]
[216150.179964]  nf_nat_icmpv6_reply_translation+0x3fc/0x5b0 [nf_nat]
[216150.187127]  nf_hook_slow+0x4c/0xf0
[216150.191707]  ipv6_rcv+0x64/0x130
[216150.196022]  __netif_receive_skb_one_core+0x48/0x60
[216150.201994]  process_backlog+0xfc/0x250
[216150.206914]  __napi_poll+0x38/0x140
[216150.211479]  net_rx_action+0xe8/0x26c
[216150.216214]  _stext+0x124/0x290
[216150.220420]  run_ksoftirqd+0x54/0x70
[216150.225051]  smpboot_thread_fn+0x154/0x1a0
[216150.230206]  kthread+0x124/0x12c
[216150.234476]  ret_from_fork+0x10/0x34
[216150.239284] eth0: hw csum failure
[216150.243730] skb len=1280 headroom=144 headlen=1280 tailroom=752
[216150.243730] mac=(130,14) net=(144,40) trans=184
[216150.243730] shinfo(txflags=0 nr_frags=0 gso(size=0 type=0 segs=0))
[216150.243730] csum(0xffff72f8 ip_summed=2 complete_sw=0 valid=0 level=0)
[216150.243730] hash(0xf3900165 sw=1 l4=1) proto=0x86dd pkttype=0 iif=2
[216150.278363] dev name=eth0 feat=0x0000010000004829
[216150.284240] skb headroom: 00000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[216150.292987] skb headroom: 00000010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[216150.301817] skb headroom: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[216150.310581] skb headroom: 00000030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[216150.319386] skb headroom: 00000040: 80 7f 50 05 80 80 00 00 8d 0f 00 00 00 00 00 00
[216150.328189] skb headroom: 00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[216150.336972] skb headroom: 00000060: 9f 45 1c e9 bb 5f f9 7e a0 96 18 52 82 4a de 40
[216150.345703] skb headroom: 00000070: c6 d0 30 45 6a 2f 53 e4 f4 ac a8 28 83 d0 9c a4
[216150.354502] skb headroom: 00000080: 00 00 dc a6 32 20 99 54 00 01 5c 68 a6 46 86 dd
[216150.363242] skb linear:   00000000: 60 08 6d 4f 04 d8 3a 6f 28 00 03 f0 40 01 08 24
[216150.372062] skb linear:   00000010: 00 00 00 00 00 00 20 13 XX XX XX XX XX XX XX XX 
[216150.380880] skb linear:   00000020: XX XX XX XX XX XX XX XX 01 04 f6 d6 00 00 00 00
[216150.389685] skb linear:   00000030: 68 08 6d 4f 04 d6 11 30 XX XX XX XX XX XX  XX XX
[216150.398447] skb linear:   00000040: XX XX XX XX XX d9 76 9c 28 00 03 f0 40 01 08 24
[216150.407271] skb linear:   00000050: 00 00 00 00 00 00 20 13 f7 f9 01 bb 04 d6 87 68
[216150.416046] skb linear:   00000060: c1 00 00 00 01 08 02 43 fe d8 5d 30 56 08 00 40
[216150.424874] skb linear:   00000070: 46 00 8e 7d 5b 7e 6d 08 26 25 cf ab 4d d5 c2 12
[216150.433710] skb linear:   00000080: 8c 47 7d ed 59 9d 58 1c 3f 2e 93 dd ef 6d e1 e3
[216150.442558] skb linear:   00000090: 7d a5 c4 9a b1 bd 9b ca df 74 c9 f1 85 b7 06 39
[216150.451419] skb linear:   000000a0: 45 84 06 7c b0 a2 b9 b4 76 3c 52 fe ef 5f a5 e8
[216150.460289] skb linear:   000000b0: cf 61 38 ca 7a 08 16 44 75 67 e8 3c 39 92 d6 f6
[216150.469088] skb linear:   000000c0: aa fe 77 03 cd 61 14 34 d3 68 84 53 a0 cb c8 e7
[216150.477965] skb linear:   000000d0: f9 fc 15 d0 1f 20 4d 9f c0 b7 11 9b 14 7e 2e 27
[216150.486839] skb linear:   000000e0: 49 a4 27 2c b5 f2 21 7e 84 9a 48 1d 7c 3d d8 a3
[216150.495723] skb linear:   000000f0: a2 91 29 ee 71 cd c7 71 ba a1 32 e6 55 ee 91 eb
[216150.504615] skb linear:   00000100: cd de a7 3a cf 42 cc 33 a0 c6 a3 50 58 b7 64 37
[216150.513446] skb linear:   00000110: 9f 51 74 28 dd c1 e8 5e 9b 4f ca 50 40 48 bc d8
[216150.522316] skb linear:   00000120: c1 b4 f1 10 8f e1 ee 65 72 72 81 9c 97 eb 14 74
[216150.531156] skb linear:   00000130: f0 67 aa 8e 2c 69 e7 41 3b 87 95 fd 8a 9b 00 7b
[216150.540071] skb linear:   00000140: a1 d8 b8 4a 5d 65 45 1a d1 1c ed 33 b8 77 1f ec`

Please let me know if you need any further information

wulfy23, 2022-02-22:
https://github.com/wulfy23/rpi4/issues/11

this has been a rare issue encountered by very few (just me?) since the switch to 5.10...

for a workaround... add this to your rc.local or whatever...
(performance impact is negligble)

```
ethtool -K eth0 rx off
```

graysky2, 2022-02-23:
@wulfy23 - Thanks for sharing the work-around.  Seems to be working on my RPi4B that was sensitive the bug.  I read the `ethtool` man page... any impacts to disabling RX checksumming?  You mentioned negligible impacts.  Thanks.

graysky2, 2022-02-23:
@wulfy23 - Seems as though this work-around does not mask the problem on my machine.  Using iperf3 from a PC on the LAN to the RPi4B router triggers it even having checksumming disabled.

```
[19493.462172] ------------[ cut here ]------------
[19493.466813] NETDEV WATCHDOG: eth0 (bcmgenet): transmit queue 0 timed out
[19493.473570] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:467 dev_watchdog+0x2c8/0x2d0
[19493.481825] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet brcmfmac wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack libchacha20poly1305 libblake2s iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables chacha_neon cfg80211 cdc_ncm cdc_ether ax88179_178a xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables usbnet usbhid slhc sch_cake rtl8150 r8152 poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libcrc32c libchacha libblake2s_generic hid_generic crc_ccitt compat
[19493.481949]  brcmutil sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact snd_bcm2835(C) hid evdev ifb ip6_udp_tunnel udp_tunnel veth snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore nls_utf8 vfat fat nls_iso8859_1 nls_cp437
[19493.602463] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G         C        5.10.102 #0
[19493.609937] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[19493.615761] pstate: 60400005 (nZCv daif +PAN -UAO -TCO BTYPE=--)
[19493.621759] pc : dev_watchdog+0x2c8/0x2d0
[19493.625759] lr : dev_watchdog+0x2c8/0x2d0
[19493.629757] sp : ffffffc010b3bd70
[19493.633062] x29: ffffffc010b3bd70 x28: 0000000000000140 
[19493.638366] x27: 00000000ffffffff x26: 0000000000000000 
[19493.643669] x25: 0000000000000002 x24: 0000000000000000 
[19493.648973] x23: ffffff8040c40480 x22: 0000000000000001 
[19493.654276] x21: ffffffc0109b6000 x20: ffffff8040c40000 
[19493.659580] x19: 0000000000000000 x18: 000000000000011e 
[19493.664883] x17: 0000000000000000 x16: 000000000000001e 
[19493.670186] x15: ffffffc0109c77d0 x14: 000000000000035a 
[19493.675489] x13: 000000000000011e x12: ffffffc010b3ba48 
[19493.680793] x11: ffffffc010a1f7d0 x10: 00000000fffff000 
[19493.686096] x9 : ffffffc010a1f7d0 x8 : 0000000000000000 
[19493.691399] x7 : ffffffc0109c77d0 x6 : 0000000000000004 
[19493.696702] x5 : 0000000000000000 x4 : 0000000000000000 
[19493.702006] x3 : ffffffc0109b8f8c x2 : 0000000000000004 
[19493.707309] x1 : 0000000000000005 x0 : 000000000000003c 
[19493.712613] Call trace:
[19493.715051]  dev_watchdog+0x2c8/0x2d0
[19493.718705]  call_timer_fn.constprop.0+0x24/0x80
[19493.723313]  __run_timers.part.0+0x240/0x280
[19493.727573]  run_timer_softirq+0x3c/0x74
[19493.731486]  _stext+0x124/0x290
[19493.734619]  irq_exit+0xa0/0xf0
[19493.737756]  __handle_domain_irq+0x80/0xe0
[19493.741845]  gic_handle_irq+0x78/0xa0
[19493.745497]  el1_irq+0xcc/0x180
[19493.748632]  arch_cpu_idle+0x18/0x30
[19493.752199]  default_idle_call+0x20/0x70
[19493.756116]  do_idle+0x21c/0x280
[19493.759334]  cpu_startup_entry+0x28/0x60
[19493.763248]  secondary_start_kernel+0x128/0x150
[19493.767768] ---[ end trace 11aa64c6c6f35b63 ]---
```

TinkerTurtle, 2022-12-23:
@wulfy23 

I ran into a similar issue, and may have found another workaround that doesn’t require turning off checksums.

Under SQM -> Queue Discipline -> Advanced, I set “Squash DSCP on inbound packets (ingress)” to “DO NOT SQUASH”.

I haven’t tested this long enough to have much confidence, but there haven’t been any checksum errors in the past 7 hours when I would otherwise see many errors.

amarcu5, 2023-01-04:
Thanks @TinkerTurtle. Since updating OpenWRT, I’d been experiencing periodic ~1s latency spikes every 40s only when cake SQM is enabled. The kernel errors eventually led me here and setting “DO NOT SQUASH” seems to have done the trick!