openwrt/openwrt #9307: Activity based LEDs not functional on Asus RT-AC56U

JustAnotherEndUser, 2022-02-22:
Version 21.02 on Asus RT-AC56U
Minor bug.  Only the power LED is functional.  None of the other status or activity based  LEDs light up.  Manually modifying LED settings from admin interface had no effect.  Thank you.

pepe2k, 2022-02-22:
@JustAnotherEndUser is the `kmod-ledtrig-activity` package installed on your device? Could you provide uci configuration for `system` (`usi show system`)?

Cheers,
Piotr

JustAnotherEndUser, 2022-02-25:
@pepe2k  , You are brilliant !   After updating the package list then manually installing kmod-ledtrig-activity package and re-configuring the individual settings for each LED  it is now working as expected. 
 I am new to OpenWRT, only having had installed it on an old router for testing purposes.  I wasn't aware of the need to manually install the Kmod package.  Thank you for replying to the solution.  Also thanks to @aparcar for adding the label to the post.  
 Cheers!

closed 2022-02-25 by JustAnotherEndUser

neheb, 2026-09-11:
@JustAnotherEndUser do you still use this device?