openwrt/openwrt #9312: WDR3500 Stuck at boot after the 21.02.2 sysupgrade from 21.02.1.

aquamacho1, 2022-02-22:
Hello, i have wdr3500 v1 and was trying to install latest not yet officially released 21.02.2 stable build. I was performing update from the Luci web ui with keep the settings option enabled. After reboot router stuck at pre init stage, and stopped loading the firmware with only one led active (power led). Router was not bricked, firmware was flashed, but for some reason router cant boot the firmware with normal restart, only with the reset button pressed upon boot until two arrows appear.

I flashed this .img https://firmware-selector.openwrt.org/?version=21.02.2&target=ath79%2Fgeneric&id=tplink_tl-wdr3500-v1
Version: 21.02.2 (r16495-bf0c965af0)
Date: 2022-02-17 21:48:55

[beeweezer](https://forum.openwrt.org/u/beeweezer) on the openwrt forum had the same ussue, he tried multiple things like:
`I tried a soft reset, a hard reset, flashing 21.02.2 sysupgrade again, flasing a 21.02.2 factory image, all resulted in the same problem.`

Link to his post: https://forum.openwrt.org/t/wdr3500-stuck-on-bootup-after-sysupgrade-from-21-02-1-to-21-02-2/120764

wulfy23, 2022-02-23:
are you cable to connect a USB<>TTL UART dongle (serial) and capture the bootlog after sysupgrade from 21.02.1?

aquamacho1, 2022-02-23:
> are you able to connect a USB<>TTL UART dongle (serial)

  nope, i don`t have a cable, nor i have knowledge how to do this.

> (edit: may as well connect it and flash from ssh commandline so you can capture the flash stage also)

Will SSH capture pre init stages/u-boot logs?

aquamacho1, 2022-02-23:
```
ssh root@192.168.1.2                                                                    ✔  6s  
root@192.168.1.2's password: 

BusyBox v1.33.1 (2021-10-24 09:01:35 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 21.02.1, r16325-88151b8303
 -----------------------------------------------------
root@OpenWrt:~# free
              total        used        free      shared  buff/cache   available
Mem:         123400       24128       78664        7232       20608       60780
Swap:             0           0           0
root@OpenWrt:~# sysupgrade -v /tmp/openwrt-21.02.2-ath79-generic-tplink_tl-wdr3500-v1-squashfs-sysupgrade.bin
Wed Feb 23 12:31:56 UTC 2022 upgrade: Saving config files...
etc/config/adblock
etc/config/adblock-opkg
etc/config/dhcp
etc/config/dropbear
etc/config/firewall
etc/config/luci
etc/config/network
etc/config/rpcd
etc/config/system
etc/config/ubootenv
etc/config/ucitrack
etc/config/uhttpd
etc/config/wireless
etc/dropbear/dropbear_ed25519_host_key
etc/dropbear/dropbear_rsa_host_key
etc/group
etc/hosts
etc/inittab
etc/luci-uploads/.placeholder
etc/opkg/keys/2f8b0b98e08306bf
etc/passwd
etc/profile
etc/rc.local
etc/shadow
etc/shells
etc/shinit
etc/sysctl.conf
etc/uhttpd.crt
etc/uhttpd.key
etc/uhttpd.key
etc/uhttpd.crt
Wed Feb 23 12:31:56 UTC 2022 upgrade: Commencing upgrade. Closing all shell sessions.
Connection to 192.168.1.2 closed by remote host.
Connection to 192.168.1.2 closed.
```

aquamacho1, 2022-02-23:
Kernel log from 21.02.2:
[log.txt](https://github.com/openwrt/openwrt/files/8124412/log.txt)

wulfy23, 2022-02-24:
not useless and your updates to the OP narrow things down significantly

yes... we'll still need someone with console/serial access on one of these devices to further diagnose

aquamacho1, 2022-03-18:
Update from **beeweezer** on openwrt forum 
Link to the full reply: https://forum.openwrt.org/t/wdr3500-stuck-on-bootup-after-sysupgrade-from-21-02-1-to-21-02-2/120764/61

> So to my conclusion the device gets stuck BETWEEN those two lines:
> [ 0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
> [ 0.000000] Writing ErrCtl register=00000000

```
Thu Mar 17 14:50:10 CET 2022 upgrade: Switching to ramdisk...
Thu Mar 17 13:50:13 UTC 2022 upgrade: Performing system upgrade...
[  271.339323] sh (2684): drop_caches: 3
Unlocking firmware ...

Writing from <stdin> to firmware ...
Thu Mar 17 13:50:55 UTC 2022 upgrade: Upgrade completed
Thu Mar 17 13:50:56 UTC 2022 upgrade: Rebooting system...
umount: can't unmount /dev: Resource busy
umount: can't unmount /tmp: Resource [  313.664820] reboot: Restarting system
¦
¦C¦UI¦¦?DB1¦ja¦¦¦¦15j 2¦'¦?¦)
¦¦D¦5?¦¦j      5¦id¦¦¦¦ 0x¦¦¦¦¦¦¦5¦+? si¦¦j¦, sY,¦Kco¦=¦28¦!¦+¦h: ¦ M¦C¦Usin"¦¦¦ult¦¦?¦on¦Y
                                                                                           5

P¦RY¦¦setz-¦¦HHHj
                 In¦¦¦?¦l
¦]¦se¦+¦5
         Err'¦¦ria¦¦Net¦
                        ¦934}¦¦¦¦_inZW+¦¦ze¦?
                                             ¦
                                              ¦¦¦id X?ss ¦
                                                          2¦¦sh.¦U¦¦¦fi¦¦¦¦¦¦s
¦
 ¦¦¦id X?¦s i
             2¦¦sh.¦U¦¦¦fi¦d ¦¦¦¦s
 W¦¦ resYj¦¦k:¦30¦
!'¦g1 ¦¦ c?¦¦¦1¦S¦e
                  ¦¦¦¦Y'V¦ce:¦?¦j
                                 s2¦¦¦inij
                                          athr.¦¦¦y_sYW¦ATH¦!e}CON?¦
¦XK¦27_¦y_¦YW¦ATH?!e}SPE¦¦Q¦S 4¦10¦¦eth0¦¦5                          ¦¦:1¦0
                                          WAS¦jjjj¦ S¦¦!e
'¦¦E 0x¦¦I 0¦j
¦.¦¦¦¦¦initb¦¦¦t+¦¦:be'V¦¦Y'¦?1
¦NT?¦0¦10¦jng ¦¦H¦AT?*¦¦¦s2¦¦¦?t d¦+5¦aK¦¦7_p¦^5¦¦up ¦J¦¦e}
           ¦K¦¦7_p¦^5¦¦up PJ¦¦e}MPEC¦*UUM 0 ¦¦j¦ath¦.¦¦¦¦}¦etu¦AT?¦e_CO?¦1¦1 :L¦¦j
                                                                                  ath¦.¦¦¦¦}¦etu¦AT?¦e_SPQ¦5EUM¦1 :1j
ONT?¦2¦10¦                                                                                                           aK¦¦7_p¦^5¦¦up ¦J¦¦e}
¦K¦¦7_p¦^5¦¦up PJ¦e}MPEC5¦U¦ 2 ¦¦j¦ath¦.¦¦¦¦}¦etu¦AT?¦e_CO¦¦=13 :L¦¦j
                                                                     at¦.¦¦¦¦}¦etu¦ATJ¦A!¦_SPQ¦5EUM¦3 :1j
                                                                                                         ethL¦¦5
                                                                                                                eth¦ e+j¦Au¦K¦¦¦in¦J¦¦1 se¦?¦5
                                                                                                                                              ## B?¦¦ imXV
                                                                                                                                                          ¦ 9f¦00¦rrrj
¦[    0.000¦00¦ printk: bootcon¦ol¦ [early0] ena¦le¦h¦st) (gcc version¦8.¦.0 (OpenWrt G¦C ¦.4.0 r16495-bf0c¦65¦f0)) #0 Wed Feb ¦6 ¦0:29:10 2022
[    0.0¦00¦0] CPU0 revis¦on¦is: 0001974c (MI¦S ¦4Kc)
[   ¦0.¦00000] MIPS: mac¦in¦ is TP-Link T¦-W¦R3500 v1
¦  ¦ 0.000000] So¦: ¦theros AR9344 re¦ 2¦
                                         [    ¦.0¦0000] Initrd not¦fo¦nd or empty -¦di¦abling initrd
[  ¦ 0¦000000] Prima¦y ¦nstruction cache¦64¦B, VIPT, 4-wa¦, ¦inesize 32 bytes¦
[    0.0¦00¦0] Primary data ¦ac¦e 32kB, 4-way¦ V¦PT, cache aliase¦, ¦inesize 32 bytes¦
¦[   ¦0.¦00000]   Norm¦l ¦ [mem 0x00000000¦00¦0000-0x000000¦00¦ffffff]                [    0.0¦00¦0] Zone ranges:
[    0.¦00¦00] Movable z¦ne¦start for each n¦de¦
                                                [    0.00000¦] ¦arly memory node¦ra¦ges
[ ¦  ¦.000000]   node ¦ 0¦ [mem 0x00000¦00¦0000000-0x000000¦00¦ffffff]
[¦   0.000000] Ini¦me¦ setup node 0¦[m¦m 0x000000000000¦00¦-0x0000000007¦ff¦ff]
[¦   0.000000] ¦ui¦t 1 zonelists, m¦bi¦ity grouping on.¦ T¦tal pages: 32¦80¦
¦[ ¦  ¦.000000] Dentry ¦ac¦e hash table ¦nt¦ies: 16384 (orde¦: ¦, 65536 bytes¦ l¦near)000] Kernel comm¦nd¦line: console¦tt¦S0,115200 rootfs¦yp¦=squashfs,jffs2
[    0.0¦00¦0] Inode-cache h¦sh¦table entries¦ 8¦92 (order: 3, 32¦68¦bytes, linear¦
```

nikolarobottesla, 2022-04-21:
I have a WDR3500 v1 and noticed the same issue on 21.02.2  some additional info:
1. It very intermittently does boot normally like 1/30 times on average
2. It can be forced to boot normally by holding the reset button for a few seconds while power is applied (I believe it goes into a recovery mode then times out into the normal boot.)
3. 21.02.3 still has the issue.
4. 21.02.1 did not have the issue.

aquamacho1, 2022-06-21:
@nikolarobottesla Sure.

closed 2022-06-21 by aquamacho1