openwrt/openwrt #9324: cypress-firmware-43455 files need updated for Raspberry Pi 4B

owenh000, 2022-02-24:
With OpenWrt 21.02.1 on the Raspberry Pi 4B, I was unable to scan for and connect to certain 5 GHz channels. For example, I was able to scan for and connect to an 802.11ac wireless network on channel 44; but when I switched that network to channel 153, OpenWrt was not able to detect it in a scan or connect to it. Also, the logs were flooded with `daemon.warn wpa_supplicant[681]: wlan0: Failed to initiate sched scan`.

The problem was fixed after I updated the firmware files in `/lib/firmware/brcm/` using the firmware from the linux-firmware repository:

- <https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/cypress/cyfmac43455-sdio.bin> (renamed to `brcmfmac43455-sdio.bin`)
- <https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/cypress/cyfmac43455-sdio.clm_blob> (renamed to `brcmfmac43455-sdio.clm_blob`)
- <https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt>

I think the file that needs updated is [package/firmware/cypress-firmware/Makefile](https://github.com/openwrt/openwrt/blob/master/package/firmware/cypress-firmware/Makefile).

---

Old, not working:

```console
# opkg status cypress-firmware-43455-sdio | grep Version
Version: v5.4.18-2020_0402-3
# opkg status cypress-nvram-43455-sdio-rpi-4b | grep Version
Version: 2019-09-03-e7b78df2-2
# sha256sum <files>
3eb7d29497280dcd79462c0b45aaebfcc46645a1f9dc5eb64f0531f56bb2eb9a  brcmfmac43455-sdio.bin
c4195e23573d1d4ef3a95a54147ce6dfe57c73885d47db71d461db02b560a8a1  brcmfmac43455-sdio.clm_blob
d90ac57cf01e9334ef1bce30f2cb00460ce910d8417ff7ba6cfa460212bd7573  brcmfmac43455-sdio.raspberrypi,4-model-b.txt
# dmesg | grep 'Firmware: BCM4345/6'
[    7.656938] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar 23 2020 02:20:01 version 7.45.206 (r725000 CY) FWID 01-febaba43
```

New, working:

```
# sha256sum <files>
d408faa9d0d5b1a2f9912dcea53ab0be48217288e398406d117f0edafe7c3edd  brcmfmac43455-sdio.bin
15f50a27020b263d1bea215c8f68d0550d912932d1d9ef19ffd59f18d82dd460  brcmfmac43455-sdio.clm_blob
edb6f4e4fb19e18940004124feb4ffe160d72fc607243a07a4480338a28b2748  brcmfmac43455-sdio.raspberrypi,4-model-b.txt
# dmesg | grep 'Firmware: BCM4345/6'
[162004.725511] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
```

See also: <https://openwrt.org/toh/raspberry_pi_foundation/raspberry_pi#updating_the_wifi_firmware>

Thank you for your work on OpenWrt.

owenh000, 2022-03-03:
There is no change in OpenWrt 21.02.2; it still has the old firmware files.

annunity, 2022-03-13:
Hi, 

Big newbie here. 

Thanks for the guide. I am a bit stuck on fixing the wifi for 21.02.2. Does adding the 3 files fix the problem? Do I need to remove the old files? Do file permissions make a difference? I am hoping to use the pi as a wifi access point.

I did a file comparison.
btc_params50=0x972c was added for the .txt
I wasn't able to compare the bin and blob files. 

Cheers

onedayfishsale, 2022-03-13:
@annunity here's what I did:
```
root@OpenWrt:~# mkdir ~/firmware_backup
root@OpenWrt:~# cp -p /lib/firmware/brcm/brcmfmac43455* ~/firmware_backup
root@OpenWrt:~# wget --no-check-certificate -O /lib/firmware/brcm/brcmfmac43455-sdio.bin https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/cypress/cyfmac43455-sdio.bin
root@OpenWrt:~# wget --no-check-certificate -O /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/cypress/cyfmac43455-sdio.clm_blob
root@OpenWrt:~# wget --no-check-certificate -O /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.tx
root@OpenWrt:~# reboot
```

You can compare checksums by using the sha256sum command:

```
root@OpenWrt:~# sha256sum ~/firmware_backup/brcmfmac43455-sdio.bin /lib/firmware/brcm/brcmfmac-43455-sdio.bin
3eb7d29497280dcd79462c0b45aaebfcc46645a1f9dc5eb64f0531f56bb2eb9a  ~/firmware_backup/brcmfmac43455-sdio.bin
d408faa9d0d5b1a2f9912dcea53ab0be48217288e398406d117f0edafe7c3edd  /lib/firmware/brcm/brcmfmac43455-sdio.bin
root@OpenWrt:~# 
```

A different checksum indicates the files are different.

BKPepe, 2022-03-16:
> The problem was fixed after I updated the firmware files in `/lib/firmware/brcm/` using the firmware from the linux-firmware repository:
>
> * [...]

I think in your issue, you are mixing two things, which are different. 
If you would like to use firmware files from the ``linux-firmware``, Broadcom is located in this path: ``openwrt/package/firmware/linux-firmware/broadcom mk`` and they are shipped in the different packages. They begin with prefix ``brcmfmac``.

So in your case, it would be enough to replace ``brcmfmac-firmware-43455-sdio-rpi-4b`` by ``cypress-nvram-43455-sdio-rpi-4b``.

And yes, those files as you correctly mentioned were renamed, and that was backported to OpenWrt 21.02, which I did in this commit: https://github.com/openwrt/openwrt/commit/d0b0ebf966a412ddc2b25dffb5eff7effbcd0220

I did not look closely at why these firmware files are redistributed twice if talking about ``cypress-firmware``, these days by Infineon company. Why they are different, why they could be better/worse and just avoid this discussiong for this time. Instead, I will take a look at how it can be updated to fix most known Wi-Fi vulnerabilities such as **Kr00k** and **FragAttacks**.  _Also, another solution would be to switch to linux-firmware packages in default RPi installations since cypress-firmware is not very well maintained here._

Btw, any pull requests, as always are welcomed instead of creating such threads as:
- https://forum.openwrt.org/t/outdated-cypress-infineon-brcmfmac-firmware-for-raspberry-pi-4-in-openwrt-21-02-1/117427
- https://community.infineon.com/t5/Wi-Fi-Bluetooth-for-Linux/Outdated-brcmfmac-firmware-for-Raspberry-Pi-4-in-OpenWrt-21-02-1/td-p/331547

Overall, not helpful to anyone.

owenh000, 2022-03-17:
@BKPepe, thanks for opening pull request #9483 to fix this. And yes, it seems like it would be better to just use from linux-firmware by default for the Raspberry Pis.

BKPepe, 2022-03-26:
All my PRs were merged and are included in OpenWrt 21.02 and newer developments versions. So, you should be safe. :+1: 

It is going to be included in OpenWrt 21.02.3.

This can be closed.

geekscrapy, 2022-11-11:
I have this same issue with Pi3B+
`wlan0: Failed to initiate sched scan`

I presume I need to update my firmware too, however I can't really do any testing as the pi is very remote...

owenh000, 2023-10-27:
As [posted](https://github.com/openwrt/openwrt/issues/9324#issuecomment-1079770745) above, this is fixed in OpenWrt 21.02.3. I have tested and can confirm this is fixed in OpenWrt 23.05.0. Thanks, @BKPepe!

closed 2023-10-27 by owenh000