openwrt/openwrt #9330: TP-Link AC1750 Version 2 -- 21.02.01 disables wireless

tmatthews5511, 2022-02-25:
I tried upgrading from 21.02.0 to 21.02.1 on my TP-Link 1750 Version 2. (TP-Link Archer C7 v2). 

After the upgrade, I could not active the wireless (which was working fine in 21.02.0).

Solution: I had to downgrade back to 21.02.0 from 21.02.1. With 21.02.0 the wireless (radio0 and radio1) both are working fine. 

Wireless: ath10k_pci

Summary: upgrading to 21.02.1 on the TP-Link Archer C7 v2 killed the wireless. It was totally impossible to reactivate.

ne20002, 2022-02-25:
I have a similar finding: wireless is disableb **if** you have more than one ssid on an interface.

I'm running WRt3200ACM and updated with attended sysupgrade.

The system.log states:
```Fri Feb 25 16:52:34 2022 daemon.notice hostapd: Remove interface 'wlan0'
Fri Feb 25 16:52:34 2022 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Fri Feb 25 16:52:34 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 8e:a9:dc:54:f8:1f
Fri Feb 25 16:52:34 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Fri Feb 25 16:52:34 2022 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Fri Feb 25 16:52:34 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Fri Feb 25 16:52:34 2022 kern.info kernel: [ 5244.882492] device wlan0 left promiscuous mode
Fri Feb 25 16:52:34 2022 kern.info kernel: [ 5244.887013] br-switch: port 5(wlan0) entered disabled state
Fri Feb 25 16:52:34 2022 kern.debug kernel: [ 5244.958642] ieee80211 phy0: staid 1 deleted
Fri Feb 25 16:52:34 2022 kern.debug kernel: [ 5244.976536] ieee80211 phy0: change: 0x40
Fri Feb 25 16:52:34 2022 kern.debug kernel: [ 5245.077498] ieee80211 phy0: change: 0x100
Fri Feb 25 16:52:34 2022 daemon.notice netifd: Network device 'wlan0' link is down
Fri Feb 25 16:52:34 2022 daemon.notice netifd: Wireless device 'radio0' is now down
Fri Feb 25 16:52:34 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Fri Feb 25 16:52:34 2022 daemon.err hostapd: Duplicate BSSID 24:f5:a2:2d:7c:c2 on interface 'wlan0-1' and 'wlan0'.
Fri Feb 25 16:52:34 2022 daemon.err hostapd: 1 errors found in configuration file '/var/run/hostapd-phy0.conf'
Fri Feb 25 16:52:34 2022 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy0.conf
Fri Feb 25 16:52:34 2022 daemon.notice netifd: radio0 (29903): Command failed: Invalid argument
Fri Feb 25 16:52:34 2022 daemon.notice netifd: radio0 (29903): Device setup failed: HOSTAPD_START_FAILED
Fri Feb 25 16:52:34 2022 daemon.notice netifd: Wireless device 'radio0' set retry=0
Fri Feb 25 16:52:34 2022 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Fri Feb 25 16:52:34 2022 daemon.notice netifd: Wireless device 'radio0' is now down
```
Having only one SSID on an interface 2.4G or 5G works. If I enable a second one, the device gets deactivated. If I disable all except one, device gets activated again.

ynezz, 2022-02-26:
@tmatthews5511 provide `/etc/config/wireless` and relevant log output `logread | grep -e netifd -e hostapd`

tmatthews5511, 2022-02-26:
I did yet another upgrade to 21.02.01 but this time the wireless decided to work with no drama. 

Hopefully the next upgrade will work with far less stress.

aparcar, 2022-02-27:
Can this be closed?

tmatthews5511, 2022-02-27:
Yes. please close. It seems solved for now at least.

closed 2022-02-27 by aparcar