openwrt/openwrt #9343: 21.02.2: wireless gets deactivated when assigning more than one SSID

ne20002, 2022-02-25:
I have a similar finding to : wireless is disabled if you have more than one ssid on an interface on 21.02.2 (it worked for me with 21.01.1).

I'm running WRT3200ACM and updated with 'Attended SysUpgrade' (which worked fine so far).

The system.log states:

Fri Feb 25 16:52:34 2022 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Fri Feb 25 16:52:34 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 8e:a9:dc:54:f8:1f
Fri Feb 25 16:52:34 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Fri Feb 25 16:52:34 2022 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Fri Feb 25 16:52:34 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Fri Feb 25 16:52:34 2022 kern.info kernel: [ 5244.882492] device wlan0 left promiscuous mode
Fri Feb 25 16:52:34 2022 kern.info kernel: [ 5244.887013] br-switch: port 5(wlan0) entered disabled state
Fri Feb 25 16:52:34 2022 kern.debug kernel: [ 5244.958642] ieee80211 phy0: staid 1 deleted
Fri Feb 25 16:52:34 2022 kern.debug kernel: [ 5244.976536] ieee80211 phy0: change: 0x40
Fri Feb 25 16:52:34 2022 kern.debug kernel: [ 5245.077498] ieee80211 phy0: change: 0x100
Fri Feb 25 16:52:34 2022 daemon.notice netifd: Network device 'wlan0' link is down
Fri Feb 25 16:52:34 2022 daemon.notice netifd: Wireless device 'radio0' is now down
Fri Feb 25 16:52:34 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Fri Feb 25 16:52:34 2022 daemon.err hostapd: Duplicate BSSID 24:f5:a2:2d:7c:c2 on interface 'wlan0-1' and 'wlan0'.
Fri Feb 25 16:52:34 2022 daemon.err hostapd: 1 errors found in configuration file '/var/run/hostapd-phy0.conf'
Fri Feb 25 16:52:34 2022 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy0.conf
Fri Feb 25 16:52:34 2022 daemon.notice netifd: radio0 (29903): Command failed: Invalid argument
Fri Feb 25 16:52:34 2022 daemon.notice netifd: radio0 (29903): Device setup failed: HOSTAPD_START_FAILED
Fri Feb 25 16:52:34 2022 daemon.notice netifd: Wireless device 'radio0' set retry=0
Fri Feb 25 16:52:34 2022 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Fri Feb 25 16:52:34 2022 daemon.notice netifd: Wireless device 'radio0' is now down

Having only one SSID on an interface 2.4G or 5G works. If I enable a second one, the device gets deactivated. If I disable all except one, device gets activated again.
wireless in /etc/config has only one BSSID per wlan interface.

anomeome, 2022-02-25:
[assign BSSID](https://github.com/openwrt/openwrt/issues/7019) to each, some reporting removing all _option macaddr_ also worked.

wurzerj, 2022-02-26:
Removing all option macaddr definitely worked in my case.

ynezz, 2022-02-26:
Can someone please check if reverting 39c8bc4422 fixes this?

ynezz, 2022-02-26:
> wireless is disabled if you have more than one ssid on an interface on 21.02.2 (it worked for me with 21.01.1).

Are you really sure? I can't see any change, which would introduce this issue in 21.02.2.

ynezz, 2022-02-26:
Could you please share your `/etc/config/wireless` ?

ne20002, 2022-02-26:
Removing all macaddr worked.
thx

closed 2022-02-26 by ne20002

wurzerj, 2022-02-26:
> @ne20002 How do you remove all macaddr in this case?

Simply remove all lines containing “option macaddr” from /etc/config/wireless

rnhmjoj, 2022-03-01:
> Command failed: Invalid argument

The source of this seems to be this line: https://github.com/openwrt/openwrt/blob/cbfce9236754700a343632fff8e035acbc1b1384/package/network/services/hostapd/src/src/ap/ubus.c#L730

Looking at the hostapd [sources](https://github.com/digsrc/wpa_supplicant/blob/515eb37dd1df3f4a05fc2a31c265db6358301988/src/ap/hostapd.c#L2295), there are at least 10 possible points of failure, so it's pretty hard to tell what's going on without a debugger.

MaxRower, 2022-03-07:
When I deleted all macaddr entries, it worked for the first two interfaces, but then the third interface (Marvell 88W8887 802.11nac) did use a mac adress of the second SSID of first one (Marvell 88W8964 802.11nac). So deleting all macaddr entries is not the final solution.

dbrand666, 2022-03-23:
> Interestingly if I change the `/lib/netifd/wireless/mac80211.sh` script to retry adding the interface immediately after it works:

This did not work for me. Removing all the macaddr entries did.

tim-hub, 2023-02-06:
Has this been resolved? I got this issue at 21.03.03

rnhmjoj, 2023-02-07:
No, the issue is closed but the problem was never fixed.

LaurenceGough, 2023-02-08:
Issue is not fixed.

EvanCarroll, 2023-02-27:
I just had this. Removed option macaddr. Seems to have resolved it for me.

```text
[ 6982.818815] ieee80211 phy0: change: 0x40
[ 6983.058848] ieee80211 phy0: change: 0x42
[ 6985.775996] ieee80211 phy0: change: 0x40
```
forever.... all gone. Will report if it returns.

benr916, 2025-04-14:
I think I am running into this issue while trying to add another SSID. I have a thread posted here about it on the OpenWRT forum ([LINK](https://forum.openwrt.org/t/having-an-issue-adding-ssids/229982)).

I have tried the workaround that most report working, editing the /etc/config/wireless file and removing all of the "option macaddr 'XX:XX:XX:XX:XX:XX'" lines.
However, as soon as I reboot they get put back (with different MAC addresses). Removing them also does not fix the issue before rebooting either.
I have confirmed I saved the changes by reopening the wireless file (using vi) and making sure the changes stayed before rebooting.

benr916, 2025-04-14:
Quick update, got em working! SSH'd in and edited the /etc/config/wireless file to remove all the "option macaddr 'XX:XX:XX:XX:XX:XX'" lines then I just needed to follow that up with an:
/etc/init.d/network restart
To commit the changes (per these instructions [LINK](https://openwrt.org/docs/guide-user/base-system/uci)).

tapper82, 2025-04-15:
Does this still happen in a newer build of OpenWrt? 21.xx OpenWrt is end of life now.

benr916, 2025-04-15:
> Does this still happen in a newer build of OpenWrt? 21.xx OpenWrt is end of life now.

Not sure about that. I had the issue in a GL.iNet GL-X3000 running the latest firmware (as of 4/15/25) 4.0 and OpenWrt v21.02.