openwrt/openwrt #9345: ipq806x:  bootcount: line 5: /lib/upgrade/asrock.sh: No such file or directory

okibcn, 2022-02-26:
When building the image for Netgear R7800 I am getting this error:

```
Finalizing root filesystem...
/home/oki/kk/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/etc/init.d/bootcount: line 5: /lib/upgrade/asrock.sh: No such file or directory
Enabling boot
./etc/init.d/bootcount: line 5: /lib/upgrade/asrock.sh: No such file or directory
Enabling bootcount
```

It is not happening for other targets, as I have tested the image for the Netgear R6020 (ramips)

at [https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg58508.html](https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg58508.html) there is a proposed modification of bootcount that could solve the issue:

```
 START=99
 
-. /lib/upgrade/asrock.sh
+. "$IPKG_INSTROOT/lib/upgrade/asrock.sh"
 
 boot() {
        case $(board_name) in
        asrock,g10)
-               asrock_bootconfig_mangle "bootcheck"
-               if [ $? -eq 0 ]; then
-                       reboot
-               fi
+               asrock_bootconfig_mangle "bootcheck" && reboot
                ;;
```

ynezz, 2022-02-26:
Should be fixed, thank you for reporting it.

closed 2022-02-26 by ynezz

okibcn, 2022-02-26:
The error persists after the fix was applied, as the script /lib/upgrade/asrock.sh successfully executed now by bootcount has the same issue as bootcount script. No need to reopen this issue as I have created issue #9350 to take care of the problem inside /lib/upgrade/asrock.sh

commit 610b2cff (2022-02-26): ipq806x: base-files: asrock: fix bootcount include

Fixes following error while executing the init script on the buildhost:

 Enabling boot
 ./etc/init.d/bootcount: line 5: /lib/upgrade/asrock.sh: No such file or directory
 Enabling bootcount

While at it fix following shellcheck issue:

 base-files/etc/init.d/bootcount line 11:
    if [ $? -eq 0 ]; then
         ^-- SC2181: Check exit code directly with e.g. 'if mycmd;', not indirectly with $?.

Fixes: #9345
Cc: Ansuel Smith <ansuelsmth@gmail.com>
Cc: Pawel Dembicki <paweldembicki@gmail.com>
Cc: Christian Lamparter <chunkeey@gmail.com>
Fixes: [... cut]