openwrt/openwrt #9352: F2FS_MINSIZE needs finetuning (x86 squashfs overlay switches between f2fs and ext4)

d12r-dev, 2022-02-27:
The FTFS_MINSIZE threshold is currently set at 100ULL * 1024ULL * 1024ULL, i.e. 100 MB, which happens to be _right_ on the edge of where X86 squashfs-combined-efi builds land for the overlay.

As a result, 21.02.1 builds have been using f2fs:

```
 OpenWrt 21.02.1, r16325-88151b8303

root@OpenWrt:~# mount | grep overlay
/dev/loop0 on /overlay type f2fs (rw,lazytime,noatime,background_gc=on,discard,no_heap,user_xattr,inline_xattr,inline_data,inline_dentry,flush_merge,extent_cache,mode=adaptive,active_logs=6,alloc_mode=reuse,fsync_mode=posix)
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/loop0               98.1M     58.2M     39.8M  59% /overlay
overlayfs:/overlay       98.1M     58.2M     39.8M  59% /

```
and 21.02.2 builds are using ext4:
```
 OpenWrt 21.02.2, r16495-bf0c965af0

root@OpenWrt:~# mount | grep overlay
/dev/loop0 on /overlay type ext4 (rw,noatime)
[...]
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/loop0               92.8M      1.5M     84.3M   2% /overlay
overlayfs:/overlay       92.8M      1.5M     84.3M   2% /
```

Suggestion: Adjust F2FS_MINSIZE comfortably lower or higher to avoid switching between the two filesystems. Personally I would prefer ext4 for the lower overhead, but that decision is up to the devs.

(With thanks to slh and PaulFerster for pinpointing the cause.)

corna, 2022-04-09:
I would like instead to have a f2fs fs on x86. After an unclean shutdown (quite common on routers), ext4 refuses to mount rw until a fsck is performed, which is not done automatically on boot (and [I can't find a way to do it](https://forum.openwrt.org/t/fsck-on-boot-on-x86-64-with-ext4/83982)).

TalalMash, 2022-04-21:
> I would like instead to have a f2fs fs on x86. After an unclean shutdown (quite common on routers), ext4 refuses to mount rw until a fsck is performed, which is not done automatically on boot (and [I can't find a way to do it](https://forum.openwrt.org/t/fsck-on-boot-on-x86-64-with-ext4/83982)).

https://github.com/openwrt/openwrt/issues/9113#issuecomment-1052604257