openwrt/openwrt #9353: If I add more than one SSID all wireles interfaces become disabled

gituletz, 2022-02-27:
Model | Asus RT-N16
Architecture | Broadcom BCM4716
Target Platform | bcm47xx/mips74k
Firmware Version | OpenWrt 21.02.2 r16495-bf0c965af0 / LuCI openwrt-21.02 branch git-22.046.85957-59c3392
Kernel Version | 5.4.179

Adding the second SSID disables all wireless interfaces.

Wed Feb 16 23:06:15 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Wed Feb 16 23:06:15 2022 kern.info kernel: [ 9438.288278] b43-phy0: Loading firmware version 666.2 (2011-02-23 01:15:07)
Wed Feb 16 23:06:15 2022 kern.info kernel: [ 9438.504414] br-lan: port 2(wlan0) entered blocking state
Wed Feb 16 23:06:15 2022 kern.info kernel: [ 9438.509905] br-lan: port 2(wlan0) entered disabled state
Wed Feb 16 23:06:15 2022 kern.info kernel: [ 9438.516018] device wlan0 entered promiscuous mode
Wed Feb 16 23:06:15 2022 kern.info kernel: [ 9438.556650] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Wed Feb 16 23:06:15 2022 kern.info kernel: [ 9438.563594] br-lan: port 2(wlan0) entered blocking state
Wed Feb 16 23:06:15 2022 kern.info kernel: [ 9438.569115] br-lan: port 2(wlan0) entered forwarding state
Wed Feb 16 23:06:15 2022 daemon.err hostapd: Could not set interface wlan0-1 flags (UP): Resource busy
Wed Feb 16 23:06:15 2022 daemon.err hostapd: Failed to add BSS (BSSID=bc:ae:xx:xx:8a:7d)
Wed Feb 16 23:06:15 2022 daemon.err hostapd: Interface initialization failed
Wed Feb 16 23:06:15 2022 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->DISABLED
Wed Feb 16 23:06:15 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Wed Feb 16 23:06:15 2022 daemon.err hostapd: wlan0: Unable to setup interface.
Wed Feb 16 23:06:15 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Wed Feb 16 23:06:15 2022 kern.info kernel: [ 9438.628712] device wlan0 left promiscuous mode
Wed Feb 16 23:06:15 2022 kern.info kernel: [ 9438.633568] br-lan: port 2(wlan0) entered disabled state
Wed Feb 16 23:06:15 2022 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Wed Feb 16 23:06:15 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Wed Feb 16 23:06:15 2022 daemon.notice hostapd: wlan0-1: CTRL-EVENT-TERMINATING
Wed Feb 16 23:06:15 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0-1 wasn't started
Wed Feb 16 23:06:15 2022 daemon.notice hostapd: wlan0-2: CTRL-EVENT-TERMINATING
Wed Feb 16 23:06:15 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0-2 wasn't started
Wed Feb 16 23:06:15 2022 daemon.notice netifd: radio0 (15390): Command failed: Invalid argument
Wed Feb 16 23:06:15 2022 daemon.notice netifd: radio0 (15390): Device setup failed: HOSTAPD_START_FAILED
Wed Feb 16 23:06:15 2022 daemon.notice netifd: Wireless device 'radio0' set retry=0
Wed Feb 16 23:06:15 2022 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Wed Feb 16 23:06:16 2022 daemon.notice netifd: Wireless device 'radio0' is now down

root@OpenWrt:~# wifi status
{
        "radio0": {
                "up": false,
                "pending": false,
                "autostart": true,
                "disabled": false,
                "retry_setup_failed": true,
                "config": {
                        "path": "bcm47xx_soc/bcma0:1",
                        "channel": "1",
                        "band": "2g",
                        "cell_density": 0
                },
                "interfaces": [
                        {
                                "section": "default_radio0",
                                "config": {
                                        "mode": "ap",
                                        "ssid": "OpenWrt",
                                        "encryption": "none",
                                        "macaddr": "bc:ae:xx:xx:8a:5d",
                                        "network": [
                                                "lan"
                                        ],
                                        "mode": "ap",
                                        "disabled": false
                                },
                                "vlans": [

                                ],
                                "stations": [

                                ]
                        },
                        {
                                "section": "wifinet1",
                                "config": {
                                        "mode": "ap",
                                        "ssid": "test",
                                        "encryption": "none",
                                        "macaddr": "bc:ae:xx:xx:8a:7d",
                                        "mode": "ap",
                                        "disabled": false
                                },
                                "vlans": [

                                ],
                                "stations": [

                                ]
                        },
                        {
                                "section": "wifinet2",
                                "config": {
                                        "mode": "ap",
                                        "ssid": "OpenWrt2",
                                        "encryption": "none",
                                        "macaddr": "bc:ae:xx:xx:8a:6d",
                                        "mode": "ap",
                                        "disabled": false
                                },
                                "vlans": [

                                ],
                                "stations": [

                                ]
                        }
                ]
        }
}

jneuhauser, 2022-03-03:
Can you post the output of the command `iw list`.

wurzerj, 2022-03-04:
Sounds like this issue: https://github.com/openwrt/openwrt/issues/9343#issuecomment-1052410915

gituletz, 2022-03-07:
> Can you post the output of the command `iw list`.

root@OpenWrt:~# iw list
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 4
        max scan IEs length: 2285 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 1:
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm) (no IR)
                        * 2472 MHz [13] (20.0 dBm) (no IR)
                        * 2484 MHz [14] (20.0 dBm) (no IR)
        interface combinations are not supported
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Supported extended features:
                * [ RRM ]: RRM
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
                * [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IE                                                                                                                     s in scans
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 con                                                                                                                     trol port support
                * [ DEL_IBSS_STA ]: deletion of IBSS station support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211                                                                                                                      control port support

bbatod, 2022-08-31:
Ran into the same issue on Linksys WRT32X on latest 22.03.0-rc6 (r19590-042d558536) firmware.

dr-ni, 2022-08-31:
Same here on wrt3200acm with 21.02.03 and latest upgrades - but removing all lines containing “option macaddr” from /etc/config/wireless will fix the problem?