openwrt/openwrt #9363: mount options not applied on extroot overlay

M95D, 2022-02-28:
OpenWrt snapshot r19000 on mvebu (Linksys WRT1900ACv1)
BTRFS included in kernel (not module), btrfs tools included in squashfs
I'm mounting extroot overlay with these options:
[image]
But mount command shows this
`root@OpenWrt:~# mount
/dev/root on /rom type squashfs (ro,relatime)
devtmpfs on /rom/dev type devtmpfs (rw,relatime,size=123928k,nr_inodes=30982,mode=755)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/sda on /overlay type btrfs (rw,noatime,noacl,space_cache=v2,subvolid=5,subvol=/)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
ubi1:syscfg on /tmp/syscfg type ubifs (rw,relatime,assert=read-only,ubi=1,vol=0)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
mountd(pid18035) on /tmp/run/blockd type autofs (rw,relatime,fd=7,pgrp=18035,timeout=30,minproto=5,maxproto=5,indirect)
root@OpenWrt:~#`
The mount options that are actually used are the defaults.

M95D, 2022-02-28:
It seems that all changes in Luci are saved in extroot overlay instead of ubi overlay! They are not available until the extroot is mounted. Whatever was saved before extroot was enabled becomes permanent. The only way to change mount parameters is to reboot without the extroot stick / hdd, make changes, and reboot again.

xabolcs, 2022-02-28:
This is how [overlay works](https://openwrt.org/docs/guide-user/additional-software/extroot_configuration).

To [change `overlay` mount options](https://openwrt.org/docs/guide-user/additional-software/extroot_configuration#configuring_rootfs_data) just mount `rootfs_data` (your original `/overlay`) somewhere and edit `etc/config/fstab` there.
To avoid tricking yourself again just remove the `/overlay` mounting section from `/etc/config/fstab` on the extroot.

And you are are also right, a note in the LuCI and the wiki would be helpful.

closed 2024-12-04 by M95D