openwrt/openwrt #9365: Error: ipv4: FIB table does not exist.

predators46, 2022-03-01:
Hy @feckert 

is this an iproute2 bug? I see this when I go to MWAN Status - Troubleshooting. and I see this warning in system.log. is this normal i should ignore it? I'm using youku router yk-1 openwrt 21.02.2

`Tue Mar  1 05:03:56 2022 daemon.err uhttpd[972]: Error: ipv4: FIB table does not exist.
Tue Mar  1 05:03:56 2022 daemon.err uhttpd[972]: Dump terminated
Tue Mar  1 05:03:56 2022 daemon.err uhttpd[972]: Error: ipv4: FIB table does not exist.
Tue Mar  1 05:03:56 2022 daemon.err uhttpd[972]: Dump terminated
Tue Mar  1 05:03:56 2022 daemon.err uhttpd[972]: Error: ipv4: FIB table does not exist.`

feckert, 2022-03-01:
I have no idea what you mean by a `FIB` table.

predators46, 2022-03-01:
> I have no idea what you mean by a `FIB` table.

this error occurs when i finish installing mwan3 and go to MWAN Status - Troubleshooting and to system.log this error is there

predators46, 2022-03-02:
@feckert 

as I suspected iproute2

https://lore.kernel.org/all/LaklAOJ--3-1@tutanota.com/T/

champtar, 2022-03-06:
The iptables-legacy warning can usually be ignored, I might have time to fix it before branching but no promise

closed 2022-03-20 by predators46

MichaelWoodc, 2025-12-02:
2025 and I had a similar error, this was one of the only places I found with a similar enough error code and details about  iproute2 and MWAN, so here's what I had going on incase anybody stumbles across this:

```
root@OpenWrt:~# /etc/init.d/mwan3 restart
Error: ipv6: FIB table does not exist.
Dump terminated
Error: ipv6: FIB table does not exist.
Dump terminated
Error: ipv6: FIB table does not exist.
Dump terminated
```
It was quite puzzling.  wan worked just fine, and if I brought wanb up forcefully, it would work just fine, but then wan would go offline, and vice versa.

It was a game of digital whac-a-mole that kept going on until I forced the tables to exist, and then forced openwrt to use them.  First, I created tables in the iproute2 rt_tables file, then I seeded them which took care most of the error messages:
```
ip route add default via 10.0.0.1 dev wan table 220
ip route add default via 10.0.0.1 dev wan table 221
ip route add default via 10.1.1.1 dev lan4 table 222
```
but it seems that I had to make some rules for both of the interfaces to show up as online and stay online:
```
ip rule add from 10.0.0.246/32 table 221
ip rule add from 10.1.1.3/32 table 222
```

after that, I can finally surf the web with mwan3 working properly.