openwrt/openwrt #9366: [21.02-snapshoot] [bcm27xx-bcm2708] - kernel/irq/chip.c:1016 irq_set_chained_handler_and_data+0x3c/0x54

kofec, 2022-03-01:
On Raspberry Pi Model B Rev 2:
`root@RpiTestowe:~# logread 
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Thu Feb  3 14:57:01 2022 kern.notice kernel: [    0.000000] Linux version 5.4.179 (MyBuild@OnServer) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16449-27225e3)) #0 Tue Feb 16 08:53:24 2021
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Model B Rev 2
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] Memory policy: Data cache writeback
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x17c00000, size 64 MiB
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Thu Feb  3 14:57:01 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 114688
Thu Feb  3 14:57:01 2022 kern.debug kernel: [    0.000000]   Normal zone: 1008 pages used for memmap
Thu Feb  3 14:57:01 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Thu Feb  3 14:57:01 2022 kern.debug kernel: [    0.000000]   Normal zone: 114688 pages, LIFO batch:31
Thu Feb  3 14:57:01 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Thu Feb  3 14:57:01 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
Thu Feb  3 14:57:01 2022 kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] Memory: 379764K/458752K available (5489K kernel code, 208K rwdata, 1388K rodata, 1024K init, 337K bss, 13452K reserved, 65536K cma-reserved)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Thu Feb  3 14:57:01 2022 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x278/0x414 with crng_init=0
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000023] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000077] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000186] bcm2835: system timer (irq = 27)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.000640] Console: colour dummy device 80x30
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.001037] printk: console [tty1] enabled
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.001105] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.060324] pid_max: default: 32768 minimum: 301
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.060644] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.060695] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.061882] cgroup: Disabling memory control group subsystem
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.062094] CPU: Testing write buffer coherency: ok
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.063120] Setting up static identity map for 0x100000 - 0x10005c
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.072508] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.072941] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.073004] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.081223] pinctrl core: initialized pinctrl subsystem
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.082744] NET: Registered protocol family 16
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.085560] DMA: preallocated 1024 KiB pool for atomic coherent allocations
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.091164] cpuidle: using governor ladder
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.091224] cpuidle: using governor menu
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.091821] Serial: AMBA PL011 UART driver
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.093950] bcm2835-mbox 2000b880.mailbox: mailbox enabled
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.110576] raspberrypi-firmware soc:firmware: Attached to firmware from 2021-02-16 14:19, variant start
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.120589] raspberrypi-firmware soc:firmware: Firmware hash is d6d82cf99bcb3e9a166a34cfde53130957a36bd3
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.162305] bcm2835-dma 20007000.dma: DMA legacy API manager, dmachans=0x1
Thu Feb  3 14:57:01 2022 kern.notice kernel: [    0.165518] SCSI subsystem initialized
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.165786] usbcore: registered new interface driver usbfs
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.165912] usbcore: registered new interface driver hub
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.166023] usbcore: registered new device driver usb
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.167809] clocksource: Switched to clocksource timer
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.176540] thermal_sys: Registered thermal governor 'step_wise'
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.177178] NET: Registered protocol family 2
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.177507] IP idents hash table entries: 8192 (order: 4, 65536 bytes, linear)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.178772] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.178869] TCP established hash table entries: 4096 (order: 2, 16384 bytes, linear)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.178976] TCP bind hash table entries: 4096 (order: 2, 16384 bytes, linear)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.179090] TCP: Hash tables configured (established 4096 bind 4096)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.179251] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.179309] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.179658] NET: Registered protocol family 1
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.181470] workingset: timestamp_bits=14 max_order=17 bucket_order=3
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.191208] squashfs: version 4.0 (2009/01/31) Phillip Lougher
Thu Feb  3 14:57:01 2022 kern.warn kernel: [    0.194504] ------------[ cut here ]------------
Thu Feb  3 14:57:01 2022 kern.warn kernel: [    0.194598] WARNING: CPU: 0 PID: 1 at kernel/irq/chip.c:1016 irq_set_chained_handler_and_data+0x3c/0x54
Thu Feb  3 14:57:01 2022 kern.warn kernel: [    0.194633] Modules linked in:
[... 8 lines cut ...]
Thu Feb  3 14:57:01 2022 kern.warn kernel: [    0.195084] [<c03a8bac>] (devm_gpiochip_add_data) from [<c03a2b94>] (bcm2835_pinctrl_probe+0x3ac/0x454)
Thu Feb  3 14:57:01 2022 kern.warn kernel: [    0.195138] [<c03a2b94>] (bcm2835_pinctrl_probe) from [<c0419e38>] (platform_drv_probe+0x48/0x98)
[... cut ...]
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.214553] bcm2708_fb soc:fb: Registered framebuffer for display 0, size 656x416
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.225312] bcm2835-rng 20104000.rng: hwrng registered
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.228881] vc-mem: phys_addr:0x00000000 mem_base=0x1ec00000 mem_size:0x20000000(512 MiB)
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.235858] vc-sm: Videocore shared memory driver
Thu Feb  3 14:57:01 2022 kern.info kernel: [    0.239511] gpiomem-bcm2835 20200000.gpiomem: Initialised: Registers at 0x2020000`

kofec, 2022-04-03:
Issue is not visible on 22-03

Noltari, 2025-01-15:
This was fixed on 22.03.

closed 2025-01-15 by Noltari