openwrt/openwrt #9367: VHT160 broken with OpenWrt 21.02.2 r16495-bf0c965af0 on Netgear R7800 / ath10k-CT - configuration error

NilsRo, 2022-03-01:
Got a configuration error is I like to use VHT160. VHT80 is working fine. Here are the logs, configuration attached.
[wireless.txt](https://github.com/openwrt/openwrt/files/8161179/wireless.txt)

Tue Mar  1 12:24:49 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED b0:a4:60:3d:1c:83
Tue Mar  1 12:24:49 2022 daemon.debug hostapd: wlan1: STA b0:a4:60:3d:1c:83 IEEE 802.1X: authorizing port
Tue Mar  1 12:24:49 2022 daemon.info hostapd: wlan1: STA b0:a4:60:3d:1c:83 RADIUS: starting accounting session 9D7DF0896931F8B7
Tue Mar  1 12:24:49 2022 daemon.info hostapd: wlan1: STA b0:a4:60:3d:1c:83 WPA: pairwise key handshake completed (RSN)
Tue Mar  1 12:24:49 2022 daemon.info dnsmasq-dhcp[3784]: DHCPREQUEST(br-lan) 192.168.177.139 b0:a4:60:3d:1c:83
Tue Mar  1 12:24:49 2022 daemon.info dnsmasq-dhcp[3784]: DHCPACK(br-lan) 192.168.177.139 b0:a4:60:3d:1c:83 Dell-7706
Tue Mar  1 12:24:50 2022 daemon.notice netifd: Wireless device 'radio0' is now up
Tue Mar  1 12:24:52 2022 daemon.err hostapd: ACS: Failed to compute ideal channel
Tue Mar  1 12:24:52 2022 daemon.err hostapd: Interface initialization failed
Tue Mar  1 12:24:52 2022 daemon.notice hostapd: wlan0: interface state ACS->DISABLED
Tue Mar  1 12:24:52 2022 daemon.notice hostapd: wlan0: AP-DISABLED
**Tue Mar  1 12:24:52 2022 daemon.err hostapd: ACS: Possibly channel configuration is invalid, please report this along with your config file.**
Tue Mar  1 12:24:52 2022 daemon.err hostapd: ACS: Failed to start
Tue Mar  1 12:24:52 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Tue Mar  1 12:24:52 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Tue Mar  1 12:24:52 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0-1 wasn't started
Tue Mar  1 12:24:52 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0-2 wasn't started
Tue Mar  1 12:24:52 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Tue Mar  1 12:24:52 2022 kern.info kernel: [ 1297.290565] device wlan0 left promiscuous mode
Tue Mar  1 12:24:52 2022 kern.info kernel: [ 1297.290672] br-lan: port 3(wlan0) entered disabled state
Tue Mar  1 12:24:52 2022 kern.warn kernel: [ 1297.331628] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
Tue Mar  1 12:24:52 2022 daemon.notice hostapd: wlan0: interface state DISABLED->DISABLED
Tue Mar  1 12:24:52 2022 daemon.debug hostapd: wlan1: STA 14:c8:8b:66:46:f7 IEEE 802.11: authenti

hnyman, 2022-03-03:
> Tue Mar 1 12:24:52 2022 daemon.err hostapd: ACS: Failed to compute ideal channel

You should probably stop using "auto" channel with VHT160.
(Your settings show that "auto")

NilsRo, 2022-03-03:
> > Tue Mar 1 12:24:52 2022 daemon.err hostapd: ACS: Failed to compute ideal channel
> 
> 
> 
> You should probably stop using "auto" channel with VHT160.
> 
> (Your settings show that "auto")
> 
> 
> 
> Vht160 works for me in R7800

It seems happening if country is not set. With that combination it does not work...I cannot set a DFS channel manually then.

Black6spdZ, 2022-05-15:
> > Tue Mar 1 12:24:52 2022 daemon.err hostapd: ACS: Failed to compute ideal channel
> 
> You should probably stop using "auto" channel with VHT160. (Your settings show that "auto")
> 
> Vht160 works for me in R7800

what are your settings and throughput?

NilsRo, 2022-05-15:
It's working with that version if full htt driver used and ACS disabled. With ACS it does not work.

zclkkk, 2023-06-07:
You need to select proper country which allows VHT160 to be used.
Check https://en.m.wikipedia.org/wiki/List_of_WLAN_channels