openwrt/openwrt #9369: [21.02.1] Pairing with WPS let device connect only once and then never again

fractalwalkr, 2022-03-01:
Name the tree/revision/version: 
* openwrt-21.02.1-ath79-generic-tplink_archer-c7-v5-squashfs-sysupgrade

Name the affected device: 
* TP Link Archer C7 v5

What does it do that it should not do / what does it not do that it should do:
* Devices which was paired with WPS will only connect once to the AP. After a network restart, it won't connect anymore.

Steps to reproduce:
* Setup WIFI AP:
    Mode = N
    Width = 20Mhz
    WPA2-PSK
    Force CCMP   
* Install WPS via CLI:
  opkg update
  opkg remove wpad-mini
  opkg remove wpad-basic-wolfssl
  opkg install wpad hostapd-utils
  -> OK
* Enable WPS for Wifi / AP
* Pair device via WPS
  hostapd_cli -i wlan1 wps_pbc
  hostapd_cli wps_get_status
  -> OK
* See device connecting to AP and it can be pinged
* Restart network or reboot device
  reboot

What it does:
* Device does not pair again anymore
* You see it only short under "Associated Stations" and then it disconnect again. Device tries over and over and it fails always
* Log says with each attempt: 
 daemon.notice hostapd: wlan1: AP-STA-POSSIBLE-PSK-MISMATCH xx:xx:xx:xx..

What it should do:
* Device should always connect to AP after a successful WPS pairing

What you have already done to workaround/fix the problem
* Downgraded to 19.07.8. No issue there. WPS pairing works with described procedure and the device can always connect again afterwards. So it is not a hardware issue.

ptpt52, 2022-03-02:
wpa key sync by wps is temporary so it would not work after router reboot.

fractalwalkr, 2022-03-02:
It's not the WPS which does not work afterwards. It is the pairing of the device to the access point which does not persist.

Ansuel, 2022-03-11:
Can confirm this. I honestly think the algo is not correctly communicated with WPS.

We should check the changes to hostpad about the WPS part...

dunglx, 2022-07-29:
> Can confirm this. I honestly think the algo is not correctly communicated with WPS.
> 
> We should check the changes to hostpad about the WPS part...

I meet the same problem.
@Ansuel Please help us check the changes. Thanks!

dunglx, 2022-08-04:
I see in the source file openwrt/package/network/services/hostapd/file/hostapd.sh, then wpa_psk_file is /var/run/hostapd-$ifname.psk. It is just a temporary file and will be a new file after the router reboot. 
I changed the path to /etc/hostapd-$ifname.psk and the connection by WPS can be keep after router reboot.

dunglx, 2022-08-05:
It's ok when pairing with one tablet via WPS (both router reboots and tablet reboots.
But, when I pair with a printer, then it's ok when router reboots and not OK when the printer reboot. 
Can anyone give me some ideas for this?

Salamandar, 2023-06-30:
I can confirm @dunglx 's problem.

lukasz1992, 2023-10-27:
I wonder if setting ap_setup_locked=1 within wifi iface would help?

OlafTitz, 2024-04-26:
After some experimenting I found out that what you need to do is:
1. Configure wpa_psk_file to a file in /etc
2. Create that file (empty) and make it writable for group "network"
3. Put that file's path in /etc/sysupgrade.conf to have it backed up
Then you will be able to use manually set per-MAC passphrases too.

The standard configuration should probably include this from the start.

ssdnvv, 2025-04-27:
> After some experimenting I found out that what you need to do is:

This works great and solves the issue - thanks alot!