openwrt/openwrt #9372: Channel 153(5765MHz) not work

intdev32, 2022-03-02:
I updated openwrt to recent snapshot build (r18953-b9251e3b40) and found 5GHz channel 153 is not working.

When I set channel to '153', it goes to 149.

But 149, 157, 161 working well.

Only 153 doesn't work.

Happened on below devices:
Linksys EA4500 (wireless driver: mwl8k) with 21.02.2
Linksys E5600 (wireless driver: mt76) with SNAPSHOT r18953

is this hostapd bug?

intdev32, 2022-03-02:
https://github.com/openwrt/mt76/issues/650

intdev32, 2022-03-02:
[https://github.com/openwrt/mt76/issues/650](url)

intdev32, 2022-03-02:
[EA5600 iw list Result]

root@OpenWrt:~# iw list
Wiphy phy1
        wiphy index: 1
        max # scan SSIDs: 4
        max scan IEs length: 2261 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
        Band 2:
                Capabilities: 0xffe
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 3-streams
                        HT Delayed Block Ack
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-23, 32
                Frequencies:
                        * 5180 MHz [36] (23.0 dBm)
                        * 5200 MHz [40] (23.0 dBm)
                        * 5220 MHz [44] (23.0 dBm)
                        * 5240 MHz [48] (23.0 dBm)
                        * 5745 MHz [149] (30.0 dBm)
                        *** 5765 MHz [153] (30.0 dBm)**
                        * 5785 MHz [157] (30.0 dBm)
                        * 5805 MHz [161] (30.0 dBm)
                        * 5825 MHz [165] (30.0 dBm)
        valid interface combinations:
                 * #{ AP } <= 8, #{ managed } <= 1,
                   total <= 8, #channels <= 1
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Supported extended features:
                * [ RRM ]: RRM
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
                * [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support

Ostring, 2024-03-06:
The same thing happens to me with the E5600 too.