openwrt/openwrt #9374: R6220 2.4GHz radio0 issues

Bartixxx32, 2022-03-02:
Hello, i have issues with 2.4ghz wireless not working on openwrt 21.2.02 on my Netgear R6220, i have also tested snapshot build, with same result.
Im pretty sure that issue is caused by bad eraseblocks, becouse every (x) reboots radio0 would be up and working util next "power outage"

```
root@OpenWrt:~# dmesg | grep eraseblock
[    1.922023] Bad eraseblock 847 at 0x0000069e0000
```
[image]

I have read on forum that kernel 4.x had applied patch and 5.x wont.

I tried to install openwrt 19.07.9 whih i guess fixed my problem (gonna test it longer) But thats not a good solution to me to stick with old firmware. 

Also, i have been trying to move partition "factory" to other offsets using imagebuilder, but seems i dont know how, or it need be rebuilded from sourcecode, becouse nothing in `/prod/mtd` is changing when builded new image.

Ps. Sorry for my bad english (not my main lang)

gj86, 2022-03-26:
I am adding to this, as it sounds similar to my issue.

Firmware versions: `OpenWRT 21.02.2 & snapshots up to 26 March, 2022` all exhibit the same issue.
Hardware: `Netgear R6220`

On reboot or power outage sometimes `2.4Ghz/MT7603E` radio doesn't get detected. However, I don't have any bad `eraseblock` issue.

For some reason if I keep the router turned off for 10 minutes or so before turning on, it will work until next reboot/power outage happens.

I have tried using the suggested fix from [https://forum.openwrt.org/t/r6220-could-not-find-phy-for-device-radio0-mt76x2-issue-continues/116665](url) without any success.

With `OpenWRT 19.07.9` everything works as expected.

Bartixxx32, 2022-05-29:
> I am adding to this, as it sounds similar to my issue.
> 
> Firmware versions: `OpenWRT 21.02.2 & snapshots up to 26 March, 2022` all exhibit the same issue.
> Hardware: `Netgear R6220`
> 
> On reboot or power outage sometimes `2.4Ghz/MT7603E` radio doesn't get detected. However, I don't have any bad `eraseblock` issue.
> 
> For some reason if I keep the router turned off for 10 minutes or so before turning on, it will work until next reboot/power outage happens.
> 
> I have tried using the suggested fix from [https://forum.openwrt.org/t/r6220-could-not-find-phy-for-device-radio0-mt76x2-issue-continues/116665](https://forum.openwrt.org/t/r6220-could-not-find-phy-for-device-radio0-mt76x2-issue-continues/116665) without any success.
> 
> With `OpenWRT 19.07.9` everything works as expected.
> 
> 

Did you resolve your problem? I forgot about this issue tbh haha

gj86, 2022-05-30:
> Did you resolve your problem? I forgot about this issue tbh haha

I'm still using the **19.07.9**
When I get some free time I will try the latest build/snapshot to see if it fixed.

krieghof, 2022-05-30:
> > Did you resolve your problem? I forgot about this issue tbh haha
> 
> I'm still using the **19.07.9** When I get some free time I will try the latest build/snapshot to see if it fixed.

I tried 22.03.0-rc1 a week ago, 2.4GHz still missing.

gj86, 2022-06-17:
I'm currently using June 17 snapshot. Same problem with 2.4Ghz radio missing. If I keep the router turned off for about 5 minutes it will work on next power on. But on next power loss it be gone again.

DragonBluep, 2022-06-26:
There are two reasons for this problem:
pcie reset pin.  #10125 
Sercomm mtd partitions. #10038

Still need someone to port to Sercomm mtd partitions.

Kline-, 2022-07-05:
Just ran into this myself after rebooting to failsafe to fix a bad config change I made. After rebooting normally radio0 (2.4ghz) won't come back up.

<pre>DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.3'
DISTRIB_REVISION='r16554-1d4dea6d4f'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02.3 r16554-1d4dea6d4f'</pre>

No messages about eraseblocks in dmesg.

gj86, 2022-07-22:
Possible solution: [#601](https://github.com/openwrt/mt76/issues/601)

Kline-, 2022-08-22:
I'm not sure if this is fixed in rc6 or if I'm just getting lucky tonight since the issue was intermittent for me. Can anyone else flash rc6 and verify?

Bartixxx32, 2022-09-10:
> I'm not sure if this is fixed in rc6 or if I'm just getting lucky tonight since the issue was intermittent for me. Can anyone else flash rc6 and verify?

For you still no issues? Ill test it in ~month when come back home

Kline-, 2022-09-10:
Thinking maybe I was just lucky. Flashed 22.03 when it went stable and right back to no 2.4GHz. Haven't tried rolling back to rc6 yet.

wifiseguro, 2022-09-12:
> I am adding to this, as it sounds similar to my issue.
>
> Firmware versions: `OpenWRT 21.02.2 & snapshots up to 26 March, 2022` all exhibit the same issue. Hardware: `Netgear R6220`
>
> On reboot or [...]

My device zbt we1326 (mt7621) has exactly the same behaviour as yours. Its funny the part where you say that turning off for ten minutes fixes it until next reboot because I also came to that workaround.
I have been googling many days until and found out many mt7621 devices with very similar issues until I found a fix that worked for me. 
This is the log message that points this pcie ports issue:

```
Sat Sep  3 02:56:31 2022 kern.err kernel: [    1.289026] mt7621-pci 1e140000.pcie: pcie0 no card, disable it (RST & CLK)
Sat Sep  3 02:56:31 2022 kern.err kernel: [    1.295959] mt7621-pci 1e140000.pcie: pcie2 no card, disable it (RST & CLK)
Sat Sep  3 02:56:31 2022 kern.info kernel: [    1.302912] mt7621-pci 1e140000.pcie: PCIE1 enabled

```
As you can see, only one port is detected, so only one wifi band.

It its about adding a delay when initializing the pcie ports so the two pcie ports in my hardware are detected properly (the third port is not detected as I dont have any card attached to it)
Adding these delays to the firmware fixes it in every release, Its been tested for many months on 21.02 and the issue is gone ,no matter I reboot or do anything, the 2,4 wlan is always there in every reboot :-)

Recently I added the patch to 22.03 because the issue was still on this new release:
I attach the patch for 22.03 if you want to compile from source- Just add a file in patches folder ie:
target/linux/ramips/patches-5.10/901-staging-mt7621-pci-delay-for-properly-detect.patch

with this inside:

```
--- a/drivers/staging/mt7621-pci/pci-mt7621.c
+++ b/drivers/staging/mt7621-pci/pci-mt7621.c
@@ -502,8 +502,9 @@
 	int err;
 
 	rt_sysc_m32(PERST_MODE_MASK, PERST_MODE_GPIO, MT7621_GPIO_MODE);
-
+	mdelay(250);
 	mt7621_pcie_reset_assert(pcie);
+	mdelay(250);
 	mt7621_pcie_reset_rc_deassert(pcie);
 
 	list_for_each_entry_safe(port, tmp, &pcie->ports, list) {
@@ -520,7 +521,7 @@
 			list_del(&port->list);
 		}
 	}
-
+	mdelay(250);
 	mt7621_pcie_reset_ep_deassert(pcie);
 
 	tmp = NULL;

```
Hope it works for you all, as it does for me

If you want to run 21.02 I can provide the patch aswell (it is not applied just like the above for 22.03)

Kline-, 2022-09-13:
@wifiseguro Thank you, this worked perfectly so far! I just finished building 22.03 with your patch and both radios came up immediately without any issues.

<pre>Sun Sep 11 09:24:51 2022 kern.err kernel: [    3.671442] mt7621-pci 1e140000.pcie: pcie1 no card, disable it (RST & CLK)
Sun Sep 11 09:24:51 2022 kern.info kernel: [    3.685319] mt7621-pci 1e140000.pcie: PCIE0 enabled
Sun Sep 11 09:24:51 2022 kern.info kernel: [    3.695030] mt7621-pci 1e140000.pcie: PCIE2 enabled</pre>

pierrepinon, 2022-09-19:
@Kline- can you provide your build ?

Kline-, 2022-09-19:
@pierrepinon Sure, here you go. I used a [docker](https://github.com/noonien/docker-openwrt-buildroot) build environment and only copied out the sysupgrade bin at the time. The only differences (aside from the patch) from the official image is that mine was built on r19701 instead of r19685, and I baked in a copy of nano for myself.

If you need the OEM flash bin let me know and I'll spin the docker back up and build another one.
[openwrt-ramips-mt7621-netgear_r6220-squashfs-sysupgrade.zip](https://github.com/openwrt/openwrt/files/9602107/openwrt-ramips-mt7621-netgear_r6220-squashfs-sysupgrade.zip)

pierrepinon, 2022-09-24:
@Kline- Thx ❤️

pprakash, 2022-09-29:
> @pierrepinon Sure, here you go. I used a [docker](https://github.com/noonien/docker-openwrt-buildroot) build environment and only copied out the sysupgrade bin at the time. The only differences (aside from the patch) from the official image is that mine was built on r19701 instead of r19685, and I baked in a copy of nano for myself.
> 
> If you need the OEM flash bin let me know and I'll spin the docker back up and build another one. [openwrt-ramips-mt7621-netgear_r6220-squashfs-sysupgrade.zip](https://github.com/openwrt/openwrt/files/9602107/openwrt-ramips-mt7621-netgear_r6220-squashfs-sysupgrade.zip)
> 
> Hashes of the extracted bin:
> 
> * MD5 d3ae5e670fc33de8e9a762b017c7f600
> * SHA256 0cd3659b0626c9462a811684c7c7637832e0da35de511b32d14a0428e66892f8

Solution works perfectly. Thanks @Kline-

kgrodrigues, 2022-10-10:
this fix worked for me too. what would it take to get this into the official builds?

Bartixxx32, 2022-10-15:
At free time without access to my router im working on repo for automatic builds of firmware for r6220 and patch applied, this build contains nano luci and few luci mods inbuild, I HAVE NOT TESTED IT YET cuz im rn working on ci optimization
Fell free to PR or just use it

https://github.com/Bartixxx32/r6220-build/releases

Djfe, 2023-02-26:
The last fix regarding the R6220, was just merged into master.
https://github.com/openwrt/openwrt/pull/12042

@Bartixxx32 If you get around to it, please test latest Snapshot and then close this issue if everything is working as expected :)
Thanks

Bartixxx32, 2023-03-07:
Hmm, i have to test it more, but when i tried restore my backup when flashed snapshot somentig wild happened, got extra radio2, im guessing that radio1 and radio2 are primary, and radio0 is leftover? im not sure, when i did restore to defaults, without backups, radio`s was normal, radio0 and radio1, now im trying to restore my backup except wireless settings file, and trying catch an issue

wifiseguro, 2023-03-07:
Hi Bartixxx32, just want let you know that I had same issue when flashed snapshot keeping settings,  2g radio was duplicated, and shown as Generic just like happened to you.
Flashing with no settings keept fixed it for me too.
I  would say that is related to new pcie ports numbering declared in the dts file, that somehow "fools" the system to keep old and new declared radio in the config so it shows duplicated.

Djfe, 2023-03-07:
good catch!

I'll look at how the devices and their radio can be migrated by some hotplug.d script on upgrade. I've never done anything like that before so if anyone's got any pointers for me how to do it, I'm all ears.
And if anyone else wants to provide the patch instead, I wouldn't mind either. It might take sometime for me to provide a patch, but I will make sure it's part of the next release :)

Djfe, 2023-03-18:
@wifiseguro would you be able to test the following patch for me? :)
https://github.com/openwrt/openwrt/pull/12198
Upgrade from release to master with my patch and then check whether it shows just two radios. uci wireless would be interesting for me as well.

I can provide a sysupgrade for people if the demand is there. (someone doesn't have the means to build it themselves)

Djfe, 2023-03-18:
The patch was tested on Netgear hw and migrated the wifi path successfully :)
Looking good so far, but the Zbt has two changes to wifi paths and one radio gets the old wifi path of the other so I still want to make sure that the migration runs successfully.

But don't stress yourself, I just wanted to give an update. Take all the time you need.

closed 2023-04-01 by openwrt-bot

Djfe, 2023-04-02:
Pls test the current snapshot for remaining bugs :)
My patch was just merged.

Djfe, 2023-05-14:
Pls post the output of `cat /etc/config/wireless`, when it happens :)

tankwrt, 2023-05-17:
@mikerodrigues  as far as I know, the fix was sent to kernel 5.15
If you want to stick with 22.03 you have to patch and build yourself.

mikerodrigues, 2023-05-19:
Thanks @tankwrt. I'm not married to 22.03, I just figured it would where this would land first.

I tried to build with the patch months ago but couldn't get it to work. I'm happy to wait for it to hit an official release.

commit cbf120c2 (2023-04-01): ramips: mt7621: add migration script for WLAN PCI paths

PCI paths of the WLAN devices have changed between kernel 5.10 and 5.15;
migrate config so existing wifi-iface definitions don't break.

This is implemented as a hotplug handler rather than a uci-defaults script
as the migration script must run before the 10-wifi-detect hotplug handler.

based on b452af23a8602ebf7bfb0eb084383ecd595face5

migration was forgotten when device trees were adjusted in
688697889c9ceb9d2d8d413d1abcc5fa7edb402b
c77913be5be5f11a0f20dba1b45b96470099e8a8

fixes #9374

affected devices:
Netgear R6220
Netgear WAC104
N [... cut]
PR #12198 (2023-04-01): ramips: mt7621: add migration script for WLAN PCI paths
PCI paths of the WLAN devices have changed between kernel 5.10 and 5.15; migrate config so existing wifi-iface definitions don't break.

This is implemented as a hotplug handler rather than a uci-defaults script as the migration script must run before the 10-wifi-detect hotplug handler.

based on b452af23a8602ebf7bfb0eb084383ecd595face5

migration was forgotten when device trees were adjusted in
688697889c9ceb9d2d8d413d1abcc5fa7edb402b
c77913be5be5f11a0f20dba1b45b96470099e8a8

fixes #9374

affected devices:
Netgear R6220
Netgear WAC104
Ne [... cut]